asus-linux-drivers/asus-numberpad-driver #34: Assign to device correct layout automatically

ldrahnik, 2022-07-24:


kamack38, 2022-07-28:
I think, that by running `setxkbmap -print` we can get the keyboard layout.
```
xkb_keymap {
	xkb_keycodes  { include "evdev+aliases(qwerty)"	};
	xkb_types     { include "complete"	};
	xkb_compat    { include "complete"	};
	xkb_symbols   { include "pc+pl+inet(evdev)"	};
	xkb_geometry  { include "pc(pc105)"	};
};
```

kamack38, 2022-07-28:
And then we can grep `aliases(qwerty)` or `aliases(azerty)`.

ldrahnik, 2022-07-29:
Can you please start a new branch and commit this to `install.sh` script? So if found successfully `aliases(qwerty)` or `aliases(azerty)` do not ask for it.
 
- a misconfigured type of keyboard layout means that the driver does not work correctly, so do not allow the manual option when it is found automatically
- automatically detection exception (not found `xkb_keymap` has not to shutdown entire `install.sh` script)

ldrahnik, 2022-07-29:
And then is required to add code even to `asus_touchpad.py` because the layout can be changed easily:

`setxkbmap fr` # -> to azerty

So in `asus_touchpad.py`:

- join all threads to only one running with 0.5s interval
- this one start only when is at least one condition for background checking fulfilled
- add function with the same code as to `install.sh`, set up variable `percentage_key`

ldrahnik, 2022-07-29:
Can you commit both? I marked it as a bug because numpad percent key does not work when is layout changed.

ldrahnik, 2022-07-29:
Also, the second part of this issue (feature) is to recognize which type of numpad device the user has. Would be a nice to guess the type of numpad keys layout from manufacturer number if possible.

ldrahnik, 2022-07-29:
What is your output (bus, version, product id..) for touchpad using command `evtest --grab` (second part)?

kamack38, 2022-07-29:
What is `evtest`?

kamack38, 2022-07-29:
For `Asus Touchpad/Numpad"
```
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Asus Touchpad/Numpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 14 (KEY_BACKSPACE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 69 (KEY_NUMLOCK)
    Event code 71 (KEY_KP7)
    Event code 72 (KEY_KP8)
    Event code 73 (KEY_KP9)
    Event code 74 (KEY_KPMINUS)
    Event code 75 (KEY_KP4)
    Event code 76 (KEY_KP5)
    Event code 77 (KEY_KP6)
    Event code 78 (KEY_KPPLUS)
    Event code 79 (KEY_KP1)
    Event code 80 (KEY_KP2)
    Event code 81 (KEY_KP3)
    Event code 82 (KEY_KP0)
    Event code 83 (KEY_KPDOT)
    Event code 96 (KEY_KPENTER)
    Event code 98 (KEY_KPSLASH)
    Event code 140 (KEY_CALC)
```

ldrahnik, 2022-07-29:
You picked up this driver. Please original Touchpad, so in your case it is `event11`.

kamack38, 2022-07-29:
```
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x4f3 product 0x319a version 0x100
Input device name: "ASUE1403:00 04F3:319A Touchpad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 325 (BTN_TOOL_FINGER)
    Event code 328 (BTN_TOOL_QUINTTAP)
    Event code 330 (BTN_TOUCH)
    Event code 333 (BTN_TOOL_DOUBLETAP)
    Event code 334 (BTN_TOOL_TRIPLETAP)
    Event code 335 (BTN_TOOL_QUADTAP)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   3906
      Min        0
      Max     4036
      Resolution      32
    Event code 1 (ABS_Y)
      Value   1562
      Min        0
      Max     2299
      Resolution      31
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     4036
      Resolution      32
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     2299
      Resolution      31
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        2
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
  Event type 4 (EV_MSC)
    Event code 5 (MSC_TIMESTAMP)
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 2 (INPUT_PROP_BUTTONPAD)
```

ldrahnik, 2022-07-30:
Well, is required have more ids to prove any association in (part of) touchpad product id and between numpad layout type.

ldrahnik, 2022-07-30:
From my fast investigation, there might be an association between touchpad product id and numpad key layout. Anyway do not know how to concretely pick up right numpad key layout automatically even when would be this prediction correct = exists serie of numbers (touchpad id) separated to specific ranges as have been used specific number key layouts. 

[offline variant] Only do a guess "Your number we do not know yet (it can be automatically done only when we know exactly this key layout use this laptop), we think you have / recommend use you have this layout during installation or this layout, for this could be used specific file with laptop product id and key layout in repository. With more rows would be a better prediction.

[or online variant, do not know if possible] Or try go deeper and check if any Asus page provides for example on laptop / touchpad specification page which key layout is used.

ldrahnik, 2022-08-01:
Table provided by Asus support with all laptops with numpad: [numpad.xlsx](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/9232793/numpad.xlsx)

closed 2022-08-03 by ldrahnik