asus-linux-drivers/asus-numberpad-driver #42: [Feature Request] Configuration file

kamack38, 2022-08-03:
### Short Description

Create config file to set layout independent configurations like activation time.

ldrahnik, 2022-08-06:
### Variables associated with each NumberPad model (still might be overridden with custom value / possibly created own new one):

TODO: will not be `*.py` in future, think about new extension (C has headers for EV_KEYS aswell but think how made it easily configurable for end-user, default NumberPad layouts could be hardcoded in driver anyway is necessary have possibility create new one without overriding driver code)

- keys
- keys_ignore_offset
- keys_activation_times
- top_left_icon_width
- top_left_icon_height
- top_left_icon_keys # [default `[EV_KEY.KEY_CALC]`] # association functionality (script) should be implemented outside of this driver just listening this key
- top_right_icon_width
- top_right_icon_height
- top_right_icon_keys # [default `[EV_KEY.KEY_NUMLOCK]`]
- top_right_icon_is_on_top_left (has priority over top_left_icon in order) # will be removed, as below
- top_offset
- right_offset
- bottom_offset
- left_offset
- touchpad_physical_buttons_are_inside_numpad
- top_left_icon_slide_func_activate_numpad # association functionality (script) should be implemented outside of this driver just listening key 
- backlight_levels

### Config

atm:
CONFIG_FILE_NAME = "asus_touchpad_dev"
["main"] 

but idea is every option below would respond to a specific file in the device interface
allowed values, default value and description is in brackets

brightness # [<hex value>] # last brightness
auto_disable_due_inactivity_time # [<positive number>, 0; default 60], 0 means function is disabled = do not auto disable due inactivity
multitouch # [1,0; default 0]
one_touch_key_rotation # [1,0; default 0]
key_repetitions # [1,0; default 0]
activation_time_left_icon # [<positive number>, default 1]
activation_time_right_icon # [<positive number>, default 1]
try_times # [<positive number>, default 5] # probably will be removed, need to look at why variable exists
try_sleep # [<positive float>, default 0.1] # probably will be removed, need to look at why variable exists
touchpad_disables_numpad # [1,0; default 1]
top_left_icon_slide_func_activation_x_ratio # [<positive float in range 0-1>, default 0.3]
top_left_icon_slide_func_activation_y_ratio # [<positive float in range 0-1>, default 0.3]
TODO: sys_numlock_enables_numpad # depends on work in issue  https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/35
default_backlight_level # [<hex value>, default `0x01`]
top_left_icon_brightness_func # [1,0; default 1 but it is <depending value on not empty backlight_levels>] # BUT file should exists only (should be looked to this variable only) if exist an array of backlight levels, otherwise automatically 0
top_left_icon_activate_numpad # [1,0; default 1]

### TODO:

- Refactor `top_right_icon_is_on_top_left`

Remove `top_right_icon_is_on_top_left` which is used only in layout `gx551`. So the top left key would not be set `None`, instead would be used `EV_KEY.KEY_NUMLOCK` with ignored offsets like this:

```
keys_ignore_offset = [
    [0, 0]
]
``` 

and `top_right_icon_width` and `top_right_icon_height` not set. Implementation of possibility set  `EV_KEY.KEY_NUMLOCK` has to be added inside keys with the same functionality as right_icon has atm.

```
keys_activation_times = [
   [[0,4], 1]
]

Also has to be renamed `top_right_icon_activation_time` to `activation_time`. Also has to be added `top_right_icon_key`. Has to be reimplemented `top_right` (and even `top_left`) internally, faked with some constant or keep outside of array probably because we do not wont allow cross finger to the same key border as it is not anymore `top_right` or `top_left` area. 

- Rename `last_brightness` back to `brightness` as it is normally used in other device interfaces without the last prefix

- Bug: `EV_KEY.KEY_CALC` as key for the top left icon is atm hardcoded 

- Bug: auto disable `top_left_icon_brightness_func` when is empty array of backlights

ldrahnik, 2022-08-12:
**TODO:**

 - TODO in code: if is numlock not found in keys array and right icon has no dimensions, in this case auto enable `sys_numlock_enables_numpad`
 - Readme.md is broken, I separated it to 2 tables
 - config file might not be created because of permission issues (check if it works)

ldrahnik, 2022-08-13:
**TODO:**
- add `enabled` config entry and add checking via thread with autoenable; the same for brightness changed via file and maybe more

closed 2022-08-19 by ldrahnik