asus-linux-drivers/asus-numberpad-driver #45: Kernel module

ldrahnik, 2022-08-06:


Stivux, 2025-04-26:
Hello!

I made some very simple and dummy draft:
```
#include <linux/i2c.h>
#include <linux/init.h>
#include <linux/module.h>
#include <linux/printk.h>
#include <linux/string.h>

MODULE_AUTHOR("Author");
MODULE_DESCRIPTION("Asus NumberPad LED control LKM");
MODULE_LICENSE("GPL");
MODULE_VERSION("0.1");

static struct i2c_client *client = NULL;

static int send_command(u8 value)
{
	u8 cmd[] = { 0x05, 0x00, 0x3d, 0x03, 0x06,  0x00, 0x07,
		     0x00, 0x0d, 0x14, 0x03, value, 0xad };

	struct i2c_msg msg = {
		.addr = client->addr,
		.flags = 0,
		.len = sizeof(cmd),
		.buf = cmd,
	};
	return i2c_transfer(client->adapter, &msg, 1);
}

static int i2c_numberpad_match(struct device *dev, void *data)
{
	if (client != NULL) {
		return 0;
	}

	const char *name = dev_name(dev);
	if (strstr(name, "ASUE140D") != NULL && dev->parent != NULL) {
		client = i2c_verify_client(dev);
	}
	if (IS_ERR(client)) {
		pr_err("Failed to create i2c client\n");
		return PTR_ERR(client);
	}
	return 0;
}

static int __init asus_numberpad_led_init(void)
{
	i2c_for_each_dev(NULL, i2c_numberpad_match);
	if (client == NULL) {
		pr_err("Failed to match ASUS NumberPad LED device\n");
		return -ENODEV;
	}
	pr_info("clienr addr: %d", client->addr);
	send_command(0x1);
	return 0;
}

static void __exit asus_numberpad_led_cleanup(void)
{
	pr_info("Removing ASUS NumberPad LED driver");
	send_command(0x0);
}

module_init(asus_numberpad_led_init);
module_exit(asus_numberpad_led_cleanup);

```

It works (I mean it can enable led when I do insmod and disable it after rmmod).

ldrahnik, 2025-04-26:
@Stivux Hi, thank you for the message. Would be great to kick-off this finally :) I saw your repository on your profile. This project is especially very good at autorecommending layouts from the data (yes, gathered data in .csv from the GA has to be put manually into the repository but that is the only flaw in beauty, almost automatic), a lot of data is already gathered, but, how would this work in the kernel driver? Because GA would not be connected so feedback would be only from the provided logs from the people I guess. I think about it like hardcode into c/c++ file these already gathered data https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/laptop_touchpad_numberpad_layouts.csv but what about the way of manual selection? Why I am writing this because I am from this reason not convinced move the development of these drivers to the kernel drivers only because selecting is good for people that are not programmers, do not have github, will not create issue, just have newest laptop that is not supported yet and downloaded .zip and run install, do not even know how to for example edit one file, change something, rebuild, reinstall a module. What do you think about this? Gathering data via GA helps maintain this project IMHO a lot. Btw I realized I should probably allow select Touchpad manually also as the names change a lot.

Stivux, 2025-04-26:
>  I’m not fully convinced about moving the entire development into kernel drivers only

Sure. In my opinion, the module is needed for friendly (if it's appropriate to say so in case kernel level) communication with the device. For example, communication via `sysfs`, as  we have for some keyboard led `/sys/devices/platform/asus-nb-wmi/leds/platform::micmute` (You can write/read 0 or 1 to/from `brightness` file in this dir to control led of mic key).

At the same time, there are many problems associated with the need to recompile the module. Of course, you can write userspace programm/script that collect stats, fetch module source, compile it and insert in the kernel.  
But the ideal module is the one that is part of kernel source tree. 

By the way, there is another official recommended way to implement approximately what we need to control touchpad and led behavior.

As I mentioned in #224 touchpad and its led can be controlled using hid. After a little research I found https://docs.kernel.org/hid/hid-bpf.html.

HID-BPF is "modern" (since kernel 6.3) way to customize HID behavior without writing real Linux Kernel Module (but bpf programs run in kernel space). Using HID-BPF it is possible to create kernel program, that catch and process hid events, like touch, swipe, ...  It looks like it's a good option because it does not require recompilation for every kernel version.

ldrahnik, 2025-04-27:
@Stivux 
>It can be implemented using procfs or sysfs (pass some parameter to special file in /proc or /sys for manual selection of current layout). 
>Yeah. Writing kernel driver needs a lot of hardcoding.

You are right, I forgot layout can be determined by hard-coded values (gathered data) and If was not found match just left (expose) configurable default in config file (virtual file) for the module (device). 

> HID-BPF is "modern" (since kernel 6.3) way to customize HID behavior without writing real Linux Kernel Module (but bpf programs run in kernel space). Using HID-BPF it is possible to create kernel [...]

It sounds good, it would extend Touchpad behaviour but I do not want rewrite it to the BPF and then as kernel module, so twice and it looks like its up to user which these programs include so would be problematic to distribute it to every asus user. I prefer kernel driver. What do you prefer? Because you wrote: But the ideal module is the one that is part of kernel source tree I guess the same.

I would start developing prototype, I will find some time. Options where start development I would prefer this org, I can add you here for writing access and we can create a new repository (something like asus-numberpad-kernel-driver, with goal to be merged to the kernel source tree), few people have already follow here and could help with testing and org sounds better to me.

Stivux, 2025-04-27:
@ldrahnik 

> I would prefer this org

Yes. It's a good idea.

By the way, I was thinking about using Rust, but it seems that at the moment there are no bindings in the Rust kernel subsystem to write i2c or hid drivers. Therefore, we will have to use C.

ldrahnik, 2025-04-27:
Everything, not just i2c. Yes, in C. Look at this work, link Is in README aswell https://github.com/chovine/asus_numpad_overlay

closed 2025-08-24 by ldrahnik

ldrahnik, 2025-11-24:
WIP here https://github.com/Batoo76/asus-numberpad-driver/tree/Conversion-to-compiled-version