asus-linux-drivers/asus-numberpad-driver #47: New laptop model for existing layout, slide not satisfactory

Spindletop, 2022-08-09:
Detected laptop: ZenBook UX425UA_UM425UA
Selected model keypad layout: up5401ea.py

Basic functionalities work :-)
[image]

But the slide function (touch top left corner, slide to right) is annoying:
- slide - it is detected because mouse cursor does not move, ok
- then nothing happens (in Windows the NumberPad is activated and calc app is opened), not ok
- then after touching somewhere else in the touchpad region, the mouse pointer can be seen moving around for about one second, then the NumberPad is activated and the mouse pointer disappears, not ok
[ux425ua.webm](https://user-images.githubusercontent.com/8633642/183663153-2782ee41-b43c-42a7-9a5d-a743861eae11.webm)

Sometimes after the slide from top left corner, the NumberPad cannot even be turned off, it is only deactivated after the 60s timeout or by disabling/enabling the touchpad

ldrahnik, 2022-08-09:
Hi, thank you for creating an issue. Incorrectly ended untouch of slide from top left icon in code should be fixed in linked branch. Can you test it please?

ldrahnik, 2022-08-09:
And when is slide longer than 30% of width and 30% of the height it should start by default NumberPad calculator. You should see in command `sudo libinput debug-events` this as me:

```
 event24  DEVICE_ADDED     Asus Touchpad/Numpad              seat0 default group14 cap:k
 ...
 event24  KEYBOARD_KEY     +5.375s	KEY_CALC (140) pressed
 event24  KEYBOARD_KEY     +5.375s	KEY_CALC (140) released
```

Then you can bind the function as a calculator to key `KEY_CALC` if you do not have it by default.

Spindletop, 2022-08-09:
Thanks for fast processing :-)

The issue with the NumberPad activation after slide seems to be fixed in branch `bug-slide-is-not-cleared-slot-when-is-done-untouch`, but now with your info about `KEY_CALC` I think this is "more faulty". The slide itself does not log any event (probably never does) but now also no follow up action, only the other actions are logged, e.g.:
```
 (...)
 event11  GESTURE_HOLD_BEGIN      +4.819s       1                                | Single Tap Begin
 event11  GESTURE_HOLD_END        +4.904s       1                                | Single Tap End
 event11  POINTER_MOTION          +6.019s         0.15/  0.00 ( +1.00/ +0.00)    |
 event11  GESTURE_HOLD_BEGIN      +6.038s       1                                | Tap move begin
 event11  POINTER_MOTION          +6.067s        -0.16/  0.00 ( -1.00/ +0.00)    | .
 event11  POINTER_MOTION          +6.074s        -0.92/  0.00 ( -3.00/ +0.00)    | .
 event11  POINTER_MOTION          +6.081s        -1.36/  0.00 ( -4.00/ +0.00)    | .
 event11  POINTER_MOTION          +6.088s        -2.38/ -0.34 ( -7.00/ -1.00)    | .
 event11  POINTER_MOTION          +6.095s        -3.06/ -0.68 ( -9.00/ -2.00)    | .
 event11  GESTURE_HOLD_END        +6.095s       1 cancelled                      | Tap move end
 event11  POINTER_MOTION          +6.102s        -4.08/ -1.02 (-12.00/ -3.00)    |
 event11  POINTER_MOTION          +6.109s        -4.42/ -1.36 (-13.00/ -4.00)    |
 (...)                                                                           |
 event11  POINTER_MOTION          +6.289s        -0.68/  0.34 ( -2.00/ +1.00)    |
 event11  POINTER_MOTION          +6.296s        -0.68/  0.34 ( -2.00/ +1.00)    |
-event13  KEYBOARD_KEY            +8.643s       *** (-1) pressed                 | NumberPad activate end (1/2) (begin not logged but 2 sec earlier)
 event13  KEYBOARD_KEY            +8.643s       *** (-1) released                | NumberPad activate end (2/2)
 event13  KEYBOARD_KEY            +12.018s      *** (-1) pressed                 | NumberPad button 1 (1/2)
 event13  KEYBOARD_KEY            +12.019s      *** (-1) released                | NumberPad button 1 (2/2)
1 event13  KEYBOARD_KEY            +12.302s     *** (-1) pressed                 | NumberPad button 2 (1/2)
 event13  KEYBOARD_KEY            +12.302s      *** (-1) released                | NumberPad button 2 (2/2)
2 event13  KEYBOARD_KEY            +12.586s     *** (-1) pressed                 | NumberPad button 3 (1/2)
 event13  KEYBOARD_KEY            +12.587s      *** (-1) released                | NumberPad button 3 (2/2)
3 event13  KEYBOARD_KEY            +15.720s     *** (-1) pressed                 | NumberPad deactivate end (1/2) (begin not logged but 2 sec earlier)
 event13  KEYBOARD_KEY            +15.720s      *** (-1) released                |
-(..)                                                                            |
```

In branch `master` I get the `KEY_CALC` event after the slide, but the activation of the NumberPad is only after a pointer motion, which unfortunately is not logged:
```
 event11  POINTER_MOTION          +4.659s        -3.40/ -0.34 (-10.00/ -1.00)
 event11  POINTER_MOTION          +4.666s        -3.06/ -0.34 ( -9.00/ -1.00)
-event13  KEYBOARD_KEY            +4.688s       KEY_CALC (140) pressed
 event13  KEYBOARD_KEY            +4.688s       KEY_CALC (140) released
 event13  KEYBOARD_KEY            +4.688s       *** (-1) pressed
 event13  KEYBOARD_KEY            +4.688s       *** (-1) released
```
So currently I would say that the master-branch is correct, but the required pointer motion disturbing... And I have to have a look for the `KEY_CALC` binding ;-)

ldrahnik, 2022-08-09:
For key binding `KEY_CALC` look here, I had to bound it by myself too. Elementary OS Loki 6.1 https://github.com/ldrahnik/elementary-os-scripts

Spindletop, 2022-08-09:
Not sure about the way for binding keys in KDE other than with the gui, with branch `master` and the slide + (buggy) pointer I could capture the "shortcut":
[image]
 which will results in the configuration:
```
[org.kde.kcalc.desktop]
_k_friendly_name=KCalc
_launch=Launch (1),none,KCalc
```
but the key event is still somewhere else hidden in the depths of the desktop environment...

This may exceed the scope of `asus-linux-drivers`, but as this behaviour differs from Windows as it only opens the application but does not close it, it may be noteworthy that this still can be accomplished with a small hack.
e.g. instead of the assigning the event to the application, create a script file somewhere, e.g. `toggle_calc.sh`, in KDE Shortus click `Add Application` and assign the script, capture the sliding motion event from the touchpad, voilà, same as in Windows :-) (excluding the above mentioned bug)

```
#!/bin/bash

# If running, close it, else open program
if pgrep kcalc; then
	pkill kcalc
else
	kcalc &
fi
```
This may also be useful for the gnome evironment.

ldrahnik, 2022-08-09:
It is definitely bug, thank you for reporting because slide was activated even when was ended unsuccessfully and than was random touch done with arena 30% width and 30 height from top left key. I pushed new commits to the linked branch. Can you test it please?

ldrahnik, 2022-08-09:
Yeah, I am thinking in the same way. Drivers should only send appropriate event (in this case event) and additionally functionality like toggling should be inside script bound to that key. Your solution is clear in my opinion.

ldrahnik, 2022-08-09:
I am interested in if can be changed application for calculating in windows. How official driver recognize which application to close. Because is marked as default application for calculating? I do not know windows very well. In linux everyone is using another app I guess. That is main problem and second there does not exist across distribution place where is keep information about default program for calculating.

Spindletop, 2022-08-10:
> It is definitely bug, thank you for reporting because slide function was start even when was slide ended unsuccessfully (< 30% w & h) and than was done random touch in area 30% width and 30% height from the top left key. I pushed new commits to the linked branch. Can you test it please?

Now I get it, the sliding motion additionally needs 30% height :exploding_head: - this wasn't the case in Windows (only a few percent top-left to the right was enough) so I did not even try :blush:.
Therefore this behaviour did already work in `master`, but in 950316c2200081471e516477e6e22d566d8a4852 the bug "slide 30% width 0% height + touch 50% w 50% h" is fixed :+1: 

The "slide 30% width 30% height" when NumberPad is off sends key event and activates NumberPad. Log:
```
 event13  KEYBOARD_KEY            +1.171s       KEY_CALC (140) pressed
 event13  KEYBOARD_KEY            +1.171s       KEY_CALC (140) released
 event13  KEYBOARD_KEY            +1.171s       *** (-1) pressed
 event13  KEYBOARD_KEY            +1.171s       *** (-1) released
```
But when the NumberPad is already on, with "slide 30% width 30% height" only the key event is sent and the NumberPad does not deactivate. Log:
```
 -event13  KEYBOARD_KEY            +4.994s       KEY_CALC (140) pressed
 event13  KEYBOARD_KEY            +4.994s       KEY_CALC (140) released
```

> I am interested in if can be changed application for calculating in windows

Application in Windows can be changed via registry or a tool accessing it (`[HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Explorer\AppKey\18]`), but as I replaced Windows on the Asus, I cannot test it :-(. However maybe the driver on the Asus does something different, as I remember a different calculator application may have been opened with the numberPad...

By the way, running `/somepath/asus-linux-drivers/asus-touchpad-numpad-driver/install.sh` fails because the scripts wants to access files relative the the executing directory, so only `./install.sh` works...

ldrahnik, 2022-08-10:
> By the way, running /somepath/asus-linux-drivers/asus-touchpad-numpad-driver/install.sh fails because the scripts wants to access files relative the the executing directory, so only ./install.sh works...

Should be fixed

ldrahnik, 2022-08-10:
> Now I get it, the sliding motion additionally needs 30% height exploding_head - this wasn't the case in Windows (only a few percent top-left to the right was enough) so I did not even try blush.

I decreased the default activation values to 5% of width and height.

ldrahnik, 2022-08-10:
> Now here it becomes complicated what the the expected result should be and what is possible. In Windows the calculator is closed and NumberPad deactivated (in my opinion the perfect behaviour). With the Linux driver another calculator is opened (or whatever is associated with the event). I would prefer disabling the NumberPad because it is faster than holding top right and does not require the 2s delay (also disabling often even does not work on my machine), but when in doubt keep the current behaviour smile.

I refactored the default behaviour of the driver to be prepared for adding toggling script to key `KEY_CALC`.

ldrahnik, 2022-08-10:
> (also disabling often even does not work on my machine)

I tried enabling and disabling in row a lot times but everything seems to be working.

closed 2022-08-11 by ldrahnik