asus-linux-drivers/asus-numberpad-driver #50: [Bug]: After sleep have to restart the asus_touchpad_numpad systemd service

imifarago, 2022-08-10:
### Describe the bug

After I wake the laptop from the sleep, the touchpad numpad doesn't activate by pressing the top right of the touchpad, it works just a single click.

To work properly again, have to do a `systemctl restart asus_touchpad_numpad` 

### Expected behavior

After wake up from sleep should work without manually restart the service.

Probably something like that (don't know, it can be solved in one service).

```
[Unit]
Description=Restart Touchpad Numpad after resume
After=suspend.target

[Service]
Type=simple
ExecStart=/bin/systemctl --no-block restart asus_touchpad_numpad

[Install]
WantedBy=suspend.target
```

### Desktop

- Program version: this git repo, commit 0d0079a48a8442851e5f7353dae839e6bcf44e44 
- Distro: Manjaro 
- Kernel: 5.19.0-2-MANJARO

ldrahnik, 2022-08-10:
Hi, thank you for creating an issue. I implemented your proposed solution with a second service to linked branch with this issue. Can you please confirm it works? I do not use atm suspend on my laptop.

imifarago, 2022-08-10:
It was super fast :)
I'll try it soon, probably later today.

ldrahnik, 2022-08-11:
@imifarago Ping

imifarago, 2022-08-12:
@ldrahnik sorry for late, i've tested it yesterday and worked for me
thanks!

closed 2022-08-12 by ldrahnik

ldrahnik, 2023-01-29:
@imifarago Hello there, suspend service unfortunately was causing black screen during waking up. I tested suspend functionality now without specific service in clearly installed Elemementary OS 6.1 Jólnir (based on Ubuntu 20.04) and NumberPad was working after woke up fine so service I have removed. Could you please test in clearly installed system Manjaro it does not work? Issue can be reopened.