asus-linux-drivers/asus-numberpad-driver #53: [Bug]: Service keeps restarting on startup

Mark12870, 2022-08-13:
### Describe the bug

I think the script was keeping restarting on startup, so after login the laptop froze.

I have solved it by adding `RestartSec=60s` to the `asus_touchpad_numpad.service` file. (I just wanted to let you know.)

### Expected behavior

Laptop shouldn't be freezing.

### Desktop

- Program version [master?]
- Distro [Ubuntu 22.04.1]
- Kernel [5.15.0-46-generic]

ldrahnik, 2022-08-13:
Hi, thank you for creating an issue. 

Can you please provide log from `systemctl` when is laptop started? With context please, so please include all services. I will look at that.

Mark12870, 2022-08-13:
Here is some output of `journalctl -u asus_touchpad_numpad.service`:
```
-- Boot 3d6ff9ff7ef04a958d31dd7686e04187 --
srp 13 21:33:31 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: start-p>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Control>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Failed >
srp 13 21:34:37 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Schedul>
srp 13 21:34:37 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:34:37 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
 ESCOD
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: start-p>
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Control>
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Failed >
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Schedul>
srp 13 21:23:49 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:23:49 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:23:52 marek-zenbook systemd[1]: Stopping Asus Touchpad to Numpad Hand>
srp 13 21:23:53 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Deactiv>
srp 13 21:23:53 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
-- Boot 6e78383af5d94c5cb696013ac7547203 --
srp 13 21:32:03 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:33:11 marek-zenbook systemd[1]: Stopping Asus Touchpad to Numpad Hand>
srp 13 21:33:11 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Deactiv>
srp 13 21:33:11 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:33:11 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Consume>
-- Boot 3d6ff9ff7ef04a958d31dd7686e04187 --
srp 13 21:33:31 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: start-p>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Control>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Failed >
srp 13 21:34:37 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Schedul>
srp 13 21:34:37 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:34:37 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
 ESCOD
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: start-p>
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Control>
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Failed >
srp 13 21:23:49 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Schedul>
srp 13 21:23:49 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:23:49 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:23:52 marek-zenbook systemd[1]: Stopping Asus Touchpad to Numpad Hand>
srp 13 21:23:53 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Deactiv>
srp 13 21:23:53 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
-- Boot 6e78383af5d94c5cb696013ac7547203 --
srp 13 21:32:03 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:33:11 marek-zenbook systemd[1]: Stopping Asus Touchpad to Numpad Hand>
srp 13 21:33:11 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Deactiv>
srp 13 21:33:11 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:33:11 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Consume>
-- Boot 3d6ff9ff7ef04a958d31dd7686e04187 --
srp 13 21:33:31 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: start-p>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Control>
srp 13 21:33:36 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Failed >
srp 13 21:34:37 marek-zenbook systemd[1]: asus_touchpad_numpad.service: Schedul>
srp 13 21:34:37 marek-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handl>
srp 13 21:34:37 marek-zenbook systemd[1]: Started Asus Touchpad to Numpad Handl>
lines 83-105/105 (END)
```

ldrahnik, 2022-08-13:
Can you please set up simple as type of service `asus_touchpad.service`, restart laptop and look at log again?

Mark12870, 2022-08-13:
Here is some log from previous sessions.
[log.zip](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/9332165/log.zip)

I'm gonna try the `simple` thing.

Mark12870, 2022-08-13:
I'm not sure if I was lucky before but now it doesn't start with either config. :/

ldrahnik, 2022-08-13:
Posted log looks almost ok, it was in the same second started successfully (Aug 13 21:23:49). Hm, hm, should be enough set `graphical.target` as driver requires xorg server and then just let the driver start.

Mark12870, 2022-08-13:
I'm on wayland btw. Might be also the difference.

Mark12870, 2022-08-13:
Maybe the Environment is wrong
[image]

ldrahnik, 2022-08-13:
Twice record of `Environment` should be ok.

Yeah, `wayland` I not tested, only `xorg display server`. Anyway is used in this driver (`xorg`) only for checking touchpad status (enabled/disabled via `xinput` command), if this is a problem, should be enough set `touchpad_disables_numpad = False` in used layout and this endless cycle will not run. But this functionality will not probably work because `xinput` i guess wayland does not have.

ldrahnik, 2022-08-13:
Can you post please log `/var/log/asus_touchpad_numpad-driver/error.log` of this driver, we should see that `xinput` failed, a lot of:

`
Getting Device Enabled via xinput failed
`

Mark12870, 2022-08-13:
[error.zip](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/9332215/error.zip)

I think I will rather continue tomorrow. Good night and thanks for help.

ldrahnik, 2022-08-13:
Thank you for your report and time. Anyway adding suport of `wayland` for checking device status is I think different problem. It should max do lags during using driver. It should not affect starting of service.

ldrahnik, 2022-08-13:
Adding support for `wayland` would be great and is it necessary because `xorg` is older. Feel free make pull request or if you know how `wayland` check devices enabled/disabled status would be helpful for implementation.

Mark12870, 2022-08-14:
Also I have found out that the "%" button triggers some nonsence.
I use `up5401ea` config.

ldrahnik, 2022-08-14:
Can you confirm unicode characters do not work on your laptop manually? Unicode code for "%" is `25` so hold (`left_shift` + `left_ctrl` + `U`) and then press `2`, unpress `2`, then press `5`, unpress `5` and release (`left_shift` + `left_ctrl` + `U`).

Mark12870, 2022-08-14:
> Can you confirm manually that unicode characters do not work on your laptop? Unicode code for `"%"` is `25` so hold (`left_shift` + `left_ctrl`) and then press `U`, unpress `U`, press `2`, unpress `2`, then press `5`, unpress `5` and release (`left_shift` + `left_ctrl`).

It works, but I've found out that the numpad `%` button triggers the unicode charater only. Like this: 
[image]

Mark12870, 2022-08-14:
I still had problem with the freezing. I changed the service config to look like this and it seems to finally work properly:
```
[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea
StandardInput=tty-force
StandardOutput=append:/var/log/asus_touchpad_numpad-driver/error.log
StandardError=append:/var/log/asus_touchpad_numpad-driver/error.log
TimeoutSec=5
Restart=on-failure
RestartSec=60s
# For some operating systems with boot failure, adding this following property might resolve the issue  
# ExecStartPre=/bin/sleep 10
# Environment="DISPLAY=:0"

# [Install]
# WantedBy=graphical.target
```
But I will try few more restarts. :)

ldrahnik, 2022-08-14:
I do not understand. So shortcut (left_shift + left_ctrl) and then press U, unpress U, press 2, unpress 2, then press 5, unpress 5 and release (left_shift + left_ctrl) print character "%"? But in driver it does not work? Can you please record it via command `evtest`:

```
evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Asus WMI hotkeys
/dev/input/event5:	ELAN9009:00 04F3:2E36
/dev/input/event6:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event7:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event8:	ELAN9009:00 04F3:2E36 Stylus
/dev/input/event9:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event10:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event11:	ASUE140D:00 04F3:31B9 Mouse
/dev/input/event12:	ASUE140D:00 04F3:31B9 Touchpad
/dev/input/event13:	ASUE140D:00 04F3:31B9 Keyboard
/dev/input/event14:	Video Bus
/dev/input/event15:	Intel HID events
/dev/input/event16:	ELAN9009:00 04F3:2E36 UNKNOWN
/dev/input/event17:	USB2.0 HD UVC WebCam: USB2.0 HD
/dev/input/event18:	sof-hda-dsp Headphone
/dev/input/event19:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event20:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event21:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event22:	Asus Stylus
/dev/input/event23:	Asus WMI hotkeys
/dev/input/event24:	Asus Touchpad/Numpad
Select the device event number [0-24]: 3
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
[... 147 lines cut ...]
Event: time 1660486997.761752, type 1 (EV_KEY), code 22 (KEY_U), value 2
Event: time 1660486997.761752, -------------- SYN_REPORT ------------
Event: time 1660486997.791791, type 4 (EV_MSC), code 4 (MSC_SCAN), value 16
Event: time 1660486997.791791, type 1 (EV_KEY), code 22 (KEY_U), value 2
Event: time 1660486997.791791, -------------- SYN_REPORT ------------
Event: time 1660486997.796811, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1660486997.796811, type 1 (EV_KEY), code 3 (KEY_2), value 1
Event: time 1660486997.796811, -------------- SYN_REPORT ------------
Event: time 1660486997.876919, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1660486997.876919, type 1 (EV_KEY), code 3 (KEY_2), value 0
Event: time 1660486997.876919, -------------- SYN_REPORT ------------
Event: time 1660486998.237861, type 4 (EV_MSC), code 4 (MSC_SCAN), value 06
Event: time 1660486998.237861, type 1 (EV_KEY), code 6 (KEY_5), value 1
Event: time 1660486998.237861, -------------- SYN_REPORT ------------
Event: time 1660486998.322992, type 4 (EV_MSC), code 4 (MSC_SCAN), value 06
Event: time 1660486998.322992, type 1 (EV_KEY), code 6 (KEY_5), value 0
Event: time 1660486998.322992, -------------- SYN_REPORT ------------
Event: time 1660486998.648756, type 4 (EV_MSC), code 4 (MSC_SCAN), value 16
Event: time 1660486998.648756, type 1 (EV_KEY), code 22 (KEY_U), value 0
Event: time 1660486998.648756, -------------- SYN_REPORT ------------
Event: time 1660486998.658770, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1d
Event: time 1660486998.658770, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1660486998.658770, -------------- SYN_REPORT ------------
%
```

Mark12870, 2022-08-14:
Sorry, it works a bit differently. I press and release `ctrl + shift + u`, then I press and release `2`, then I press and release `5`. And then I need to press and release `enter` key.

ldrahnik, 2022-08-14:
Now it does make sense. Can you anyway record that please.

Mark12870, 2022-08-14:
At the end is recorded `ctrl + c` to end the evtest. :upside_down_face:
```
marek@marek-zenbook:~$ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	AT Translated Set 2 keyboard
/dev/input/event3:	Video Bus
/dev/input/event4:	Asus WMI hotkeys
/dev/input/event5:	USB2.0 HD UVC WebCam: USB2.0 HD
/dev/input/event6:	HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event7:	ELAN9008:00 04F3:2E36
/dev/input/event8:	ELAN9008:00 04F3:2E36 UNKNOWN
/dev/input/event9:	ELAN9008:00 04F3:2E36 UNKNOWN
/dev/input/event10:	ELAN9008:00 04F3:2E36 Stylus
/dev/input/event11:	ELAN9008:00 04F3:2E36 UNKNOWN
/dev/input/event12:	ELAN9008:00 04F3:2E36 UNKNOWN
/dev/input/event13:	ELAN9008:00 04F3:2E36 UNKNOWN
/dev/input/event14:	Microsoft Microsoft® 2.4GHz Transceiver v8.0
/dev/input/event15:	Microsoft Microsoft® 2.4GHz Transceiver v8.0 Mouse
/dev/input/event16:	Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control
/dev/input/event17:	Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control
/dev/input/event18:	Microsoft Microsoft® 2.4GHz Transceiver v8.0 System Control
/dev/input/event19:	HD-Audio Generic HDMI/DP,pcm=7
/dev/input/event20:	HD-Audio Generic HDMI/DP,pcm=8
/dev/input/event21:	ASUE140D:00 04F3:31B9 Mouse
/dev/input/event22:	ASUE140D:00 04F3:31B9 Touchpad
/dev/input/event23:	ASUE140D:00 04F3:31B9 Keyboard
/dev/input/event24:	Asus Touchpad/Numpad
/dev/input/event25:	HD-Audio Generic Headphone
Select the device event number [0-25]: 2
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
[... 149 lines cut ...]
Event: time 1660488328.487818, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1660488328.487818, type 1 (EV_KEY), code 3 (KEY_2), value 1
Event: time 1660488328.487818, -------------- SYN_REPORT ------------
Event: time 1660488328.628066, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1660488328.628066, type 1 (EV_KEY), code 3 (KEY_2), value 0
Event: time 1660488328.628066, -------------- SYN_REPORT ------------
Event: time 1660488329.129710, type 4 (EV_MSC), code 4 (MSC_SCAN), value 06
Event: time 1660488329.129710, type 1 (EV_KEY), code 6 (KEY_5), value 1
Event: time 1660488329.129710, -------------- SYN_REPORT ------------
Event: time 1660488329.219860, type 4 (EV_MSC), code 4 (MSC_SCAN), value 06
Event: time 1660488329.219860, type 1 (EV_KEY), code 6 (KEY_5), value 0
Event: time 1660488329.219860, -------------- SYN_REPORT ------------
Event: time 1660488330.634146, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1660488330.634146, type 1 (EV_KEY), code 28 (KEY_ENTER), value 1
Event: time 1660488330.634146, -------------- SYN_REPORT ------------
Event: time 1660488330.704230, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c
Event: time 1660488330.704230, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1660488330.704230, -------------- SYN_REPORT ------------
Event: time 1660488336.311104, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1d
Event: time 1660488336.311104, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1660488336.311104, -------------- SYN_REPORT ------------
Event: time 1660488336.516701, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2e
Event: time 1660488336.516701, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1660488336.516701, -------------- SYN_REPORT ------------
```

ldrahnik, 2022-08-14:
Pull master branch.

Can you please confirm that refactored unicode sendings form works for you as well?

Mark12870, 2022-08-14:
Everything works now. Great work! :+1:

ldrahnik, 2022-08-14:
For supporting feature **disabling NumberPad when is touchpad disabled** for you with `wayland` I created an issue
https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/54 for creating and installing udev rule together with driver. (so no `xinput` or `wayland` dependencies would be required).

When will be implemented created issue mentioned in previous paragraph will be removed `xorg` rows as you already removed from service file:

```
Environment="DISPLAY=:0"
Environment="XAUTHORITY=$XAUTHORITY"
```

and `[Install]` `graphical.target` changed back to:

```
WantedBy=default.target
```

What I do not like from your solution and what we have to figure out in this thread is why do you need:

```
RestartSec=60s
```

Can you please test if a driver works correctly with `type=simple`, `WantedBy=default.target` and `RestartSec` not used?

Mark12870, 2022-08-15:
I've just tried this config and it seems to work fine:
```
[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea
StandardInput=tty-force
StandardOutput=append:/var/log/asus_touchpad_numpad-driver/error.log
StandardError=append:/var/log/asus_touchpad_numpad-driver/error.log
TimeoutSec=5
Restart=on-failure
[Install]
WantedBy=default.target
```

closed 2022-08-29 by ldrahnik