asus-linux-drivers/asus-numberpad-driver #55: [Bug]: Problem touch with layout g533

Gevert92, 2022-08-15:
### Describe the bug

Some keys of touch pad using layout g533, does not write correctly.

The keys that type well are "1, 2, 3, 4, 5, 6, *, -, +, Enter".
The slash key (/) does not write.
The keys "0, 7, 8, 9, Back" must be pressed several times to type.

### Expected behavior

Each key must write the value at the first contact.

It is desirable that when leaving the key pressed, the value is written several times.

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2022-08-16:
Hello, thank you for creating an issue.  

> It is desirable that when leaving the key pressed, the value is written several times.

For second part please try enable `key_repetitions` in your layout file. So add row `key_repetitions=True`. It is disabled by default.

ldrahnik, 2022-08-16:
> Each key must write the value at the first contact.

Can you please run driver in debug mode:

> LOG=DEBUG sudo -E ./asus_touchpad.py

and post result?

Gevert92, 2022-08-17:
Thank you for your response.

For the first point I put key_repetitions and it worked fine.

For the second part I am not sure if I did it right, but I will detail the tests.

First I uncommented the line you indicate.
[image]

Then I ran the install.sh and this is the result
[image]

As you can see I had to comment out the line "sudo -E ./asus_touchpad.py" because when running the install.sh, ubuntu experiences a syntax error problem.
[image]

ldrahnik, 2022-08-17:
> Troubleshooting
> To show debug logs run:
> LOG=DEBUG sudo -E ./asus_touchpad.py

Not uncomment line but run command. Readme suggest the same.

ldrahnik, 2022-08-19:
Closed because of inactivity of issue owner.

closed 2022-08-19 by ldrahnik

Gevert92, 2022-08-22:
Hi, sorry for the inactivity, I had problems with my pc.

I share the image of the debug, I stopped the service and executed the command

[image]

ldrahnik, 2022-08-24:
Discussion continues here https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/58