asus-linux-drivers/asus-numberpad-driver #57: [Bug]: Config file keeps getting overwritten

davorvr, 2022-08-23:
### Describe the bug

I am trying to change some of the config options in `/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev`. First, I stop the two related systemd services, then I edit the configuration options, and then I start the services. The numberpad is usable again, but behaving according to the default config, and, indeed, the config file is overwritten with the default values.

### Expected behavior

The config file should keep newly specified options, and the numberpad should behave accordingly.

### Relevant log output

```shell
New config file that I'm trying to use:

[main]
numpad_disables_sys_numlock = 0
disable_due_inactivity_time = 60
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 1
one_touch_key_rotation = 0
activation_time = 0.5
sys_numlock_enables_numpad = 0
top_left_icon_activation_time = 0.5
top_left_icon_slide_func_activates_numpad = 1
top_left_icon_slide_func_deactivates_numpad = 1
top_left_icon_slide_func_activation_x_ratio = 1   
top_left_icon_slide_func_activation_y_ratio = 1
top_right_icon_slide_func_activation_x_ratio = 1
top_right_icon_slide_func_activation_y_ratio = 1 
default_backlight_level = 0x01
enabled = 0
top_left_icon_brightness_func_disabled = 0
```

### Desktop

- Program version [e.g. 57c742a]: 5d2023d
- Distro [e.g. Arch Linux]: Fedora 36 Workstation
- Kernel [e.g. 5.18.15.arch1-2]: 5.18.18-200.rog.fc36.s86_64

ldrahnik, 2022-08-23:
Hi, thank you for creating an issue.

ldrahnik, 2022-08-23:
@davorvr Can you test it now, please?

davorvr, 2022-08-23:
Yep, that fixed it! Thanks for the quick resolve, and thank you for your work on this driver!

closed 2022-08-23 by davorvr

ldrahnik, 2022-08-23:
No, me thank to you. I hurried and merged pul request on Yolo (without review). Sorry. At least I have a badge. Put a star is another way how support my/us activity in this repo (y)