asus-linux-drivers/asus-numberpad-driver #58: [Bug]: Problem touch with layout g533

Gevert92, 2022-08-24:
### Describe the bug

I want to follow the thread of the issue #55 which was closed due to inactivity, sorry, I had problems with my pc.

Some keys of touch pad using layout g533, does not write correctly.

The keys that type well are "1, 2, 3, 4, 5, 6, *, -, +, Enter".
The slash key (/) does not write.
The keys "0, 7, 8, 9, Back" must be pressed several times to type.

I share the image of the debug.

### Expected behavior

Each key must write the value at the first contact.

### Relevant log output

```shell
I share the image of the debug "LOG=DEBUG sudo -E ./asus_touchpad.py", I stopped the service and executed the command

![image](https://user-images.githubusercontent.com/42220220/186288245-163ae197-ce59-470c-97a9-2d9602362ebc.png)
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

Gevert92, 2022-08-24:
I share the image of the debug "LOG=DEBUG sudo -E ./asus_touchpad.py", I stopped the service and executed the command

[image]

ldrahnik, 2022-08-24:
Hi, thank you for creating an issue. I improved documentation about debugging arguments. I attach it here also. Please re-run this command with params and attach the log again.

## Troubleshooting

To show debug logs run this command in terminal (**Do not forget specify numpad layout and which config do you want to use**):

```bash
# LOG=DEBUG sudo -E ./asus_touchpad.py <REQUIRED:numpad layout file name without extension .py> <OPTIONAL:directory where is located config file with name: asus_touchpad_numpad_dev, by default is taken CWD - current working directory>

cd asus-touchpad-numpad-driver
LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" # now driver use root of repository as directory for config file named asus_touchpad_numpad_dev

cd asus-touchpad-numpad-driver
LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "/usr/share/asus_touchpad_numpad-driver/" # now driver use installed config
```

Gevert92, 2022-08-24:
Hello, i share the debug log printout:

LOG=DEBUG sudo -E ./asus_touchpad.py "g533" ""

`
DEBUG:Pad:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
DEBUG:Pad:Set keyboard 4 from H: Handlers=sysrq kbd event4 leds
DEBUG:Pad:Detecting touchpad from string: "N: Name="ELAN1405:00 04F3:315E Touchpad""
DEBUG:Pad:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ELAN1405:00/0018:04F3:315E.0001/input/input24
DEBUG:Pad:Set touchpad id 20 from H: Handlers=mouse1 event20
DEBUG:Pad:Touchpad min-max: x 0-3081, y 0-1627
DEBUG:Pad:Numpad min-max: x 200-2881, y 200-1547
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN1405:00 04F3:315E Touchpad
`

Image:
[image]

ldrahnik, 2022-08-24:
But in this log you not used NumberPad anytime. Can you please post relevant log with attempts for example numbers from 1 to 9 then special chars?

Gevert92, 2022-08-24:
It's true I missed that.

1) I did the test by pressing the slash (/) which does not type

[image]

2) I also noticed that if I press slash (/) quickly the touchpad shuts down and the log shows the following:
[image]

4) When I press 7 it still happens that I have to press it repeatedly for it to write, however the log shows the following whether or not the digit is painted.
[image]

5) Para el 8 y 9 ocurre lo mismo que el 7
[image]

ldrahnik, 2022-08-25:
1. But `/` is typed. Is displayed in log.
2. More information needed.
3. More information needed, is not possible that log is the same when is type and when not.
4. Looks like from log you touched 3 times  almost the same spot, for key 7, so nothing wrong.

Anyway, because I do not have your NumberPad type, `g533` I can not test if paddings are valid. Please check paddings in NumberPad layout and make pull request if not. Everything else should be working.

closed 2022-08-25 by ldrahnik

Gevert92, 2022-08-25:
Ok, I need to go deeper into the logic of the service to be able to give more information.

One question, when you say paddings what do you mean?

ldrahnik, 2022-08-25:
I recommend you read file Readme.md first. Padding = offset where is not possible touch any key and which is used for calculation widths and heights of rows and columns for key grid.

Gevert92, 2022-08-25:
Ok, I understand, I will go into analyzing what you mention.

Thanks for the clarification.