asus-linux-drivers/asus-numberpad-driver #59: [Bug]: Does not work key_repetitions with layout g533

Gevert92, 2022-08-24:
### Describe the bug

Hello, i have added `key_repetitions = 1` row in the layout g533.py but it does not work.

[image]

Previously it was working.

### Expected behavior

When leaving the key pressed, the value is written several times.

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2022-08-24:
Hello, thank you for creating an issue. You have to now change this value in config `/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev`. Readme is updated.

closed 2022-08-24 by ldrahnik

Gevert92, 2022-08-24:
Thank you.

I tried to change the asus_touchpad_numpad_dev file but it won't let me while the service is active, so I had to change the configuration directly in the asus_touchpad.py file.

ldrahnik, 2022-08-24:
I have no problem change config when is service running. Please do not change anything in code. You may be affected be found bug of overwriting config file, in this case please do a `git pull` to get latest state of repository.

Gevert92, 2022-08-24:
I found issue #57 where this issue is solved, thank you very much for the clarification, everything was solved with a git pull, I will be more pending.