asus-linux-drivers/asus-numberpad-driver #61: [Bug]:

AbubakrSamsodien, 2022-09-04:
### Describe the bug

installed, chose model up5401ea, but nothing happens, even after i rebooted my machine. 

log:
Testing interface i2c-0 : sucess
Detected laptop: ZenBook UX434IQ_UM433IQ
Could not automatically detect numpad layout for your laptop. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).

Select your model keypad layout:
1) b7402.py	3) g533.py	5) gx701.py	7) ux433fa.py	9) Quit
2) e210ma.py	4) gx551.py	6) up5401ea.py	8) ux581l.py
Please enter your choice 6 
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Asus touchpad service enabled
Asus touchpad suspend service enabled
Asus touchpad service started
Asus touchpad suspend service started

_OS: fedora 35_

### Expected behavior

when holding on the numpad icon on the touchpad, the numpad outline should light up.

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2022-09-04:
Hello, 
thank you for creating an issue. Could you please use the current latest version of repository and try again? (uninstall, `git pull`, install). In case it will does not work again please check `/var/log/asus_touchpad_numpad-driver/error.log` and check whether is service running using this command `systemctl status asus_touchpad_numpad` and try run driver manually from root of repository like this:

```
cd asus-touchpad-numpad-driver
LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" # now driver use root of repository as directory for config file named asus_touchpad_numpad_dev
```

and post result here.

ldrahnik, 2022-09-04:
Also, you have to hold the top right icon for at least 1s.

AbubakrSamsodien, 2022-09-04:
it works!, was missing numpy and libevdev python packages

closed 2022-09-04 by AbubakrSamsodien