asus-linux-drivers/asus-numberpad-driver #62: [Bug]: AUR installation doesnt work on Manjaro

Lamdarer, 2022-09-05:
### Describe the bug

Using a Asus VivoBook Flip 14 TM420UA and installing the `asus-touchpad-numpad-driver-up5401ea-git` package from the AUR results in an error saying that it couldn't be configured.

When installing via the GUI once it is finished a pop-up mentions that there was an issue with configuring it `Fehler beim Konfigurieren von asus-touchpad-numpad-driver-up5401ea-git` (I wasnt able to change the language sry for that), 
when I close it and view the details I notice that there is a warning about a link in the package which appears multiple times `==> WARNUNG: Paket enthält einen Verweis auf $pkgdir
usr/lib/systemd/system/asus_touchpad_numpad.service` 
and at the bottom there seems to be the important part:
```Job for asus_touchpad_numpad.service failed because the control process exited with error code.
Fehler beim Konfigurieren von asus-touchpad-numpad-driver-up5401ea-git
See "systemctl status asus_touchpad_numpad.service" and "journalctl -xeu asus_touchpad_numpad.service" for details.
```
and that the command couldn't be executed correctly `Fehler: asus-touchpad-numpad-driver-up5401ea-git: Befehl konnte nicht korrekt ausgeführt werden`

When running both of the mentioned debug commands for details I get this:
```
systemctl status asus_touchpad_numpad.service                                                        
× asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Mon 2022-09-05 18:34:02 CEST; 4min 27s ago
    Process: 18528 ExecStartPre=/bin/sleep 2 (code=exited, status=209/STDOUT)
        CPU: 1ms

Sep 05 18:34:02 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Scheduled restart job, restart counter is at 6.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Start request repeated too quickly.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
```

<details>
  <summary>journalctl -xeu asus_touchpad_numpad.service</summary>

```
journalctl -xeu asus_touchpad_numpad.service
Sep 05 18:34:01 PPC-VBA-22 systemd[1]: Starting Asus Touchpad to Numpad Handler...
░░ Subject: A start job for unit asus_touchpad_numpad.service has begun execution
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ A start job for unit asus_touchpad_numpad.service has begun execution.
░░ 
░░ The job identifier is 2858.
Sep 05 18:34:01 PPC-VBA-22 systemd[18496]: asus_touchpad_numpad.service: Failed to set up standard output: No such file or directory
Sep 05 18:34:01 PPC-VBA-22 systemd[18496]: asus_touchpad_numpad.service: Failed at step STDOUT spawning /bin/sleep: No such file or directory
░░ Subject: Process /bin/sleep could not be executed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ The process /bin/sleep could not be executed and failed.
░░ 
░░ The error number returned by this process is ERRNO.
Sep 05 18:34:01 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Control process exited, code=exited, status=209/STDOUT
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ An ExecStartPre= process belonging to unit asus_touchpad_numpad.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 209.
Sep 05 18:34:01 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ The unit asus_touchpad_numpad.service has entered the 'failed' state with result 'exit-code'.
Sep 05 18:34:01 PPC-VBA-22 systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
░░ Subject: A start job for unit asus_touchpad_numpad.service has failed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
[... 708 lines cut ...]
░░ The job identifier is 3422 and the job result is done.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Start request repeated too quickly.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ The unit asus_touchpad_numpad.service has entered the 'failed' state with result 'exit-code'.
Sep 05 18:34:02 PPC-VBA-22 systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
░░ Subject: A start job for unit asus_touchpad_numpad.service has failed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ A start job for unit asus_touchpad_numpad.service has finished with a failure.
░░ 
░░ The job identifier is 3422 and the job result is failed.
```

### Desktop

- Program version: [r304.c7452b8-1
](https://aur.archlinux.org/packages/asus-touchpad-numpad-driver-up5401ea-git)
- Distro: Manjaro Linux x86_64
- Kernel: 5.19.1-3-MANJARO
- DE: GNOME 42.4

ldrahnik, 2022-09-05:
@kamack38 Can you look at that please?

ldrahnik, 2022-09-08:
@kamack38 any progres?

kamack38, 2022-09-08:
I'm working on it.

kamack38, 2022-09-08:
@Lamdarer are you using wayland?

Lamdarer, 2022-09-08:
@kamack38 no I am Starting Gnome under x.org and `echo $XDG_SESSION_TYPE` gives me `x11` as output.
Though I have enabled the "Wayland session" in the gnome-layout-switcher (manjaro application) I will try disabling it though I am not sure if it even has any effect when Gnome is started in x.org mode, since I way able to use the manual installed one after I switched from default to x.org Gnome. 
sry until now I hadn't much time to do other testing but I will do that now. Thank you very much for the help!

Lamdarer, 2022-09-08:
I disabled the wayland session in the gnome-layout-switcher (even though it seems to have no effect).
Now it works, also no matter if I activate it ok not, sometimes its just really hard to disable the numpad.
Just to be sure that it wasnt me not beeing able to click it I run 
`systemctl status asus_touchpad_numpad.service` and it indeed seems to be active this time around:
```
● asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; preset: disabled)
     Active: active (running) since Thu 2022-09-08 17:48:26 CEST; 7min ago
   Main PID: 18393 (python3)
      Tasks: 5 (limit: 8781)
     Memory: 58.5M
        CPU: 10.165s
     CGroup: /system.slice/asus_touchpad_numpad.service
             └─18393 python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea /usr/share/asus_touchpad_numpad-driver/

Sep 08 17:48:24 PC-xx-22 systemd[1]: Starting Asus Touchpad to Numpad Handler...
Sep 08 17:48:26 PC-xx-22 systemd[1]: Started Asus Touchpad to Numpad Handler.
```
Also here the other output
of `journalctl -xeu asus_touchpad_numpad.service`
```
Sep 08 17:48:24 PPC-VBA-22 systemd[1]: Starting Asus Touchpad to Numpad Handler...
░░ Subject: A start job for unit asus_touchpad_numpad.service has begun execution
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ A start job for unit asus_touchpad_numpad.service has begun execution.
░░ 
░░ The job identifier is 2463.
Sep 08 17:48:26 PC-xx-22 systemd[1]: Started Asus Touchpad to Numpad Handler.
░░ Subject: A start job for unit asus_touchpad_numpad.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
░░ A start job for unit asus_touchpad_numpad.service has finished successfully.
░░ 
░░ The job identifier is 2463.
```
Not really sure what I did to get a different result, I already reinstalled the AUR package multiple times, otherwise I tried the manual install iirc once again before, but besides that and since the session should also be alright I really dont know.

I will make a timeshift snapshot now, reboot, check if its still working and then try to install some of the other packages, please tell me if I can provide you any other log files etc if necessary to figure out the cause.

Lamdarer, 2022-09-08:
Unfortunately now I am reliably getting a similar issue to [this](https://github.com/asus-linux-drivers/asus-fliplock-driver/issues/4), at first every second boot or so I just got a blank screen after login, otherwise it seemed fine, I tried to disable and re enable the "Wayland session" option  within the gnome-layout-switcher but I couldn't make out a pattern or proof if it has an effect(still starting a usual Gnome x.org session all the time).
Than after a while and some restarts, backups not even the login screen showed up, except for one time when I got a login screen but neither the touchpad nor touchscreen worked I launched both of the diagnostic commands but it gave the same response as above so it should have been working.
I timeshifted back to snapshot where just the screen turned black after login but either the timeshift dint work or the issue was present much earlier, but I didnt even came close to the login screen.
So here I am now at my backup pre AUR package install.
I will try to install the AUR driver again and see if I can reproduce the issue.
Else maybe you can tell me where to search for any leftover config files of the AUR or manual install so that I can remove them before reinstallation(in case they mess something up).

Lamdarer, 2022-09-09:
So I wiped my system, and I get basically the same issue as in my initial comment. (with tue AUR install)
Except that I am asked to restart instead of the failure pop-up.

ldrahnik, 2022-09-11:
I removed `/bin/sleep` by default because was displayed an error inside the log in the first message.

```
░░ The job identifier is 2858.
Sep 05 18:34:01 PPC-VBA-22 systemd[18496]: asus_touchpad_numpad.service: Failed to set up standard output: No such file or directory
Sep 05 18:34:01 PPC-VBA-22 systemd[18496]: asus_touchpad_numpad.service: Failed at step STDOUT spawning /bin/sleep: No such file or directory
░░ Subject: Process /bin/sleep could not be executed
░░ Defined-By: systemd
░░ Support: https://forum.manjaro.org/c/support
░░ 
```

Also I removed dependency `xinput` which is optional when is used wayland and can cause problems (but should not).

ldrahnik, 2022-09-11:
@kamack38 Warning should not be there, look at the example packages and fix that please. ```WARNING: Package contains a reference to $pkgdir usr/lib/systemd/system/asus_touchpad_numpad.service```

ldrahnik, 2022-09-11:
@Lamdarer `Pull master branch`, please. Could you install the driver via git and summarize the current state? With attached logs which prove how is not working state (if is) related to this driver?

ldrahnik, 2022-09-11:
@Lamdarer Ok, can you please post here logs from systemd serices when is laptop started and service is loaded but you can not get NumberPad on? Also please attach NumberPad own logs from `/var/log/asus_touchpad_numpad-driver/error.log`

ldrahnik, 2022-09-11:
@Lamdarer Now when you download the latest version of driver (`git pull`) you do not have to manually pick up `up5401ea`. I added your model too.

ldrahnik, 2022-09-11:
Is weird that getting touchpad status via `xinput` fails (log: `INFO:Pad:Getting Device Enabled via xinput failed`). Can you manually try check your touchpad status (enabled/disabled) via terminal please? This command is for mine: ```xinput --list-props "ASUE140D:00 04F3:31B9 Touchpad" | grep "Device Enabled"```

ldrahnik, 2022-09-11:
- fixed discovered OSError, added exit(1) so systemd driver service will try restart yourself after 5s
- 
```
INFO:Pad:Saving config file: "/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev"
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 1250, in <module>
    listen_touchpad_events()
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 1028, in listen_touchpad_events
    for e in d_t.events():
  File "/usr/lib/python3.10/site-packages/libevdev/device.py", line 536, in events
    ev = self._libevdev.next_event(flags)
  File "/usr/lib/python3.10/site-packages/libevdev/_clib.py", line 893, in next_event
    raise OSError(-rc, os.strerror(-rc))
```

ldrahnik, 2022-09-11:
@Lamdarer reverted back as not needed when is detected during running installation process `wayland` and installed when is detected `X11`

ldrahnik, 2022-09-11:
@Lamdarer Without `xinput` installed does not work this:

```
Disabling Touchpad (e.g. Fn+special key) disables by default NumberPad as well (can be disabled, this functionality supports atm only xinput from xorg, no wayland support https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/54)
```

ldrahnik, 2022-09-11:
Only when problems occur again. I hope main problem was `/bin/sleep` and `OSError` without except and auto restarting driver service.

ldrahnik, 2022-09-11:
```
Also one issue I noticed is that when using the swiping gesture(Top-left to button right) it brings up the calculator, but when I do it again it brings up another calculator instead of closing the first one(enabling/disabling of the numpad works fine though).
```

It's up to you (I can not do it in driver as Windows developers can) setting own script to `KEY_CALC` key which will have toggling functionality. In Readme I linked mine toggling script bound to `KEY_CALC` for elementary OS.

ldrahnik, 2022-09-11:
@Lamdarer @kamack38 What is left I think is fix warning below and then can be issue closed? (can be reopened if any problems occur)

@kamack38 Warning should not be there, look at the example packages and fix that please. WARNING: Package contains a reference to $pkgdir usr/lib/systemd/system/asus_touchpad_numpad.service

ldrahnik, 2022-09-11:
In elementary I am setting it via `dconf` like this:
[image]

And then in settings > keyboard > custom shortcuts I see:

[image]

ldrahnik, 2022-09-11:
@Lamdarer Now it should be fixed. Thank you. It was a bug because I do not remove comments `#` from each line during reading file `laptop_numpad_layouts`.

closed 2022-09-12 by ldrahnik