asus-linux-drivers/asus-numberpad-driver #63: [Feature Request] Better auto-detection

kamack38, 2022-09-08:
### Short Description

By using [this site](https://linux-hardware.org/), we can easily detect the layout based on the Touchpad serial number.

### New dependencies

none

### Additional Info

Example:
By looking at multiple probes, I realized that we can associate `g533` layout with `ASUE1403:00 04F3:Touchpad`.

kamack38, 2022-09-11:
As far as I'm concerned, only one layout is associated to one touchpad ID, but one touchpad model exists in multiple variations of devices. For example: `ASUE1403:00 04F3:Touchpad` exists in `G513RM`, `G533QS`, `G533QZ` and `G533ZW`

kamack38, 2022-09-11:
In every of this devices the ID detected by xinput is `ASUE1403:00 04F3:Touchpad`

ldrahnik, 2022-09-11:
Understand, I'am concerned too. Part of `ASUE1403:00 04F3:Touchpad` is product id: `04F3` and NumberPad is included inside touchpad so it does make sense. 

How can be used [site](https://linux-hardware.org/) for improve autodetection? Is possible somehow parse from that website layout with input Touchpad product id? If yes, can you please post here example?

kamack38, 2022-09-11:
This site can be used for creating new layouts.

Steps:
1. Get laptop model name to be associated with the new layout.
2. Use the site to get the Touchpad ID.
3. Create new layout.
4. Associate new layout with the Touchpad ID. - this means that even if user with different laptop model uses the same Touchpad it will detect it automatically.

I would replace the model list with the Touchpad ID list.

ldrahnik, 2022-09-12:
@kamack38 About only one product ID for one layout type: From mentioned issue and table you can see that is not true. More product ids can have the same NumberPad layout (e.g. from the table `0x3145` and `0x31b9` is both layout `up5401ea.py` and `0x30dc`, `0x30F2`, `0x30F4` are `ux433fa.py`.

ldrahnik, 2022-09-12:
@kamack38 But the idea is not bad, I like it and I agree with refactoring laptop model to part of touchpad ID named product ID. I will post here a comment if any work from my side will be done. I had no time yet unfortunately.

kamack38, 2022-09-12:
OK, I've made a mistake, because I was looking at the device name instead of ID. I should have been looking at the `ID/Class` position. And could you please add a link to the *mentioned issue*.

ldrahnik, 2022-11-29:
@kamack38 Would be possible do something like this:

```
1. Based on each touchpad ID sent a get request by e.g. curl `https://linux-hardware.org/?view=search&vendorid=04f3&deviceid=31b9&typeid=input%2Fkeyboard#list` and parse e.g. by `xmllint` from HTML all laptop names and save them to list
2. In case list contains the laptop which is already written (associated to NumberPad layout) in static offline file `laptop_numpad_layouts` to NumberPad) can be read and used associated NumberPad layout

WIP parsing obtained HTML's via xmllint part:

```
xmllint --html --xpath '//td[@class="device"]//a[1]/@href' <<< cat ./Input_keyboard\ devices.html | cut -f2 -d"\""
xmllint --html --xpath '//table/tbody/tr/td[3]/span/@title' <<< cat ./ASUE140D_00\ 04F3_31B9\ Keyboard.html | cut -f2 -d"\""
```

Do you want to try to implement this (your) idea to `install.sh` script by yourself?

kamack38, 2022-11-29:
I think better idea would be to store all data locally in some file and then based on this we would assign the correct layout

kamack38, 2022-11-29:
But, yes It would be possible to do something like that. IMO `htmlq` (I think it was called something like it) is the best HTML parser.

ldrahnik, 2022-11-29:
@kamack38 Problem with your idea is maintenance, my idea has the advantage that all the latest Asus laptops uploaded to `linux-hardware` when use already known "old" NumberPad layout would be automatically recommended during install process (running `install.sh`) = immediately when would be uploaded laptop info to **linux-hardware website**. No maintenance is needed. But what is required is a connection. I still trust it is the best solution. Because when driver's user does not have a connection there still exists a possibility to select layout manually. What I see as the second (a combination of both approaches) is routine - updating file in the repository by simple script? github hook? and periodically (1 per day?) update this file in the repository with new laptops and with them associated layouts. Do not know if it is possible, I have experience only with building/deploying hooks in GitLab.

ldrahnik, 2023-01-21:
@kamack38 ping

kamack38, 2023-01-21:
Looks good!

linuxhw, 2023-03-11:
Hello

Info for ASUE1403 can be quickly found with the help of auxiliary repo https://github.com/linuxhw/Dmesg

Clone it and grep what you want. Result for ASUE1403 is the following:

```
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:942:[    1.404147] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input9
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:943:[    1.404228] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input10
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:944:[    1.404289] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input11
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:945:[    1.404408] hid-generic 0018:04F3:319A.0002: input,hidraw1: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:954:[    1.595630] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input12
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:955:[    1.694347] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input13
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:956:[    1.711169] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input14
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:957:[    1.728191] hid-multitouch 0018:04F3:319A.0002: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:957:[    0.880093] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input5
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:958:[    0.880306] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input6
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:959:[    0.880413] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input7
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:960:[    0.880491] hid-generic 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1227:[    4.333701] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input29
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1229:[    4.333768] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input30
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1230:[    4.333876] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input31
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1231:[    4.334129] hid-multitouch 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1029:[    1.858668] input: ASUE1403:00 04F3:319A Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input4
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1030:[    1.858841] input: ASUE1403:00 04F3:319A Touchpad as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input5
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1031:[    1.858984] input: ASUE1403:00 04F3:319A Keyboard as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input6
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1032:[    1.859052] hid-generic 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
[... cut ...]
./ASUSTek Computer/ROG/ROG Strix G533ZW_G533ZW/560A787C4010/UBUNTU-22.04/5.19.0-051900-GENERIC/X86_64/FFAB2B901D:1445:[    5.509800] input: ASUE1403:00 04F3:319A Keyboard as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input32
./ASUSTek Computer/ROG/ROG Strix G533ZW_G533ZW/560A787C4010/UBUNTU-22.04/5.19.0-051900-GENERIC/X86_64/FFAB2B901D:1448:[    5.510173] hid-multitouch 0018:04F3:319A.0001: input,hidraw7: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
```

ldrahnik, 2023-03-11:
@linuxhw 

Hello there, 
thanks a lot for letting me know!! :)

About the recommendation of the correct NumberPad layout for users of this Asus driver:

Negative: 
- has to be downloaded the entire set of all laptops to the user who wants to install the driver (to his local system, **EDIT:** 46% downloaded and size is 2.18GB already) and the NumberPad layout is not found in the static file
Q: I can not grep in GitHub repository without downloading, I am right?
- set might have a delay
Q: Is the update planned every day, week, month, or non periodically?

Benefits:
- the structure of repository dMesg looks solid, via curl can be implemented code easily broken because the code uses HTML tags/ids/classes of website https://linux-hardware.org/ (currently used implementation starts in code line here: https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/install.sh#L99)
- I can now (when will be the entire repository downloaded) more easily extend the existing static file with associated NumberPad layouts to laptop names via the same Device ID's using script. Manual work is required only when a new Touchpad device's ID which contains NumberPad has been released by Asus company for any Asus laptops, then has to be manually created NumberPad layout in the driver and added to static file with associated newly created layout.

So I appreciate a lot created repository as Linux kernel and kernel modules contain a lot of hardcoded laptop types etc. And created repository as I mentioned above helps me extend offline file which should help for correct driver settings automatically during installation.

But in summary of both negatives & props IMHO would be great to add also REST API as possibility have up-to-date-data as a second option for people with connection without any intervention of maintainers of drivers like me who create static files and preset device drivers for different laptops. I would even vote for having on https://linux-hardware.org/ additional information as NumberPad layout in my case. This can be checked only visually by eys AFAIK.

Best regards

Lukas

ldrahnik, 2023-03-11:
- Write a script which extends existing static file with associated data for auto-detection via [linuxhw/Dmesg](https://github.com/linuxhw/Dmesg)

ldrahnik, 2023-12-17:
Results from new attempt of check whether is migration to association NumberPad template by Touchpad product id:

- Find all unique Asus Touchpad product ids (with but even without NumberPad):

```
$ grep -hRP "(^|\s)(ELAN|ASUE).*04F3:.*Touchpad(?=\s|$)" Convertible/ASUSTek\ Computer/ Notebook/ASUSTek\ Computer/ | cut -b 23-43 | sort -u > recommendation-based-on-dmesg-repo-17-12-2023-of-all-asus-unique-product-touchpad-ids.txt
```

and then one by one get all laptops for each touchpad product id:

```
$ grep -hRP "(^|\s)(ELAN|ASUE).*04F3:.*Touchpad(?=\s|$)" Convertible/ASUSTek\ Computer/ Notebook/ASUSTek\ Computer/ | cut -b 23-43 | sort -u
```

and result is (WIP):

```
ASUE1200:00 04F3:3087 - not found any laptop with numberpad
ASUE1200:00 04F3:30F7 - not found any laptop with numberpad
ASUE1200:00 04F3:3132 - not found any laptop with numberpad
ASUE1201:00 04F3:3125 - e210ma (found here https://www.elasslihitech.com/laptop-occasion/asus-vivobook-x421/)
ASUE1201:00 04F3:3148 - 
ASUE1209:00 04F3:319F
ASUE120A:00 04F3:319B
ASUE120B:00 04F3:31C0
ASUE120B:00 04F3:3257
ASUE120C:00 04F3:31C1
ASUE120D:00 04F3:31FB
ASUE1210:00 04F3:31FA
ASUE1211:00 04F3:3211
ASUE1212:00 04F3:3233
ASUE1300:00 04F3:3104
ASUE1301:00 04F3:3128
ASUE1304:00 04F3:3201
ASUE1305:00 04F3:3212
ASUE1307:00 04F3:3279
ASUE1403:00 04F3:319A
ASUE1406:00 04F3:3101
ASUE1407:00 04F3:310D
ASUE1408:00 04F3:3137
ASUE1409:00 04F3:3157
ASUE140A:00 04F3:3134
ASUE140C:00 04F3:3145
ASUE140D:00 04F3:31B9
ASUE140D:00 04F3:31BC
ASUE140F:00 04F3:31F7
ASUE140F:00 04F3:32A2
ASUE1410:00 04F3:322D
ASUE1900:00 04F3:31AF
ASUE1A00:00 04F3:31DE
ASUE1A01:00 04F3:31D4
ELAN1200:00 04F3:301A
ELAN1200:00 04F3:3022
ELAN1200:00 04F3:303C
ELAN1200:00 04F3:303E
ELAN1200:00 04F3:3044
ELAN1200:00 04F3:3045
ELAN1200:00 04F3:3049
ELAN1200:00 04F3:304A
ELAN1200:00 04F3:304E
ELAN1200:00 04F3:3054
ELAN1200:00 04F3:3058
ELAN1200:00 04F3:3059
ELAN1200:00 04F3:3066
ELAN1200:00 04F3:3067
ELAN1200:00 04F3:306F
ELAN1200:00 04F3:307A
ELAN1200:00 04F3:3086
ELAN1200:00 04F3:3087
ELAN1200:00 04F3:3090
ELAN1200:00 04F3:309C
ELAN1200:00 04F3:309F
ELAN1200:00 04F3:30B9
ELAN1200:00 04F3:30BA
ELAN1200:00 04F3:30C0
ELAN1200:00 04F3:30C6
ELAN1200:00 04F3:30CE
ELAN1200:00 04F3:30D4
ELAN1200:00 04F3:30E9
ELAN1200:00 04F3:30EF
ELAN1200:00 04F3:30F7
ELAN1200:00 04F3:3104
ELAN1200:00 04F3:3107
ELAN1200:00 04F3:3168
ELAN1201:00 04F3:301B
ELAN1201:00 04F3:3054
ELAN1201:00 04F3:3098
ELAN1201:00 04F3:30B8
ELAN1201:00 04F3:30C6
ELAN1201:00 04F3:30CE
ELAN1201:00 04F3:313F
ELAN1201:01 04F3:30CE
ELAN1203:00 04F3:301E
ELAN1203:00 04F3:3043
ELAN1203:00 04F3:304E
ELAN1203:00 04F3:307A
ELAN1204:00 04F3:30B2
ELAN1205:00 04F3:30E9
ELAN1206:00 04F3:30F1
ELAN1207:00 04F3:310E
ELAN1300:00 04F3:3028
ELAN1300:00 04F3:3032
ELAN1300:00 04F3:3057
ELAN1300:00 04F3:3059
ELAN1300:00 04F3:3087
ELAN1300:00 04F3:30BE
ELAN1300:00 04F3:3104
ELAN1300:00 04F3:310C
ELAN1301:00 04F3:3034
ELAN1301:00 04F3:3035
ELAN1301:00 04F3:30C6
ELAN1301:00 04F3:30EF
ELAN1301:00 04F3:3115
ELAN1301:00 04F3:3128
ELAN1400:00 04F3:30AD
ELAN1400:00 04F3:30E2
ELAN1400:00 04F3:30E5
ELAN1401:00 04F3:30BB
ELAN1401:00 04F3:30DC
ELAN1401:00 04F3:30F2
ELAN1401:00 04F3:30F4
ELAN1402:00 04F3:30BF
ELAN1404:00 04F3:3048
ELAN1404:00 04F3:30A7
ELAN1404:00 04F3:30C2
ELAN1405:00 04F3:30DF
ELAN1405:00 04F3:315E
ELAN1406:00 04F3:3101
ELAN2301:00 04F3:309F
ELAN9009:00 04F3:2C1B
ELAN9009:00 04F3:2C23
ELAN9009:00 04F3:2F2A
ELAN9009:00 04F3:2F32
ELAN9009:00 04F3:4068
ELAN9009:00 04F3:41D9
```

ldrahnik, 2023-12-18:
All unique touchpad product ids only (no ASUE/ELAN):

```
$ grep -hRP "(^|\s)(ELAN|ASUE).*04F3:.*Touchpad(?=\s|$)" Convertible/ASUSTek\ Computer/ Notebook/ASUSTek\ Computer/ | cut -b 36-43 | sort -u
4F3:2C1B
4F3:2C23
4F3:2F2A
4F3:2F32
4F3:301A
4F3:301B
4F3:301E
4F3:3022
4F3:3028
4F3:3032
4F3:3034
4F3:3035
4F3:303C
4F3:303E
4F3:3043
4F3:3044
4F3:3045
4F3:3048
4F3:3049
4F3:304A
4F3:304E
4F3:3054
4F3:3057
4F3:3058
4F3:3059
4F3:3066
4F3:3067
4F3:306F
4F3:307A
4F3:3086
4F3:3087
4F3:3090
4F3:3098
4F3:309C
4F3:309F
4F3:30A7
4F3:30AD
4F3:30B2
4F3:30B8
4F3:30B9
4F3:30BA
4F3:30BB
4F3:30BE
4F3:30BF
4F3:30C0
4F3:30C2
4F3:30C6
4F3:30CE
4F3:30D4
4F3:30DC
4F3:30DF
4F3:30E2
4F3:30E5
4F3:30E9
4F3:30EF
4F3:30F1
4F3:30F2
4F3:30F4
4F3:30F7
4F3:3101
4F3:3104
4F3:3107
4F3:310C
4F3:310D
4F3:310E
4F3:3115
4F3:3125
4F3:3128
4F3:3132
4F3:3134
4F3:3137
4F3:313F
4F3:3145
4F3:3148
4F3:3157
4F3:315E
4F3:3168
4F3:319A
4F3:319B
4F3:319F
4F3:31AF
4F3:31B9
4F3:31BC
4F3:31C0
4F3:31C1
4F3:31D4
4F3:31DE
4F3:31F7
4F3:31FA
4F3:31FB
4F3:3201
4F3:3211
4F3:3212
4F3:322D
4F3:3233
4F3:3257
4F3:3279
4F3:32A2
4F3:4068
4F3:41D9

$ wc -l 
100
```

ldrahnik, 2023-12-20:
Commands updated about `ASUP` vendor id `093A`

ldrahnik, 2023-12-21:
Note: Online using linux-hardware would have advantage even if will be reworked recommendations to touchpad product id = layout, even like github bot with auto creating issues in repository, hey - probe of new touchpad product id has been uploaded and looks like it may contains NumberPad, Is necessarry manually identify layout (existing or create new type) and add to repository

closed 2024-03-23 by ldrahnik