asus-linux-drivers/asus-numberpad-driver #64: [Layout Request] Layout Name

eyallior, 2022-09-11:
### Layout Name

ROG Strix G513QR

### Your device (sudo dmidecode -s system-product-name)

ROG Strix G513QR

### Supported Devices

Asus ROG Strix G513QR

### Numpad Layout (photo/picture or code)

Asus zen
https://www.linuxuprising.com/ezoimgfmt/1.bp.blogspot.com/-PzCdrsuCY9o/YJ6dwazM4VI/AAAAAAAAFgw/IxlqYAqZSCQSMYO2ERXAtuVF-irEROQ7wCLcBGAsYHQ/s640/asus-zenbook.png?ezimgfmt=rs%3Adevice%2Frscb272-1

### Additional Info

Right now when I install - after restart I can't login (get kicked back to login screen again)

ldrahnik, 2022-09-11:
Hello, 

thank you for creating an issue. 

Can you please provide more information? Like for example systemd logs. Do you installed driver from git or via arch package? Do you use X11 or wayland?

ldrahnik, 2022-09-11:
Please try `git pull`. Might be fixed.

eyallior, 2022-09-12:
I use X11 (PopOS)
I've installed from git on 20 july
Not daring to install it again (I'll need to reboot in safe mode and I'm afraid I don't remember all that's needed to fix after it goes wrong)

ldrahnik, 2022-09-12:
I am 99 % sure blank screen you can not get from installing this driver. Prove me wrong.

ldrahnik, 2022-09-12:
Can be reopened if you provide more information associated with this driver.

closed 2022-09-12 by ldrahnik

eyallior, 2022-09-14:
Not a blank screen. When I enter password and click login, it jumps after 1 second back to login screen.

ldrahnik, 2022-09-14:
Edited. Be constructive, that means we need logs to see the association between your statement and this driver. You can use linux live USB (which is even better because the system is at that moment clean).