asus-linux-drivers/asus-numberpad-driver #66: [Bug]:

satva2, 2022-09-12:
### Describe the bug

Cannot get it to work on the g533, in fedora it doesnt have the option for this layout. i downloaded the python file for this layout, but couldnt figure it out ...

any ideas or help. basically newbie w/linux.

### Expected behavior

was hoping for numberpad to work

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

kamack38, 2022-09-12:
If you don't use arch you have to use the install script. Instructions are in the README.

ldrahnik, 2022-09-12:
@satva2 Hello, as @kamack38 recommends to you you should use `git clone` to get latest version of driver or download repository as archive from this place or on arch via package manager and then run in downloaded repository `install.sh` script with `sudo`. Carefully first read `Readme.MD` in every project, then try it and ask if something does not work or you do not understand.

Can be reopened.

closed 2022-09-12 by ldrahnik

satva2, 2022-09-12:
> <asus-linux-drivers/asus-touchpad-numpad-driver/issues/66/1244505011@
> github.com>

I did use it though... When options popup to pick a layout. The g533 layout
isn't there. But it is available on GitHub with the rest of them, but I'm
not sure what to do to get the install script to use it. I mean, I put it
in the Correct folder. I even tried changing the contents of one that was
available to the g533
...I just couldn't get the script to see that layout ..
..

ldrahnik, 2022-09-12:
Can you post what you see in terminal during installing via `install.sh`, copy paste terminal or do a screenshot.

satva2, 2022-09-12:
> <asus-linux-drivers/asus-touchpad-numpad-driver/issues/66/1244692634@

Just this. g533.py isn't in the layouts folder and if I put it in there
(from off GitHub) it will list it but it won't let me chose it. It says l
Iike "invalid option" or something.

satva2, 2022-09-12:
When I first installed, I picked one of the other layouts just... Because,
and the pad did light up but then after it shut off, I can't get it back on

ldrahnik, 2022-09-13:
@satva2 Can you post what you see in terminal during installing via install.sh, copy paste terminal or do a screenshot

satva2, 2022-09-14:
I posted the other day, and you asked for a screenshot. i was replying from email and it appears the photo i inserted (twice) didnt make its way through.. none the less i appreciate the help. heres a screenshot
[image]

ldrahnik, 2022-09-14:
@satva2 It looks like you do not have the latest version of this driver. Can you do a `git pull`?