asus-linux-drivers/asus-numberpad-driver #68: [Bug]: no g533 layout

satva2, 2022-09-14:
### Describe the bug

I posted the other day, and you asked for a screenshot. i was replying from email and it appears the photo i inserted (twice) didnt make its way through.. none the less i appreciate the help. heres a screenshot

also, bear with me. new here. i replied on the closed issue, but wasnt sure if youd see it so im reopening one.

[image]

### Expected behavior

cannot get numpad on g533 working.

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2022-09-14:
Duplication of https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/66

closed 2022-09-14 by ldrahnik

satva2, 2022-09-14:
ok, heres that

[image]

satva2, 2022-09-14:
when you say root of downloaded driver repository, i dont know exactly what you mean?

ldrahnik, 2022-09-14:
As last time you did it fine. You do not have this driver repository. You have this one https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver. You are welcome.

satva2, 2022-09-14:
are you sure? i did git clone https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver and still have issue. please bare with me tho man, i understand its frusterating working with someone who doesnt understand completely

ldrahnik, 2022-09-14:
As is in the picture `git pull`, please run the same way `git remote show origin`.

satva2, 2022-09-14:
ill figure it out.. thanks :)

ldrahnik, 2022-09-14:
For getting this drive repository to your laptop please run `git clone git@github.com:asus-linux-drivers/asus-touchpad-numpad-driver.git`.

ldrahnik, 2022-09-14:
@satva2 Confirmation for me that you had the wrong repository were sizes of files:

```
(base) ldrahnik@ZenbookUP5401EAUP5401EA5d04e179:~/projects/copy-driver$ git remote show origin
* remote origin
  Fetch URL: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver
  Push  URL: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver
  HEAD branch: main
  Remote branch:
    main tracked
  Local branch configured for 'git pull':
    main merges with remote main
  Local ref configured for 'git push':
    main pushes to main (up to date)
(base) ldrahnik@ZenbookUP5401EAUP5401EA5d04e179:~/projects/copy-driver$
(base) ldrahnik@ZenbookUP5401EAUP5401EA5d04e179:~/projects/copy-driver$ ls -la numpad_layouts
total 24
drwxrwxr-x 2 ldrahnik ldrahnik 4096 Sep 14 13:05 .
drwxrwxr-x 6 ldrahnik ldrahnik 4096 Sep 14 13:05 ..
-rw-rw-r-- 1 ldrahnik ldrahnik  619 Sep 14 13:05 gx701.py
-rw-rw-r-- 1 ldrahnik ldrahnik  539 Sep 14 13:05 m433ia.py
-rw-rw-r-- 1 ldrahnik ldrahnik  542 Sep 14 13:05 ux433fa.py
-rw-rw-r-- 1 ldrahnik ldrahnik  740 Sep 14 13:05 ux581l.py
```

satva2, 2022-09-14:
i know... i see that. here is what happens now.
[image]

satva2, 2022-09-14:
ill try the command you gave me. as i downloaded it manually this last time. and tried. thanks.

ldrahnik, 2022-09-14:
Np, using terminal is always win against clicking websites, a lot of mistakes happens to everyone because of that. One tab, another tab, the same driver name.. so driver is working fo you?

satva2, 2022-09-14:
well.. im imagining it will. but.... now i have this problem
[image]

satva2, 2022-09-14:
ugh im sorry. its early and im not focused.. as to what im doing

satva2, 2022-09-14:
well this is why i switched to linux. there is a challenge. ill let you know when i get it and thanks for making it available,

ldrahnik, 2022-09-14:
@sotva2 For using `git clone <ssh_url>` you need to have your public key uploaded to your github profile. Search google with something like `add public key to github profile`.

satva2, 2022-09-14:
i got it. but this is what it says now... numpad isnt lighting up. 
[image]

ldrahnik, 2022-09-14:
@satva2 Post result of command `sudo systemctl status asus_touchpad_numpad.service`.

satva2, 2022-09-14:
[image]

ldrahnik, 2022-09-14:
@satva2 And result of status command after you run `sudo systemctl start asus_touchpad_numpad`?

satva2, 2022-09-14:
is this what you meant? 

[image]

ldrahnik, 2022-09-14:
Ok, post here please entire content of this file (drivers log file): `/var/log/asus_touchpad_numpad-driver/error.log`

satva2, 2022-09-14:
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

ldrahnik, 2022-09-14:
@satva2 Numpy python package is missing. What is output of `sudo dnf -y install python3-numpy`? Try run `sudo dnf -y install python3-numpy`.

ldrahnik, 2022-09-14:
@satva2 Then try `sudo install.sh` again.

satva2, 2022-09-14:
[image]

satva2, 2022-09-14:
really appreciate the time and effort man

ldrahnik, 2022-09-14:
@satva2 Post log file again.

ldrahnik, 2022-09-14:
@satva2 Feel free to put a star to this driver after it will be running.

ldrahnik, 2022-09-14:
@satva2 Try run `pip3 install numpy` and then restart, then status, then watch log if does not running yet.

satva2, 2022-09-14:
this is what it says... 
[image]

satva2, 2022-09-14:
should i jiust got ahead with the other stuff you said to do ? (restart driver...)

ldrahnik, 2022-09-14:
No. Instead of `pip3` try `python3 -v` and post here your python3 version from which you can exit using writing `exit()` and then try `python3 -m pip install numpy`.

satva2, 2022-09-14:
[image]

ldrahnik, 2022-09-14:
Weird. Can you now try to run driver but via this: `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" ""` and post here result.

satva2, 2022-09-14:
says command not found
 
[image]

satva2, 2022-09-14:
oh oopos
whoa,. i apoligize

satva2, 2022-09-14:
[image]

ldrahnik, 2022-09-14:
@satva2 Do a `git pull` and then again run previous command for debugging driver but with your layout of course `LOG=DEBUG sudo -E ./asus_touchpad.py "g533" ""`

satva2, 2022-09-14:
how do i find my exact layout?

ldrahnik, 2022-09-14:
@satva2 `g533` should fit your laptop, during install.sh is detected automatically, during debugging you have to use name of file from folder `numpad_layouts`. So run `sudo -E ./asus_touchpad.py "g533" ""`.

satva2, 2022-09-14:
it keeps doing this, after git pull

[image]

satva2, 2022-09-14:
installing but is not moving at all, so maybe problem between here and server or something

satva2, 2022-09-14:
probably have to pick this up later im assuming..... it just says downloading packages,,, and remains on zero,

satva2, 2022-09-14:
nevermind it completed,

ldrahnik, 2022-09-14:
@satva2 When will be installed try to run again `sudo -E ./asus_touchpad.py "g533" ""`

satva2, 2022-09-14:
[image]

it just jeeos repeating saving config file

ldrahnik, 2022-09-14:
@satva2 It's ok. Try open now NumberPad so hold top right icon for at least 1s, in this debugging mode.

satva2, 2022-09-14:
its working

ldrahnik, 2022-09-14:
@satva2 So exit this debugging driver and again `sudo systemctl start asus_touchpad_numpad` and then  `sudo systemctl status asus_touchpad_numpad` and it should be ok.

satva2, 2022-09-14:
hey man youre a godsend. have a good day. :)

ldrahnik, 2022-09-14:
@satva2 You can customize the driver via changes in file `numpad_layout/g533.py` (if is something missing or wrong, please make pull request or create issue here again, others use this layout also) in the repository and then remember to run `install.sh` again and other values not depending on your NumberPad model you can customize via config file `/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev`. Customization is described in the table in `Readme.MD` file.

satva2, 2022-09-14:
ok. cool thanks. thumbs up