asus-linux-drivers/asus-numberpad-driver #69: [Bug]:

bnisevic, 2022-12-09:
### Describe the bug

Numpad does not switch on by tapping numpad icon on the touchpad.

### Expected behavior

Numpad should be LED illuminated and working after tapping numpad icon.

### Desktop

- I am on UX582L
- Program version: latest downloaded by suggested git clone command.
- Distro: Ubuntu 20.04
- Kernel: 5.15.0-56-generic

ldrahnik, 2022-12-09:
Hello,
I greatly appreciate your issue because this vertical NumberPad layout was taken from the previous repository and has not been fixed yet. I pushed an attempt how could be layout fixed but I do not have your laptop, I can not test it, could you please play with offset & icon values inside file `numpad_layouts/ux581l.py` and figure it out? Then create comment/pull request. Thank you a lot.

// re-clone/pull the latest master and re-install the driver to reflect my changes what I did now and could help

ldrahnik, 2022-12-11:
Closed for issue owner inactivity, can be reopened.

closed 2022-12-11 by ldrahnik