asus-linux-drivers/asus-numberpad-driver #71: Numpad working but no backlight

EscarM1, 2022-12-26:
### Describe the bug

Installed asus-touchpad-numpad-driver v1.7.0 by downloading it from release page and installed with '> sudo ./install.sh' (no change done in the files). Installation went fine, NumberPad layout 'b7402' was identified correctly. In the end, the NumberPad works but not the backlight: it can be activated/deactivated via holding the top right icon on the touchpad but remains invisible. Computer restart does not help.

Is a further configuration step necessary or is it an issue with this laptop model? Not sure what additional (log) info is needed. Please ask!

### Expected behavior

NumberPad backlight should switch on when activated.

### Desktop

- Program version: 1.7.0
- Distro: Ubuntu 22.04.1 LTS
- Kernel: 5.15.0-56
- Computer: Asus ExpertBook B9 B9400

ldrahnik, 2022-12-26:
Hello there,
backlight can be tested (in the same way how driver use set up that) in command line:

Enable backlight: (only one change is hex value `0x01`)

```
sudo i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
```

Disable backlight: (only one change is hex value `0x00`)
```
sudo i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad r13
```

Usage of hex value between `0x41-46` should result in different brightness.

EscarM1, 2022-12-27:
Hello @ldrahnik,
Thanks a lot for the quick reply. I ran again `sudo install.sh` install script and after that the backlight started working all right. The install script output is

```
# ./install.sh
apt is here
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
libevdev2 est déjà la version la plus récente (1.12.1+dfsg-1).
i2c-tools est déjà la version la plus récente (4.3-2build1).
python3-pip est déjà la version la plus récente (22.0.2+dfsg-1).
git est déjà la version la plus récente (1:2.34.1-1ubuntu1.5).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  chromium-codecs-ffmpeg-extra gstreamer1.0-vaapi i965-va-driver intel-media-va-driver
  libaacs0 libaom3 libass9 libavcodec58 libavformat58 libavutil56 libbdplus0 libblas3
  libbluray2 libbs2b0 libchromaprint1 libcodec2-1.0 libdav1d5 libflashrom1 libflite1
  libftdi1-2 libgme0 libgsm1 libgstreamer-plugins-bad1.0-0 libigdgmm12 liblilv-0-0 libmfx1
  libmysofa1 libnorm1 libopenmpt0 libpgm-5.3-0 libpostproc55 librabbitmq4 librubberband2
  libserd-0-0 libshine3 libsnappy1v5 libsord-0-0 libsratom-0-0 libsrt1.4-gnutls
  libssh-gcrypt-4 libswresample3 libswscale5 libudfread0 libva-drm2 libva-wayland2
  libva-x11-2 libva2 libvdpau1 libvidstab1.1 libx265-199 libxvidcore4 libzimg2 libzmq5
  libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers pocketsphinx-en-us va-driver-all
  vdpau-driver-all
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
xinput est déjà la version la plus récente (1.6.3-1build2).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  chromium-codecs-ffmpeg-extra gstreamer1.0-vaapi i965-va-driver intel-media-va-driver
  libaacs0 libaom3 libass9 libavcodec58 libavformat58 libavutil56 libbdplus0 libblas3
  libbluray2 libbs2b0 libchromaprint1 libcodec2-1.0 libdav1d5 libflashrom1 libflite1
  libftdi1-2 libgme0 libgsm1 libgstreamer-plugins-bad1.0-0 libigdgmm12 liblilv-0-0 libmfx1
  libmysofa1 libnorm1 libopenmpt0 libpgm-5.3-0 libpostproc55 librabbitmq4 librubberband2
  libserd-0-0 libshine3 libsnappy1v5 libsord-0-0 libsratom-0-0 libsrt1.4-gnutls
  libssh-gcrypt-4 libswresample3 libswscale5 libudfread0 libva-drm2 libva-wayland2
  libva-x11-2 libva2 libvdpau1 libvidstab1.1 libx265-199 libxvidcore4 libzimg2 libzmq5
  libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers pocketsphinx-en-us va-driver-all
  vdpau-driver-all
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
Requirement already satisfied: libevdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 2)) (1.6.0)
Requirement already satisfied: numpy in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 3)) (1.24.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ASUS EXPERTBOOK B9400CBA_B9400CBA
Automatically recommended numpad layout: b7402 (associated to B9400CBA-KC0074X). You can specify numpad layout later by yourself and please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Is recommended numpad layout correct? [y/N]y
Selected key layout b7402
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Asus touchpad service enabled
Asus touchpad suspend service enabled
Asus touchpad service started
Asus touchpad suspend service started
```
Unfortunately I don't have the output of the first install any more so can't compare and also not sure why it did not work the first time.

EscarM1, 2022-12-27:
Hello again,

Little addon: after a few restarts of the system, the backlight is not working any more. So this seems to be a rather unstable thing. I tried the following commands if this is any useful:

```
# i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	i915 gmbus dpa                  	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-6	i2c       	i915 gmbus tc1                  	I2C adapter
i2c-7	i2c       	i915 gmbus tc2                  	I2C adapter
i2c-8	i2c       	i915 gmbus tc3                  	I2C adapter
i2c-9	i2c       	i915 gmbus tc4                  	I2C adapter
i2c-10	i2c       	i915 gmbus tc5                  	I2C adapter
i2c-11	i2c       	i915 gmbus tc6                  	I2C adapter
i2c-12	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-13	i2c       	AUX A/DDI A/PHY A               	I2C adapter
i2c-14	i2c       	AUX USBC1/DDI TC1/PHY TC1       	I2C adapter
i2c-15	i2c       	AUX USBC2/DDI TC2/PHY TC2       	I2C adapter
```

```
# i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
msg 0: addr 0x15, write, len 13, buf 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
msg 1: addr 0x15, read, len 13, buf 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```
(this latter command did not have any visible effect on the NumberPad if it was supposed to). Any suggestion to ensure the persistence of the backlight function?

ldrahnik, 2022-12-27:
Weird, when it will be broken, please try `0x01` to enable and then send at least one more command with specific brightness like this for example:

```
i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x46 0xad r13
```

ldrahnik, 2022-12-27:
When it will be broken, please try `0x01` to enable and then send at least one more command with specific brightness like this for example (`0x46`, it should be maximum of brightness):

```
i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x46 0xad r13
```

ldrahnik, 2022-12-27:
Also, plase make sure you send commands to correct i2c. Simple this command below confirms `i2c-2` is touchpad `ASUE140D:00`, where I found this `ASUE...`?, for example here `evtest --grab`:

```
/dev/input/event11:	ASUE140D:00 04F3:31B9 Mouse
/dev/input/event12:	ASUE140D:00 04F3:31B9 Touchpad
/dev/input/event13:	ASUE140D:00 04F3:31B9 Keyboard
```

So in terminal try this or find correct i2c-xyz:

```
cd /sys/bus/i2c/devices/i2c-2/device/i2c-2/i2c-ASUE140D:00
```

EscarM1, 2022-12-27:
I tried the following commands:
```
# evtest --grab
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Asus WMI hotkeys
/dev/input/event5:	USB2.0 HD UVC WebCam: USB2.0 HD
/dev/input/event6:	USB2.0 HD UVC WebCam: USB2.0 IR
/dev/input/event7:	Video Bus
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
/dev/input/event10:	ASUE1410:00 04F3:3230 Mouse
/dev/input/event11:	ASUE1410:00 04F3:3230 Touchpad
/dev/input/event12:	ASUE1410:00 04F3:3230 Keyboard
/dev/input/event13:	sof-hda-dsp Headset Mic
/dev/input/event14:	sof-hda-dsp Headphone
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event16:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event17:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event18:	Asus Touchpad/Numpad
/dev/input/event19:	Logitech Pebble Mouse
```
It looks like there are 2 related entries (event11 and event18).
```
# ll /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 déc 27 12:06 ./
drwxr-xr-x 4 root root 0 déc 27 12:06 ../
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-0/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-1 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-12 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-12/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-13/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-14/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-15 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-15/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-2 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-3 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-3/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-4 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-4/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-5 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-5/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-ALED0217:00 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-12/i2c-ALED0217:00/
lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1410:00/
```
Now I'm a bit confused: which is the correct i2c? It looks like it is i2c-2 or i2c-ASUE1410:00 (last line above). Anyway the 2 commands
```
# i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
msg 0: addr 0x15, write, len 13, buf 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
msg 1: addr 0x15, read, len 13, buf 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
# i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x46 0xad r13
msg 0: addr 0x15, write, len 13, buf 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x46 0xad
msg 1: addr 0x15, read, len 13, buf 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```
did not produce any visible effect on the NumberPad.

ldrahnik, 2022-12-27:
@EscarM1 `event18` is this driver for NumberPad, `event 10 - 12` are events for touchpad but without NumberPad (touchpad send event keyboard and also support behaviour like mouse, that is reason why exist 3 different events for 1 touchpad) functionality. So this row confirms i2c-2 is correct `lrwxrwxrwx 1 root root 0 déc 27 11:06 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1410:00/`. Well, atm i do not know

EscarM1, 2022-12-27:
I installed Ubuntu 22.04.1 LTS from a freshly downloaded ISO written on an USB key 3 days ago (24.12.2022) in parallel with Windows 11 on the same laptop (encrypting the partition with LUKS during install). Besides installing a few software from the Ubuntu distro I did not do much with this system yet, so it should be pretty clean.

ldrahnik, 2022-12-27:
I am right when I say when the problem occurred running again `sudo ./install.sh` solved the problem? Immediately? Without any restart? Every time? Please confirm.

EscarM1, 2022-12-27:
Unfortunately this does not seem to be that clear:
- After the first ./install it did not work at all (neither immediately nor after restart). That's why I started reporting the issue here.
- After the second ./install it did work immediately and again after the first restart (not sure how many restarts were done before it stopped working).
- After the third ./install (which I just tested now) it did not work immediately but it did after the first restart (and after the second also). Note that the first restart did not work smoothly (I got a black screen after selecting 'Restart' in the menu and I had to switch the laptop off by holding the power button), so maybe it was not in a ideal state.
So just now it is working fine. I'll report about the next tries in the next few days.

EscarM1, 2022-12-28:
After a fresh login the next morning, the backlight no longer works (touchpad/numpad works for entering numbers, but I can't see which key I'm hitting). Still no clue what I should do to fix this. Running ./install after every new login seems a little too much...
I'll be glad to check and report  more system parameters if necessary.

ldrahnik, 2022-12-28:
@EscarM1 
With described below should be excluded problems with i2c.

Uninstall driver `sudo ./uninstall.sh`, restart and again try to send values by i2c manually so we can be sure code of the driver is not a problem and we can 100% exclude it. Then set up tracing i2c (like is described here https://michlstechblog.info/blog/linux-enable-i2c-kernel-tracing), rewritten with using tee here:

```
echo nop | sudo tee /sys/kernel/debug/tracing/current_tracer
echo 1 | sudo tee /sys/kernel/debug/tracing/events/i2c/enable
echo 1 | sudo tee /sys/kernel/debug/tracing/tracing_on
```

Then again try to change manually and check log `sudo cat /sys/kernel/debug/tracing/trace`, check if send commands exist for example with pipeline || grep|. Should be there. But then clear the log `echo > /sys/kernel/debug/tracing/trace
`, then restart and after restart looks again and attach entire log here. Also could be nice clean attached log after restart when it is broken and manually sending do nothing with backlight and when it's ok.

EscarM1, 2022-12-28:
OK, I'll do that, but first I'd like to report that after a new restart the backlight started working again. So it looks like this is not a regular phenomenon, but rather it seems to work intermittently. Here is the fresh output of the two commands from yesterday:

```
# evtest --grab
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Asus WMI hotkeys
/dev/input/event5:	USB2.0 HD UVC WebCam: USB2.0 HD
/dev/input/event6:	USB2.0 HD UVC WebCam: USB2.0 IR
/dev/input/event7:	Video Bus
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
/dev/input/event10:	ASUE1410:00 04F3:3230 Mouse
/dev/input/event11:	ASUE1410:00 04F3:3230 Touchpad
/dev/input/event12:	ASUE1410:00 04F3:3230 Keyboard
/dev/input/event13:	Asus Touchpad/Numpad
/dev/input/event14:	sof-hda-dsp Headset Mic
/dev/input/event15:	sof-hda-dsp Headphone
/dev/input/event16:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event17:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event18:	sof-hda-dsp HDMI/DP,pcm=5
/dev/input/event19:	Logitech Pebble Mouse
```
It's basically identical, only the "Asus Touchpad/Numpad" line is higher in the list (event13 instead of event18).
```
# ll /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 déc 28 11:54 ./
drwxr-xr-x 4 root root 0 déc 28 11:54 ../
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-0/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-1 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-12 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-12/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-13/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-14/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-15 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-15/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-2 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-2/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-3 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-4 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-4/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-5 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-5/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-ALED0217:00 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-12/i2c-ALED0217:00/
lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ASUE1410:00/
```
So it looks like it is now i2c-3 instead of i2c-2! Is it really supposed to change? Or was it during the 3rd ./install? Anyway, when the backlight is working when tapping the upper right corner of the touchpad, it is also working when entering the command
```
# i2ctransfer -v -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad r13
msg 0: addr 0x15, write, len 13, buf 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
msg 1: addr 0x15, read, len 13, buf 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

```
and yesterday's tests have shown that the converse is true (i.e., when tapping the upper right corner of the touchpad fails to switch on the backlight, so does the manual command).
Any new idea? Should I still perform the steps of your previous message?

ldrahnik, 2022-12-28:
@EscarM1 The way how to confirm the driver has saved the wrong i2c device id. Every laptop start driver loads the device id again. Well, first pull the latest master. 

- uninstall `sudo ./uninstall.sh`
- clear error log `sudo rm -f /var/log/asus_touchpad_numpad-driver/error.log
- install `sudo ./install.sh`
- look which i2c is saved (in the log it will be a row like this: `INFO:Pad:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input29`) when it works and when it does not work and compare with `/sys/bus/i2c/devices/`

Also, happened situation it was working OK and then it stopped working without any restart?

EscarM1, 2022-12-28:
Ok, did all that. Relevant log file line is
`INFO:Pad:Set touchpad device id 3 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ASUE1410:00/0018:04F3:3230.0001/input/input14`
and `ll /sys/bus/i2c/devices/` shows `lrwxrwxrwx 1 root root 0 déc 28 10:54 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ASUE1410:00/` just after install (and the backlight works fine).
I'll check and report this info + backlight working status the next few times I restart.
No, I did never notice that the backlight working status changed without restart.

EscarM1, 2022-12-28:
Ok, I restarted a few times and the backlight worked each time, so could not reproduce the problem so far. I'll try again tomorrow.
By the way, I tried to use some functions described on the 'https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver' main page but did not succeed. For that purpose I edited some parameters in the `/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev` file. Could this be a sign of a problematic install? Some examples:
- Brightness levels are not available (only on and off), even when using the command line `i2ctransfer -v -f -y # w13 ...`. On the other hand, there is no "up left" button on the touchpad, so maybe this is just a feature that does not exist on that particular model.
- I couldn't find how to enable the feature described as "By default ignoring touchpad physical buttons events (left, right and middle one) when is NumberPad activated, unless when are buttons outside the NumberPad area like layout gx701 has (ignoring can be disabled)". So when the NumberPad is enabled, the mouse pointer cannot be moved using the touchpad.

ldrahnik, 2022-12-28:
@EscarM1 Ok, let me know about any news.
 
- Unfortunately, your laptop does not have an icon for brightness change so it looks like brightness levels without on/off are not supported, if i2c does not work it only confirms the missing icon.
- `touchpad_physical_buttons_are_inside_numpad` in the config file, IMHO when you have enabled NumberPad, you do not want to have enabled left, right and middle clicks (not touches, but clicks), at least I think so because you want use calculator and click on left for example means changed opened app to another etc. So atm it is only possible to change mentioned option and let the driver know the physical buttons are outside so they will not be disabled when is NumberPad enabled. Option is used only for this, so functionality is the same as the name would be 'disabled_touchpad_buttons_when_is_numberpad_activated = (default true)`

ldrahnik, 2023-01-01:
@EscarM1 Any news?

EscarM1, 2023-01-02:
Restarted the computer twice this morning, unfortunately Numpad backlight does not work (I was away and could not test it these past two days). Not sure what I can do. Reinstall may again solve the problem for a while but most probably not for very long. I'm ready to perform more checks if needed.

ldrahnik, 2023-01-02:
@EscarM1 When enabling/disabling backlight does not work, please do the same as was written in the previous post. First `git pull master` and `reinstall`, I removed the logging not-error status about saving the config file. Was there a lot of mess with that.

From previous post:

The way how to confirm the driver has saved the wrong i2c device id. 

```
look which i2c is saved (in the log cat /var/log/asus_touchpad_numpad-driver/error.log it will be a row like this in mine: INFO:Pad:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31B9.0002/input/input29) when it works and when it does not work and compare with row in cat /sys/bus/i2c/devices/

EscarM1, 2023-01-04:
After the last restart the backlight was not working any more, so I looked for the relevant information:

```
# ls -alF /sys/bus/i2c/devices/ | grep ASUE1410
lrwxrwxrwx 1 root root 0 jan  4 08:36 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1410:00/
```
```
# cat /var/log/asus_touchpad_numpad-driver/error.log | grep ASUE1410
...
INFO:Pad:Detecting touchpad from string: "N: Name="ASUE1410:00 04F3:3230 Touchpad""
INFO:Pad:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1410:00/0018:04F3:3230.0001/input/input14
```
It looks like the ID is correct. I also uploaded the full error.log file... which still contains quite a lot of things. 
[error.log](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/10342303/error.log)

Need more info?

ldrahnik, 2023-01-05:
@EscarM1 About button pointers: confirmed as a bug. The option `touchpad_physical_buttons_are_inside_numpad` did nothing. Now, config ption renamed to `enabled_pointer_buttons` and by default enabled! Now should be possible to use LEFT, MIDDLE, RIGHT touchpad button when is NumberPad activated. On the bottom area where is no NumberPad number, character alone or even with number. Can not be excluded situation when touch persists for example 10s and then is pressed with a higher pressure button. In this case is send number even used button. Could you please test it? Changes are in branch https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/pull/72

ldrahnik, 2023-01-05:
@EscarM1 About backlight: and sending manually i2c commands to the Touchpad device to enable backlight was not helpful too?

EscarM1, 2023-01-05:
**About button pointers:**
I uninstalled, downloaded the latest git and installed again, added the `touchpad_physical_buttons_are_inside_numpad` keyword in the `/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev` file and set its value = 0. I could not see any effect, but actually on my laptop, the numpad indeed covers the full touchpad, so the buttons are really inside it, so maybe this just does not work for me.
Also not 100% sure how to test for sure. I clicked on both the left and right buttons at the button of the touchpad, but no effect (e.g. no context menu shows up with right button).

EscarM1, 2023-01-05:
**About backlight:**
Made several restarts again. When backlight works with touchpad upper right button, then the manual ic2 command also works, but when backlight does not work with touchpad upper right button, neither does it with the manual i2c command. So both are perfectly correlated. Here the latest outputs when backlight does **not** work:
```
# ls -alF /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 jan  5  2023 ./
drwxr-xr-x 4 root root 0 jan  5  2023 ../
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-0/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-1 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-12/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-13/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-14/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-15 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-15/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-2 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-3 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-3/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-4 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-4/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-5 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-5/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-ALED0217:00 -> ../../../devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-3/i2c-ALED0217:00/
lrwxrwxrwx 1 root root 0 jan  5 18:37 i2c-ASUE1410:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1410:00/
```
The error.log file is attached.
[error_2023-01-05_18-48.log](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/10354473/error_2023-01-05_18-48.log)
Any other suggestion for solving backlight problem?

ldrahnik, 2023-01-05:
About button pointers: changes were done in the branch, and now I pull request (branch) merged to master, so you can download even .zip or via .git `git pull origin master`. `touchpad_physical_buttons_are_inside_numpad` has been removed, so in config is now valid option named `enabled_pointer_buttons = 1` (valid values are 0, 1, 1 is default value). Try 1, you should be able use after installing left, middle and right button when is NumberPad activated, with value 0 not able.

EscarM1, 2023-01-05:
**About button pointers:**
Yes, now the 3 buttons do work when the numpad is enabled, so that's very nice. I do not have `libinput-tools` installed so I can't see the output of `sudo libinput debug-events` command, but it would certainly show the correct info as well since it's working.

**About backlight:**
Still no clue. I can only summarise the situation as follow:

- When pressing upper right touchpad button, numpad always works, but backlight works only sometimes.
- When backlight works with touchpad upper right button, then the manual ic2 command also works. The negative of this statement is also true.
- When the backlight switch works, then it continues working until reboot (it does not stop working by itself). The negative of this statement is also true. So it is only upon system restart that the behaviour may change.
- The `ls -alF /sys/bus/i2c/devices/` output and the `cat /var/log/asus_touchpad_numpad-driver/error.log | grep ASUE1410` output do not look really different whether backlight is working or not, so this information does not look to be a good diagnosis criterium.

Any idea where I should look into?

closed 2023-01-05 by ldrahnik

EscarM1, 2023-01-06:
OK, thanks for the advice. I may look into that if I find more time later on.