asus-linux-drivers/asus-numberpad-driver #73: [Bug]: Does not work on ux582hm

jordann93, 2023-01-06:
### Describe the bug

Numpad does not switch on by tapping numpad icon on the touchpad.

### Expected behavior

Numpad should be LED illuminated and working after tapping numpad icon.

### Relevant log output

```shell
LOG=DEBUG sudo -E ./asus_touchpad.py "ux581l" ""
INFO:Pad:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:Pad:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:Pad:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:Pad:Set touchpad device id 3 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-3/i2c-ELAN9009:00/0018:04F3:2C23.0002/input/input47
INFO:Pad:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:Pad:Touchpad min-max: x 0-3984, y 0-1152
INFO:Pad:Numpad min-max: x 80-3904, y 200-1072
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN9009:00 04F3:2C23 Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN9009:00 04F3:2C23 Touchpad
INFO:Pad:Getting Device Enabled via xinput failed
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ELAN9009:00 04F3:2C23 Touchpad
```

### Desktop

- Asus Zenbook UX582HM
- Program version : latest
- Distro : Ubuntu 22.04.1
- Kernel : 5.15.0-56-generic

ldrahnik, 2023-01-06:
Hello there,
try to start the driver via the option in the config file (`enabled = 1`) and me know about the result.

jordann93, 2023-01-06:
> via the option in the config file

Hello,
Thank you for your response.

Unfortunately, Still doesn't work :
```
INFO:Pad:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:Pad:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:Pad:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:Pad:Set touchpad device id 3 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-3/i2c-ELAN9009:00/0018:04F3:2C23.0002/input/input46
INFO:Pad:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:Pad:Touchpad min-max: x 0-3984, y 0-1152
INFO:Pad:Numpad min-max: x 80-3904, y 200-1072
INFO:Pad:Numpad activated
```

ldrahnik, 2023-01-06:
@jordann93 Can you more describe the problem? The backlight was not enabled? (if yes, try commands i2c transfer what I send to previously closed issue). Because in the log is nothing wrong.

ldrahnik, 2023-01-06:
@jordann93 Could you attach the entire log when was driver installed and not running stand-alone?

jordann93, 2023-01-06:
@ldrahnik the exact problem is backlight numpad doesn't work, script does nothing to my laptop and nothing happens when I touch numpad after active log.

- sudo ./install.sh 
```
apt is here
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
libevdev2 est déjà la version la plus récente (1.12.1+dfsg-1).
i2c-tools est déjà la version la plus récente (4.3-2build1).
python3-pip est déjà la version la plus récente (22.0.2+dfsg-1).
git est déjà la version la plus récente (1:2.34.1-1ubuntu1.5).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  chromium-codecs-ffmpeg-extra gstreamer1.0-vaapi i965-va-driver intel-media-va-driver libaacs0 libaom3 libass9 libavcodec58 libavformat58 libavutil56 libbdplus0 libblas3 libbluray2 libbs2b0 libchromaprint1
  libcodec2-1.0 libdav1d5 libflashrom1 libflite1 libftdi1-2 libgme0 libgsm1 libgstreamer-plugins-bad1.0-0 libigdgmm12 liblilv-0-0 libmfx1 libmysofa1 libnorm1 libopenmpt0 libpgm-5.3-0 libpostproc55 librabbitmq4
  librubberband2 libserd-0-0 libshine3 libsnappy1v5 libsord-0-0 libsratom-0-0 libsrt1.4-gnutls libssh-gcrypt-4 libswresample3 libswscale5 libudfread0 libva-drm2 libva-wayland2 libva-x11-2 libva2 libvdpau1
  libvidstab1.1 libx265-199 libxvidcore4 libzimg2 libzmq5 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers pocketsphinx-en-us va-driver-all vdpau-driver-all
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 23 non mis à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
xinput est déjà la version la plus récente (1.6.3-1build2).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  chromium-codecs-ffmpeg-extra gstreamer1.0-vaapi i965-va-driver intel-media-va-driver libaacs0 libaom3 libass9 libavcodec58 libavformat58 libavutil56 libbdplus0 libblas3 libbluray2 libbs2b0 libchromaprint1
  libcodec2-1.0 libdav1d5 libflashrom1 libflite1 libftdi1-2 libgme0 libgsm1 libgstreamer-plugins-bad1.0-0 libigdgmm12 liblilv-0-0 libmfx1 libmysofa1 libnorm1 libopenmpt0 libpgm-5.3-0 libpostproc55 librabbitmq4
  librubberband2 libserd-0-0 libshine3 libsnappy1v5 libsord-0-0 libsratom-0-0 libsrt1.4-gnutls libssh-gcrypt-4 libswresample3 libswscale5 libudfread0 libva-drm2 libva-wayland2 libva-x11-2 libva2 libvdpau1
  libvidstab1.1 libx265-199 libxvidcore4 libzimg2 libzmq5 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers pocketsphinx-en-us va-driver-all vdpau-driver-all
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 23 non mis à jour.
Requirement already satisfied: libevdev in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (0.5)
Requirement already satisfied: evdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 2)) (1.6.0)
Requirement already satisfied: numpy in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 3)) (1.24.1)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ZenBook Pro Duo UX582HM_UX582HM
Automatically recommended numpad layout: ux581l (associated to UX582HM-OLED032W). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad.service → /etc/systemd/system/asus_touchpad_numpad.service.
Asus touchpad service enabled
Created symlink /etc/systemd/system/suspend.target.wants/asus_touchpad_numpad_suspend.service → /etc/systemd/system/asus_touchpad_numpad_suspend.service.
Asus touchpad suspend service enabled
Asus touchpad service started
Asus touchpad suspend service started
```

jordann93, 2023-01-06:
> @jordann93 Can you more describe the problem? The backlight was not enabled? (if yes, try commands i2c transfer what I send to previously closed issue). Because in the log is nothing wrong. Numpad was activated.

I can enable/disable numpad backlight with i2c commands :
```
sudo i2ctransfer -v -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad r13
```
But still numpad doesn't work

ldrahnik, 2023-01-07:
@jordann93 Could you attach the entire log `/var/log/asus_touchpad_numpad-driver/error.log` and could you check if is driver service running with command: `systemctl status asus_touchpad_numpad`

jordann93, 2023-01-07:
@ldrahnik Hello,
Full error log :
```
INFO:Pad:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:Pad:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:Pad:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:Pad:Set touchpad device id 3 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-3/i2c-ELAN9009:00/0018:04F3:2C23.0002/input/input46
INFO:Pad:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:Pad:Touchpad min-max: x 0-3984, y 0-1152
INFO:Pad:Numpad min-max: x 80-3904, y 200-1072
INFO:Pad:Numpad activated
Error: Sending messages failed: Remote I/O error
INFO:Pad:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:Pad:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:Pad:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:Pad:Set touchpad device id 3 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-3/i2c-ELAN9009:00/0018:04F3:2C23.0002/input/input46
INFO:Pad:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:Pad:Touchpad min-max: x 0-3984, y 0-1152
INFO:Pad:Numpad min-max: x 80-3904, y 200-1072
Unable to connect to X server
INFO:Pad:Getting Device Enabled via xinput failed
INFO:Pad:System numlock activated
INFO:Pad:Numpad activated
Error: Sending messages failed: Remote I/O error
Unable to connect to X server
INFO:Pad:Getting Device Enabled via xinput failed
Unable to connect to X server
INFO:Pad:Getting Device Enabled via xinput failed
Unable to connect to X server
INFO:Pad:Getting Device Enabled via xinput failed
Unable to connect to X server
INFO:Pad:Getting Device Enabled via xinput failed
Unable to connect to X server
```

And use of systemctl command give me :
```
● asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-01-07 14:28:27 CET; 3min 47s ago
   Main PID: 34595 (python3)
      Tasks: 20 (limit: 38071)
     Memory: 23.2M
        CPU: 4.481s
     CGroup: /system.slice/asus_touchpad_numpad.service
             └─34595 python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py ux581l /usr/share/asus_touchpad_numpad-driver/

janv. 07 14:28:27 jordann-ZenBook-Pro-Duo-UX582HM-UX582HM systemd[1]: Started Asus Touchpad to Numpad Handler.
```

ldrahnik, 2023-01-07:
@jordann93 Thank your, in log is line `Error: Sending messages failed: Remote I/O error`. Are you able run the driver in for example Visual Code like me and get a call stack when is error thrown? Mine launch file `.vscode/launch.json` for debugging driver python file `asus_touchpad.py` is:

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal",
            "justMyCode": true,
            "sudo": true,
            "args": ["up5401ea", ""]
        }
    ]
}
```

jordann93, 2023-01-07:
@ldrahnik Thank you,
Error after debug asus_touchpad.py, line 635 :
```
NoOptionError
No option 'brightness' in section: 'main'

During handling of the above exception, another exception occurred:

  File "/home/jordann/Asus/asus-touchpad-numpad-driver/asus_touchpad.py", line 635, in load_all_config_values
    default_backlight_level = config.get(CONFIG_SECTION, CONFIG_LAST_BRIGHTNESS)
  File "/home/jordann/Asus/asus-touchpad-numpad-driver/asus_touchpad.py", line 674, in <module>
    load_all_config_values()
```

ldrahnik, 2023-01-07:
@jordann93 Both exceptions are thrown inside try-catch blocks and already handled. So nothing wrong. What I need from you is a call stack when is logged `Error: Sending messages failed: Remote I/O error`

ldrahnik, 2023-01-07:
IMHO atm it looks like i2c problem but is weird that command for changing backlights in terminal works for you, so I need that call stack / line / function for identify source of problem for sure. Could you comment out all these lines `subprocess.call(` when is changed backlight via i2c so we can confirm it is i2c problem?

jordann93, 2023-01-07:
> 

Is that what you are asking ?
[image]
This is call stack when I have error `Error: Sending messages failed: Remote I/O error`

jordann93, 2023-01-07:
> @jordann93 IMHO atm it looks like i2c problem but is weird that command for changing backlights in terminal works for you, so I need that call stack / line / function for identify source of problem for sure. Could you comment out all these lines `subprocess.call(` when is changed backlight via i2c so we can confirm it is i2c problem?

I comment all theses line and I no longer have the error (but still doesn't work).
Yes I confirm, use i2ctransfer command to change backlight work

ldrahnik, 2023-01-08:
@jordann93 When were lines commented out, what was wrong? Could you  please comment these lines out and then install the driver and attach here the entire log file `/var/log/asus_touchpad_numpad-driver/error.log`?

ldrahnik, 2023-01-08:
@jordann93 Also, please put a breakpoint on each line `subprocess.call(` and check variable `device_id`, use this `device_id` in terminal via `i2c transfer`, so the same command, but yourself. So we can exclude command try change register which for example does not exist. So like this:

```
i2ctransfer -f -y <device_id> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
````

ldrahnik, 2023-01-11:
@jordann93 Closed for issue owner inactivity, can be reopened, is required more information for confirmation that error message has something to do with driver or can be something changed/added to prevent this error occurred `Error: Sending messages failed: Remote I/O error`. Anyway, it is i2c problem.

closed 2023-01-11 by ldrahnik