asus-linux-drivers/asus-numberpad-driver #74: [Bug]: Fedora 37 - installation fails

ShinJii89, 2023-01-07:
### Describe the bug

After sudo ./install.sh there's errors and I can't install that driver :( 

### Expected behavior

Numpad driver to be installed ;) 

### Relevant log output

```shell
Requirement already satisfied: libevdev in /usr/lib/python3.11/site-packages (from -r requirements.txt (line 1)) (0.11)
Collecting evdev
  Using cached evdev-1.6.0.tar.gz (26 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in /usr/lib64/python3.11/site-packages (from -r requirements.txt (line 3)) (1.22.0)
Using legacy 'setup.py install' for evdev, since package 'wheel' is not installed.
Installing collected packages: evdev
  Running setup.py install for evdev ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for evdev did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      running install
      /usr/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-311
      creating build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/util.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/uinput.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/genecodes.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/ff.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/evtest.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/events.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/eventio_async.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/eventio.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/ecodes.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/device.py -> build/lib.linux-x86_64-cpython-311/evdev
      copying evdev/__init__.py -> build/lib.linux-x86_64-cpython-311/evdev
      running build_ext
      running build_ecodes
      writing ecodes.c (using /usr/include/linux/input.h /usr/include/linux/input-event-codes.h /usr/include/linux/uinput.h)
      building 'evdev._input' extension
      creating build/temp.linux-x86_64-cpython-311
      creating build/temp.linux-x86_64-cpython-311/evdev
      gcc -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python3.11 -c evdev/input.c -o build/temp.linux-x86_64-cpython-311/evdev/input.o -std=c99 -Wno-error=declaration-after-statement
      evdev/input.c:10:10: fatal error: Python.h: Nie ma takiego pliku ani katalogu
         10 | #include <Python.h>
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> evdev

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
pip dependencies via file requirements.txt cannot be loaded correctly.
```

### Desktop

- Program version [e.g. 57c742a]: 1.7.0
- Distro [e.g. Arch Linux] Fedora 37 (KDE)
- Kernel [e.g. 5.18.15.arch1-2] 6.1.3 from asus-linux.org

ldrahnik, 2023-01-07:
Hello there,
your system probably just does not have installed headers for python (associated StackOverflow topic https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory). Helped? Let me know, please.

ShinJii89, 2023-01-07:
> Hello there, your system probably just does not have installed headers for python (associated StackOverflow topic https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory). Helped? Let me know, please.

It worked! Thanks!!! :)

closed 2023-01-07 by ShinJii89

ldrahnik, 2023-01-07:
@ShinJii89 You're welcome. Keep asking about anything. If you think the driver is useful, do not forget to put a start on the repository :)