asus-linux-drivers/asus-numberpad-driver #79: [Bug]: Mouse not working after NumberPad input

Lud1ma, 2023-01-18:
### Describe the bug

On my TP3402Z Notebook, starting the NumberPad by pressing the right-corner icon works, and all numbers work as intended. But when I press the top-right-corner icon again, the touchpad doesn't work for mouse input anymore, until I stop the driver.
But I can start the NumberPad again and it works as intended.

I tested it on the commit f6e002f69617e509884d78cc4b8ef344bfbc79b2 (Monday, January 16th, 2023), where everything works a intended.
It appears to be a problem with the newest commit.

Any ideas, what could lead to this problem?

### Expected behavior

I expected the NumberPad to turn off and the Touchpad to start working again, as in previous versions.

### Relevant log output

```shell
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31BC Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31BC.0001/input/input18
INFO:asus-touchpad-numpad-driver:Set touchpad id 15 from H: Handlers=mouse1 event15
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3880, y 0-2299
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3680, y 200-2219
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3799 y 87
INFO:asus-touchpad-numpad-driver:Touched numlock in time: 1674085305.041168
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.0031204223632812
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Numpad activated
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3767 y 162
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1994 y 498
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP9:73
INFO:asus-touchpad-numpad-driver:grab
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP9:73
9INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1976 y 483
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3236 y 797
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:grab
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 0
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1983 y 1961
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KPENTER:96
INFO:asus-touchpad-numpad-driver:grab
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KPENTER:96

INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1985 y 1959
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3814 y 108
INFO:asus-touchpad-numpad-driver:Touched numlock in time: 1674085312.6813283
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.0020999908447266
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Numpad deactivated
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3798 y 125
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1938 y 1065
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2312 y 1306
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2295 y 933
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1972 y 1400
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2216 y 916
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1950 y 1550
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1788 y 1633
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1984 y 1501

## After line "Numpad deactivated", the mouse movement wasn't working.
```

### Desktop

- Program version: 0aa476fb5a86a57f56318840bc539b9b75474fdf  (Tuesday, January 17th, 2023) 
- Distro: Ubuntu 22.04 with KDE Plasma 5.24.7
- Kernel: 5.15.0-58-generic (64-bit)

ldrahnik, 2023-01-19:
@Lud1ma Thank you for the issue, it should be fixed now.

ldrahnik, 2023-01-20:
@Lud1ma Let me please know that update fixed this issue.

ldrahnik, 2023-01-21:
@Lud1ma Closed because of issue owner inactivity can be reopened.

closed 2023-01-21 by ldrahnik

ddalt10, 2023-02-12:
Having same issue as OP on current commit on UX5401ZAS, restarting service is only work around to get touchpad mouse working again

ldrahnik, 2023-02-12:
@ddalt10 Thank you. It should be fixed now.

ddalt10, 2023-02-12:
> @ddalt10 Thank you. It should be fixed now.

Confirming it works now - Thank you @ldrahnik  !