asus-linux-drivers/asus-numberpad-driver #80: [Bug]: Installation looks successful but doesn't work

bettker, 2023-01-20:
### Describe the bug

I can't activate the numpad after installation (not even during the session, unlike #78), but I can use it normally when troubleshooting.

Following are the outputs for `sudo ./install.sh` and `sudo systemctl status asus_touchpad_numpad`.

```
~/Projects/asus-touchpad-numpad-driver master ❯ sudo ./install.sh  
pacman is here
warning: libevdev-1.13.0-1 is up to date -- skipping
warning: i2c-tools-4.3-3 is up to date -- skipping
warning: git-2.39.0-1 is up to date -- skipping
warning: python-3.10.8-3 is up to date -- skipping
warning: python-pip-22.3.1-1 is up to date -- skipping
 there is nothing to do
warning: xorg-xinput-1.6.3-3 is up to date -- skipping
 there is nothing to do
Requirement already satisfied: libevdev in /home/bettker/.pyenv/versions/3.9.10/lib/python3.9/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /home/bettker/.pyenv/versions/3.9.10/lib/python3.9/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /home/bettker/.pyenv/versions/3.9.10/lib/python3.9/site-packages (from -r requirements.txt (line 3)) (1.22.3)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-0 : failed
Testing interface i2c-1 : sucess
Detected laptop: ZenBook UX325JA_UX325JA
Could not automatically detect numpad layout for your laptop. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).

Select your model keypad layout:
1) b7402.py       3) g513.py       5) gx551.py      7) up5401ea.py   9) ux581l.py
2) e210ma.py      4) g533.py       6) gx701.py      8) ux433fa.py   10) Quit
Please enter your choice 7
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad.service → /etc/systemd/system/asus_touchpad_numpad.service.
Asus touchpad service enabled
Created symlink /etc/systemd/system/suspend.target.wants/asus_touchpad_numpad_suspend.service → /etc/systemd/system/asus_touchpad_numpad_suspend.service.
Asus touchpad suspend service enabled
Asus touchpad service started
Asus touchpad suspend service started
```

```
× asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2023-01-20 19:37:38 -03; 5min ago
   Duration: 133ms
    Process: 58782 ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea /usr/share/asus_touchpad_numpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 58782 (code=exited, status=1/FAILURE)
        CPU: 132ms

Jan 20 19:37:38 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Scheduled restart job, restart counter is at 4.
Jan 20 19:37:38 bettker-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jan 20 19:37:38 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Start request repeated too quickly.
Jan 20 19:37:38 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
Jan 20 19:37:38 bettker-zenbook systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
```

### Expected behavior

Work after installation.

### Desktop

- Program version: up5401ea
- Distro: Manjaro
- Kernel: 5.15.85-1-MANJARO
- Laptop: ux325ja

ldrahnik, 2023-01-20:
@bettker Thank you for creating an issue. Please attach entire log file `/var/log/asus_touchpad_numpad-driver/error.log`

bettker, 2023-01-20:
Apparently my problem is that it's not finding `evdev`:
```
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
```

From what I understand what is happening is that I use `pyenv` to manage the python installations and my default installation `/usr/bin/python` does not have `evdev`. Somewhere the driver is invoked with `/usr/bin/python` instead of using the `python` or `python3` alias.

Any idea where this happens? Obviously, a quick solution is to install `evdev` in `/usr/bin/python`, but in the long run it could be a problem again since I don't intend to keep this installation up to date.

ldrahnik, 2023-01-20:
Did you check `which python3` and `which python`?

bettker, 2023-01-20:
Yes, it's in the `pyenv` install that has the `evdev`.

```
~ ❯ which python3
/home/bettker/.pyenv/shims/python3
~ ❯ which python
/home/bettker/.pyenv/shims/python
```

ldrahnik, 2023-01-20:
Please try to change `ExecStart` line in service 3 files `*service` to: `ExecStart=/bin/sh -c 'python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py $LAYOUT $CONFIG_FILE_DIR'`

bettker, 2023-01-21:
I uninstalled and installed again and it still doesn't work.

This is weird because I run the exact same process command... and it works.
```
 /etc/systemd/system/graphical.target.wants ❯ systemctl start asus_touchpad_numpad.service
 /etc/systemd/system/graphical.target.wants ❯ systemctl status asus_touchpad_numpad.service
× asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2023-01-20 21:10:41 -03; 3s ago
   Duration: 140ms
    Process: 7435 ExecStart=/bin/sh -c python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea /usr/share/asus_touchpad_numpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 7435 (code=exited, status=1/FAILURE)
        CPU: 139ms

Jan 20 21:10:41 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Scheduled restart job, restart counter is at 5.
Jan 20 21:10:41 bettker-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jan 20 21:10:41 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Start request repeated too quickly.
Jan 20 21:10:41 bettker-zenbook systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
Jan 20 21:10:41 bettker-zenbook systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
 /etc/systemd/system/graphical.target.wants ❯ sudo /bin/sh -c 'python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py up5401ea /usr/share/asus_touchpad_numpad-driver/'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140A:00 04F3:3134 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input11
INFO:asus-touchpad-numpad-driver:Set touchpad id 7 from H: Handlers=event7 mouse1
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-1972
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-1892
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2606 y 640
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2616 y 827
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3915 y 91
INFO:asus-touchpad-numpad-driver:Touched numlock in time: 1674259873.3478348
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.0070791244506836
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Numpad activated
```

FYI I use zsh and had tested both `which` in sh and bash but they are all pointing to the same pyenv path.
I'm trying to find out more about.

ldrahnik, 2023-01-21:
@bettker Let me know if the latest commit resolved the reported issue. Using `/usr/bin/env/` in python serves one important purpose. As python supports virtual environments, using `/usr/bin/env` python will make sure that your scripts runs inside the virtual environment, if you are inside one. Whereas, `/usr/bin/python` will run outside the virtual environment.

ldrahnik, 2023-01-21:
@bettker ping

bettker, 2023-01-21:
I was working on it just now.

Even with the last commit it doesn't seem to work on my machine, and when using `/usr/bin/env python3 --version` the version of `/usr/bin/python3` is printed instead of pyenv, i.e. it still has the same problem.

I tried a few more things, and even when starting pyenv from the service file, my python installations are not present (even though `which pyenv` both point to the same pyenv path). Apparently the whole environment is installed just for my user and root can't identify it.

Anyway, it seems to be a problem in my environment and not something that should be fixed in the driver, so the solution I took was to hardcode the path to my python3 (`/home/bettker/.pyenv/shims/python3` ) in the service files and now everything works.

Thanks a lot for the help.

closed 2023-01-21 by bettker

ldrahnik, 2023-01-21:
@bettker I can confirm I installed `.pyenv` and the last commit was helpful to me when I pushed the commit. Without that it was not working as you described.