asus-linux-drivers/asus-numberpad-driver #82: Close calculator

encrustace, 2023-01-22:
### Short Description

The top left icon touch and slide to centre enables the touchpad and launch the calculator and again touching the left icon and slide to centre disables the touchpad but instead of closing the calculator, it launches a new instance of the calculator.

if the challenge is different user uses different calculator apps, there can be added option in the configuration file to take a custom command to execute on the second slide of the top left icon, ie to close the gnome calculator "pkill gnome-calculator" to kill gnome calculator.

One catch also I found I do not know if it is a bug or limitation but if Numpad is already enabled from the top right icon then if trying to launch the calculator by the top left icon then it launches the calculator but it also disabled the touchpad which makes to manually enable Numpad again.

encrustace, 2023-01-22:
https://github.com/encrustace/asus-touchpad-numpad-driver 

@ldrahnik 
I have implemented a calculator open only if it has not been already launched (line number 360), it has other changes as well which I have customized as per my usage please ignore those. for now, the process name is hardcoded but we can add that in the configuration file to make it customizable so that the user can launch and close whatever app the user wants.

ldrahnik, 2023-01-22:
@encrustace Preferred calculator app / script should be set up outside of this drive. Now is already possible to set up activation/deactivation of the preferred calculator app with a slide function starting from top left, driver sends customizable key, by default it is of course `EV_KEY.KEY_CALC` to which can be bound own toggling script. Why I prefer this solution is also because when is set up toggling script once and then on external keyboard you press calculator key, it does the same. Mine script as is mentioned in Readme.md is on Elementary OS based on Ubuntu [this one](https://github.com/ldrahnik/elementary-os-scripts/blob/master/toggle_calculator.sh):

```
#!/bin/bash

if [[ $(ps aux | grep io.elementary.calculator | grep -v grep) ]]; then
	flatpak kill io.elementary.calculator &>/dev/null &
else
	flatpak run io.elementary.calculator &>/dev/null &
fi
```

What I changed and what was confusing was enabling/disabling the touchpad. I removed that code. I hope this solves this issue. Please, let me know.

encrustace, 2023-01-22:
I am using Arch with Gnome 43, for me, it is launching multiple instances of gnome-calculator. it is not closing the previous calculator which was already launched on the first swipe.

ldrahnik, 2023-01-22:
@encrustace Did you add toggling script to key `EV_KEY.KEY_CALC`? I do that like this:

> Toggle calculator (custom keyboard shortcut (key XF86Calculator has to be added via dconf /org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/customX/binding) with command: bash /home/ldrahnik/projects/elementary-os-scripts/toggle_calculator.sh, key for calculator can be tested with command xdotool key XF86Calculator)

[image]

encrustace, 2023-01-23:
[image]

I did this but still, it is just launching multiple calculators but not closing, I have tested the script file it is working fine so the script is ok, I think I am making some mistake in the binding please check the screenshots

encrustace, 2023-01-23:
Setting the custom script and rebooting the system solved the issue.

closed 2023-01-23 by encrustace