asus-linux-drivers/asus-numberpad-driver #87: flatpak dependency missing

bmcgough, 2023-02-08:
I am running a minimal Ubuntu 22.10 install and I get:

```
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
./install.sh: line 259: flatpak: command not found
Install finished
```

ldrahnik, 2023-02-09:
Hello @bmcgough ,
thank you for created an issue. Please try it now, it should be fixed, let me know. Adding support of Ubuntu default calculator is planned in feature request [here](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/83). Then I will probably want add support for selecting used app for calculating but script has to be refactored to be applicable accross linux distributions, `gsettings` is used atm only for binding key.

ldrahnik, 2023-02-09:
@bmcgough I have added installing toggle script automatically for `gnome-calculator`, commit https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/commit/b2f20c7752f1b2127cb93698057957c7aac9b4ad. Please, let me know it works. Thank you.

ldrahnik, 2023-02-09:
@bmcgough ping

ldrahnik, 2023-02-10:
@bmcgough Can be reopened in case the issue remains to be resolved.