asus-linux-drivers/asus-numberpad-driver #88: Brightness button and touchpad are not working for up5401ea

kramarov-evg, 2023-02-08:
I have an ASUS Expertbook B9400CEA-KC0062X. According to the README, it has a layout, corresponding to the "up5401ea" model. Before the installation of the driver, I could only use touchpad as a mouse without "touch to click" behaviour and any numpad functionality. After the installation, I can use the numpad in the touchpad, which is great.
However, I still cannot "tap to click", even with numpad disabled. Also numpad brightness regulation (top left corner button) doesn't do anything. Here's the installation log for the driver:
```
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libevdev2 is already the newest version (1.12.1+dfsg-1).
i2c-tools is already the newest version (4.3-2build1).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 122 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput is already the newest version (1.6.3-1build2).
0 upgraded, 0 newly installed, 0 to remove and 122 not upgraded.
Requirement already satisfied: libevdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 3)) (1.24.2)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ASUS EXPERTBOOK B9400CEA_B9400CEA
Automatically recommended numpad layout: up5401ea (associated to B9400CEA-KC0519R). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]n

Select your model keypad layout:
 1) b7402.py
 2) e210ma.py
 3) g513.py
 4) g533.py
 5) gx551.py
 6) gx701.py
 7) up5401ea.py
 8) ux433fa.py
 9) ux581l.py
10) Quit
Please enter your choice 7
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad.service → /etc/systemd/system/asus_touchpad_numpad.service.
Asus touchpad service enabled
Asus touchpad service started
```

ldrahnik, 2023-02-09:
Hello @kramarov-evg ,
thank you for created an issue. From screenshot of your laptop I can confirm you have the same layout as me. And I use up5401ea.

kramarov-evg, 2023-02-09:
If you need any logs provided, I'll be glad to. Just let me know, what you need

ldrahnik, 2023-02-09:
@kramarov-evg Not working fast touch to click on Touchpad without this driver installed looks like touchpad driver problem.

ldrahnik, 2023-02-09:
@kramarov-evg About not working brightness it is weird, because i2c was identified, could you reinstall, try change brightness and attach entire log please `Reinstall and break touch as you said and attach entire log please: `/var/log/asus_touchpad_numpad-driver/error.log`

ldrahnik, 2023-02-09:
@kramarov-evg Which driver you have installed for Touchpad?

kramarov-evg, 2023-02-09:
@ldrahnik, I use Linux Mint 21 with kernael version 5.15.0-58-generic. I didn't install any extra drivers for touchpad except yours. Regarding the error log, here it is:
```
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140A:00 04F3:3134 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input30
INFO:asus-touchpad-numpad-driver:Set touchpad id 12 from H: Handlers=mouse1 event12
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-1972
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-1892
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1526 y 979
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3364 y 1039
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2722 y 1188
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2722 y 1183
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3781 y 195
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1675904142.2811143
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3801 y 186
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.7589921951293945
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Un-touched top_right_icon (numlock) in time: 1675904144.0402544
INFO:asus-touchpad-numpad-driver:Numpad activated
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2031 y 941
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2030 y 915
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP6:77
INFO:asus-touchpad-numpad-driver:grab current slot
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP6:77
INFO:asus-touchpad-numpad-driver:un-grab current slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1091 y 1370
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1090 y 1365
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP2:80
INFO:asus-touchpad-numpad-driver:grab current slot
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP2:80
INFO:asus-touchpad-numpad-driver:un-grab current slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1890 y 986
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1896 y 934
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP6:77
INFO:asus-touchpad-numpad-driver:grab current slot
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP6:77
INFO:asus-touchpad-numpad-driver:un-grab current slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2019 y 1430
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2017 y 1423
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP3:81
INFO:asus-touchpad-numpad-driver:grab current slot
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_KP3:81
INFO:asus-touchpad-numpad-driver:un-grab current slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3804 y 123
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1675904146.8174706
INFO:asus-touchpad-numpad-driver:Ended existing slot
[... 50 lines cut ...]
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0053071975708008
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:1
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0048868656158447
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:2
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0032546520233154
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:3
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 110 y 179
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3856 y 64
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1675904163.3047237
[... cut ...]
INFO:asus-touchpad-numpad-driver:Activation time: 1
INFO:asus-touchpad-numpad-driver:Un-touched top_right_icon (numlock) in time: 1675904165.232194
INFO:asus-touchpad-numpad-driver:Numpad deactivated
```
Actually, I've just noticed `Touched top_left_icon in time ...`  records in the log. And it just seems, that I didn't wait long enough for brightness to change. It just takes significantl longer, than it does in Windows system, that's why I first thought it didn't work at all, but that's ok. The only issue left is non-functional touch as click

ldrahnik, 2023-02-09:
@kramarov-evg Time to brightness change when you hold top left icon can be changed in config, default is 1s.

ldrahnik, 2023-02-09:
@kramarov-evg Try to find out settings as I have on this screenshot *Tapping: Tap to click*

ldrahnik, 2023-02-09:
@kramarov-evg

[image]

closed 2023-02-09 by ldrahnik