asus-linux-drivers/asus-numberpad-driver #92: [Bug]: Failure when setting activation_time to a decimal number

misolasan, 2023-02-15:
### Describe the bug

When I set activation_time to 0.1, it breaks down the entire system - after a login, it just freezes.

### Expected behavior

I would like to be able to set activation time even to a decimal number, one second just seems too impractical. I also noticed that the config file is constantly overwritten which is hardly an efficient practice.

### Desktop

Ubuntu 22.04.1 LTS
5.15.0-60-generic

ldrahnik, 2023-02-16:
Hello @misolasan,

> When I set activation_time to 0.1, it breaks down the entire system - after a login, it just freezes. Immediately after setting this value, the config file stops being overwritten constantly which hints to an error.

1s is by default set up for activation to avoid random enabling the NumberPad with palm or finger (in my experience easy to let finger by mistake on the top right corner for ms). For faster activation is recommended to use the slide function from top right corner to center, of Touchpad, by default has to be exceeded with the finger at least 30% of the touchpad width and height before is finger released. Does not solve this approaches your issue? Anyway int changed to float for config values (tested with mentioned values, nothing borderline like 0.01 etc.)

```
disable_due_inactivity_time = 4.5
activation_time = 0.5
top_left_icon_activation_time = 1.5
```

Let me know if it works for you also.

> I also noticed that the config file is constantly overwritten which is hardly an efficient practice.

Created [issue](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/93). Feel free to suggest in created issue a concrete more efficient way.

misolasan, 2023-02-17:
Hi, in the meantime, I switched to Mohamed's implementation of this numpad, it worked right away just like I like and also without any configuration files. For me personally, 1 second is too long and bothersome for practical usage, his implementation reacts to any touch, I do not mind accidental turning on, I almost never reach corners with my fingers and I do not play games with the touchpad.

As for the constant config overwriting, I really do not understand how it is possible that someone who managed to write such a complex piece of sw (I suppose you wrote it, really, hat off to you) then proceeded to constantly overwrite the config file. If the whole kernel or Gnome was written this way, it probably would not even run and just turn the PC into a short-lasting heater while frying CPU and disc. I would suggest that since you probably know the implementation better than me, you could also be more capable of making it reasonable - why should not it be sufficient to read the config once and require logging off after any change to it, just like Gnome does it, for example?

ldrahnik, 2023-02-17:
@misolasan Thank you for letting me know. This driver written in python is still temporary, is need to have a written kernel driver for linux as even one of the feature request in this repo [says](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/45) and is waiting for time (mine or someone else). Anyway as I write meanwhile I re-implemented checking only changes in config file using inotify (wrapper for python is pynotify, inotify is considered as one of best practices when waiting for changes in any file in linux distros), which was indented to overwrite since development of driver for NumberPad continued in this repository (basic code is derived from mohamad repo because maintainer does not maintained repo anymore), basic endless while true cycle in the thread was temporary because of lack of my time and atm does not exist anymore, [commit](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/commit/b03e829811dac0aed05ecb87306257e04dc0476a5).

Feel free to test it and let me know if it works as you want and I am open to any other ideas from using. As you write it looks like you are not familiar a lot with linux yet but I recommend you use this driver as you will face a lot of bug/problems with driver in mohamed repo when you will be setting up more advanced things in your distro. Even not advanced, is enough for example add more keyboard layouts and mohamed driver will stop working properly. And I do not count features etc.

Btw configuration files are not your enemy, every well written driver contains/should offer configurations because everyone is looking for customization for own needs and default config file settings should be for the majority of users.

So as is now possible set up float to all 3 variables associated with activation as I mentioned in comment above and changes are already in master branch, the same as is re-implemented watching only changes in config file and is read only once and is rewritten only when some value is changed by using driver/NumberPad I will close this issue as solved. I can only mention more about for example driver for your keyboard (concretely backlight level of keyboard) do the same, is changed config file, in this concrete case one file everytime when you change backlight via shortcut and so on as every driver in your distro does. Only difference is in kernel drivers exist more files for each value and not one usually but everyone writes drivers for linux by own way. In this case I considered is good option have one config file and when will be driver reimplemented from python to c++/c one file per config value. The issue can be reopened.

closed 2023-02-17 by ldrahnik