asus-linux-drivers/asus-numberpad-driver #95: [Bug]: not working for UX582LR

mbrouillet, 2023-02-22:
### Describe the bug

I installed it in my home (for now) and there is no apparent error during install. I chose the closest match UX581LR but I have a UX582LR.
I hit the top line of the keypad (there is a printed numpad) and 

### Expected behavior

I believe when touching the right corner of the touchpad a numeric keypad would light up and I would be able to type in numbers. Wherever I tap, long-press, double-tap, triple-tap on the top of the keypad, nothing alike happens.

### Relevant log output

```shell
According to the Troubleshooting section of the README.md:

marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ LOG=DEBUG sudo -E ./asus_touchpad.py "ux581l" ""
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
DEBUG:asus-touchpad-numpad-driver:Writting to config file: "<_io.TextIOWrapper name='asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"

[nothing else, program keeps running, no additional line whatever I do on the touchpad]
```

### Desktop

- Program version : fresh from Git as of today, using the instructions in README.md
- Distro Ubuntu Studio 22.04 : No LSB modules are available. Distributor ID: Ubuntu; Description: Ubuntu 22.04.2 LTS; Release: 22.04; Codename: jammy
- Kernel : Linux ux582lr 5.15.0-60-lowlatency #66-Ubuntu SMP PREEMPT Tue Jan 24 14:26:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
- apt update and upgrade done.

mbrouillet, 2023-02-22:
Note : I have bios set up to have F1 F2 F3… instead of volume and brightness keys (which I control using the function key). Would that make a difference ?

ldrahnik, 2023-02-22:
@mbrouillet Please attach also the entire output of `sudo ./install.sh` (first run `sudo ./uninstall.sh`), then try to activate NumberPad again a post to me also log file located here `/var/log/asus_touchpad_numpad-driver/error.log`

mbrouillet, 2023-02-22:
Thank you. Interestingly it tells me a layout file is locally modified, and that's surprising because the directory is fresh from this morning. Second attempt (expectedly) does not tell warning. Then there is a Curl issue.

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./uninstall.sh
[sudo] password for marcel: 
Removed /etc/systemd/system/default.target.wants/asus_touchpad_numpad.service.
Installed numpad layouts contain modifications compared to the default ones. Do you want remove them [y/N]y
Uninstall finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./uninstall.sh
Uninstall finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./install.sh
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libevdev2 is already the newest version (1.12.1+dfsg-1).
i2c-tools is already the newest version (4.3-2build1).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: libevdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.21.5)
Requirement already satisfied: inotify in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: nose in /home/marcel/.local/lib/python3.10/site-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ZenBook Pro Duo UX582LR_UX582LR
curl: (3) URL using bad/illegal format or missing URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 83132    0 83132    0     0  71170      0 --:--:--  0:00:01 --:--:-- 71174
Could not automatically detect numpad layout for your laptop. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).

Select your model keypad layout:
 1) b7402.py
 2) e210ma.py
 3) g513.py
 4) g533.py
 5) gx551.py
 6) gx701.py
 7) up5401ea.py
 8) ux433fa.py
 9) ux581l.py
10) Quit
Please enter your choice 9
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
Wayland or X11 is not detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Created symlink /etc/systemd/system/default.target.wants/asus_touchpad_numpad.service → /etc/systemd/system/asus_touchpad_numpad.service.
Asus touchpad service enabled
Asus touchpad service started
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).
Install finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

ldrahnik, 2023-02-22:
@mbrouillet Please attach content of file `$ cat /var/log/asus_touchpad_numpad-driver/error.log` and  `$ curl --version`

ldrahnik, 2023-02-22:
@mbrouillet Layout for your laptop will be now recommended

ldrahnik, 2023-02-22:
@mbrouillet What is output for `$ echo $XDG_SESSION_TYPE`? Because during install process is written `Wayland or X11 is not detected`, so implemented way in the driver for detection xorg or wayland does not work for you, in case you use one of them, do you?

ldrahnik, 2023-02-22:
@mbrouillet In case you have doubts that the difference should be zero you can run `diff` command manually in touchpad numpad git repo root `$ diff numpad_layouts /usr/share/asus_touchpad_numpad-driver/numpad_layouts/`. But first remove `rm -rf /usr/share/asus_touchpad_numpad-driver/numpad_layouts/__pycache__`, `uninstall` script does the same before comparison.

ldrahnik, 2023-02-22:
@mbrouillet Question during uninstalling about diff numpad layouts was displayed because was taken into account even temp folder: `__pycache__`, in location `/usr/share/asus_touchpad_numpad-driver/numpad_layouts/` was removed before comparison but not inside the root of cloned git repository (pushed master)

mbrouillet, 2023-02-22:
> @mbrouillet Please attach content of file `$ cat /var/log/asus_touchpad_numpad-driver/error.log` and `$ curl --version`

Apologies for forgetting that part. Here it is. 
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ curl --version
curl 7.81.0 (x86_64-pc-linux-gnu) libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.13
Release-Date: 2022-01-05
Protocols: dict file ftp ftps gopher gophers http https imap imaps ldap ldaps mqtt pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: alt-svc AsynchDNS brotli GSS-API HSTS HTTP2 HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM NTLM_WB PSL SPNEGO SSL TLS-SRP UnixSockets zstd
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ cat /var/log/asus_touchpad_numpad-driver/error.log
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 16, in <module>
    from evdev import InputDevice, ecodes as ecodess
ModuleNotFoundError: No module named 'evdev'
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```
Here are the relevant output for evdev, I think (there's a lot more… I selected)
```
libevdev-dev/jammy 1.12.1+dfsg-1 amd64
  wrapper library for evdev devices - development files

libevdev-doc/jammy,jammy 1.12.1+dfsg-1 all
  wrapper library for evdev devices - development docs

libevdev-tools/jammy 1.12.1+dfsg-1 amd64
  wrapper library for evdev devices - tools

libevdev2/jammy,now 1.12.1+dfsg-1 amd64 [installed]
  wrapper library for evdev devices

libevdevplus-dev/jammy 0.1.1-2 amd64
  Development files for the libevdevplus library
```

ldrahnik, 2023-02-22:
@mbrouillet Run manually and attach result: `python3 -m pip install -r requirements.txt` in root of cloned git repository

mbrouillet, 2023-02-22:
> What is output for `$ echo $XDG_SESSION_TYPE`?
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ echo $XDG_SESSION_TYPE
x11
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

BTW, I reinstalled my full system 10 days ago, using a Ubuntu Studio usb stick. I was on Ubuntu Studio 20.xx (non-LTS) + updates, but there was no migration to 22.LTS so I re-installed. BUT I did keep all my .xyz in my home (I rsynced back my entire /home/marcel). I noticed this was not a great idea and ended up removing both .config/plasma* and .kde. Just so you know where my system comes from.

mbrouillet, 2023-02-22:
> Run manually in root of cloned git repository and attach result: `python3 -m pip install -r requirements.txt`

I hope I did the right command to retrieve your commits on main branch. If not, please correct me.
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ git pull
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 12 (delta 6), reused 12 (delta 6), pack-reused 0
Unpacking objects: 100% (12/12), 2.91 KiB | 372.00 KiB/s, done.
From https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver
   b7f9135..5f3f0f3  master     -> origin/master
Updating b7f9135..5f3f0f3
Fast-forward
 install.sh            | 29 +++++++++++++++++++++++++++++
 laptop_numpad_layouts |  4 +++-
 uninstall.sh          |  9 ++-------
 3 files changed, 34 insertions(+), 8 deletions(-)
marcel@ux582lr:~/asus-touchpad-numpad-driver$ python3 -m pip install -r requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: libevdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.21.5)
Requirement already satisfied: inotify in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: nose in /home/marcel/.local/lib/python3.10/site-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

ldrahnik, 2023-02-22:
@mbrouillet Please run manually `loginctl | grep 'marcel' | awk '{print $1}'` and then with inserted output from previous command to this one `loginctl show-session <something like string `c2` - output from previous command>`

mbrouillet, 2023-02-22:
How do I tell if the deamon is running ?  At the end of this command, I only see the drivers for the WMI (which, I understand, are not for my laptop and should preferably be uninstalled).
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./install.sh 
[sudo] password for marcel: 
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libevdev2 is already the newest version (1.12.1+dfsg-1).
i2c-tools is already the newest version (4.3-2build1).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: libevdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.21.5)
Requirement already satisfied: inotify in /home/marcel/.local/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: nose in /home/marcel/.local/lib/python3.10/site-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ZenBook Pro Duo UX582LR_UX582LR
Automatically recommended numpad layout: ux581l (associated to UX582LR_UX582LR). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
Wayland or X11 is not detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Asus touchpad service enabled
Asus touchpad service started
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).
Install finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$ pstree -pau | grep asus
  |-python3,98850 /usr/share/asus_wmi_hotkeys-driver/asus_wmi_hotkeys.py fa507r
  |   |       |   |   |   |   |-grep,117632 --color=auto asus
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

ldrahnik, 2023-02-22:
@mbrouillet `sudo systemctl status asus_touchpad_numpad.service`

mbrouillet, 2023-02-22:
> Please run manually `loginctl | grep 'marcel' | awk '{print $1}'` and then with inserted output from previous command to this one `loginctl show-session <something like string `c2` - output from previous command>`

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ loginctl | grep 'marcel' | awk '{print $1}'
124
3
5
8
marcel@ux582lr:~/asus-touchpad-numpad-driver$ loginctl show-session 124
Id=124
User=1000
Name=marcel
Timestamp=Tue 2023-02-21 17:55:25 CET
TimestampMonotonic=60535998255
VTNr=1
Seat=seat0
Display=:0
Remote=no
Service=sddm
Desktop=KDE
Scope=session-124.scope
Leader=79312
Audit=124
Type=x11
Class=user
Active=yes
State=active
IdleHint=no
IdleSinceHint=0
IdleSinceHintMonotonic=0
LockedHint=no
marcel@ux582lr:~/asus-touchpad-numpad-driver$ loginctl show-session 3
Id=3
User=1000
Name=marcel
Timestamp=Mon 2023-02-20 13:41:15 CET
TimestampMonotonic=44339185
VTNr=2
Seat=seat0
TTY=tty2
Remote=no
Service=login
Scope=session-3.scope
Leader=2212
Audit=3
Type=tty
Class=user
Active=no
State=online
IdleHint=yes
IdleSinceHint=1676896968203082
IdleSinceHintMonotonic=0
LockedHint=no
marcel@ux582lr:~/asus-touchpad-numpad-driver$ loginctl show-session 5
Id=5
User=1000
Name=marcel
Timestamp=Mon 2023-02-20 13:43:57 CET
TimestampMonotonic=206679267
VTNr=1
Seat=seat0
Display=:0
Remote=no
Service=sddm
Desktop=KDE
Scope=session-5.scope
Leader=2414
Audit=5
Type=x11
Class=user
Active=no
State=closing
IdleHint=no
IdleSinceHint=0
IdleSinceHintMonotonic=0
LockedHint=no
marcel@ux582lr:~/asus-touchpad-numpad-driver$ loginctl show-session 8
Id=8
User=1000
Name=marcel
Timestamp=Mon 2023-02-20 13:46:55 CET
TimestampMonotonic=384309056
VTNr=1
Seat=seat0
Display=:0
Remote=no
Service=sddm
Desktop=KDE
Scope=session-8.scope
Leader=3397
Audit=8
Type=x11
Class=user
Active=no
State=closing
IdleHint=no
IdleSinceHint=0
IdleSinceHintMonotonic=0
LockedHint=no
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

mbrouillet, 2023-02-22:
> sudo systemctl status asus_touchpad_numpad.service

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl status asus_touchpad_numpad.service
× asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-02-22 13:05:36 CET; 5min ago
    Process: 117626 ExecStart=/usr/bin/env python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py ux581l /usr/share/asus_touchpad_numpad-driver/ (code=exited, status=1/FAILURE)
   Main PID: 117626 (code=exited, status=1/FAILURE)
        CPU: 127ms

Feb 22 13:05:36 ux582lr systemd[1]: asus_touchpad_numpad.service: Scheduled restart job, restart counter is at 5.
Feb 22 13:05:36 ux582lr systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Feb 22 13:05:36 ux582lr systemd[1]: asus_touchpad_numpad.service: Start request repeated too quickly.
Feb 22 13:05:36 ux582lr systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
Feb 22 13:05:36 ux582lr systemd[1]: Failed to start Asus Touchpad to Numpad Handler.
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

mbrouillet, 2023-02-22:
Note : I usually run a non standard keyboard layout (fr bepo). I did « setxkbmap us ; sudo ./install.sh ; sudo systemctl status asus_touchpad_numpad.service » and there is no difference with the above.

ldrahnik, 2023-02-22:
@mbrouillet `ModuleNotFoundError: No module named 'evdev'` (pushed to master), let me know it fixed installation and NumberPad works. Created feature request https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/96.

mbrouillet, 2023-02-22:
> Please `git pull` changes and run again `sudo ./install.sh` and attach output here

The output is quite different, X11 is detected, one of the i2c probes fails, and the XF86Calculator option fails (maybe as a result of « No such schema “org.gnome.settings-daemon.plugins.media-keys” » ; reminder, I am under KDE.

The daemon is running, but no matter what I do with the touchpad **it stays as a touchpad**. Tapping, swiping from the corner… no effect. I don't have a NumLock key on my keyboard, but can I emulate it ?

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ git pull
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 6 (delta 2), reused 6 (delta 2), pack-reused 0
Unpacking objects: 100% (6/6), 5.47 KiB | 2.73 MiB/s, done.
From https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver
   5f3f0f3..7e67d51  master     -> origin/master
Updating 5f3f0f3..7e67d51
Fast-forward
 install.sh | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./install.sh 
[sudo] password for marcel: 
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libevdev2 is already the newest version (1.12.1+dfsg-1).
i2c-tools is already the newest version (4.3-2build1).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput is already the newest version (1.6.3-1build2).
xinput set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Requirement already satisfied: libevdev in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (0.5)
Collecting evdev
  Downloading evdev-1.6.1.tar.gz (26 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.21.5)
Collecting inotify
  Downloading inotify-0.2.10.tar.gz (9.9 kB)
  Preparing metadata (setup.py) ... done
Collecting nose
  Downloading nose-1.3.7-py3-none-any.whl (154 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 154.7/154.7 KB 2.4 MB/s eta 0:00:00
Building wheels for collected packages: evdev, inotify
  Building wheel for evdev (setup.py) ... done
  Created wheel for evdev: filename=evdev-1.6.1-cp310-cp310-linux_x86_64.whl size=83754 sha256=a607fd6d9ddfddc80a3d9bf8aebd00be1c98915cbc96ec2add5b7e9a13a58e99
  Stored in directory: /root/.cache/pip/wheels/bb/6e/51/0c0f5677bd58c532c42e13e7d93803fbf8d5d3bf5bfc5b0220
  Building wheel for inotify (setup.py) ... done
  Created wheel for inotify: filename=inotify-0.2.10-py3-none-any.whl size=13446 sha256=30001db8d2532e30ecf3e1e7a11a48e86825a5d3ce83e22c797841ac64c58892
  Stored in directory: /root/.cache/pip/wheels/10/35/5e/ceacd5d1f4164b2034cbaae006f9323d177e57a93a7633d43b
Successfully built evdev inotify
Installing collected packages: nose, evdev, inotify
Successfully installed evdev-1.6.1 inotify-0.2.10 nose-1.3.7
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ZenBook Pro Duo UX582LR_UX582LR
Automatically recommended numpad layout: ux581l (associated to UX582LR_UX582LR). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad.service → /etc/systemd/system/asus_touchpad_numpad.service.
Asus touchpad service enabled
Asus touchpad service started
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).
Install finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl status asus_touchpad_numpad.service
● asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/etc/systemd/system/asus_touchpad_numpad.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-02-22 17:30:52 CET; 53s ago
   Main PID: 135122 (python3)
      Tasks: 5 (limit: 38095)
     Memory: 24.0M
        CPU: 699ms
     CGroup: /system.slice/asus_touchpad_numpad.service
             └─135122 python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py ux581l /usr/share/asus_touchpad_numpad-driver/

Feb 22 17:30:52 ux582lr systemd[1]: Started Asus Touchpad to Numpad Handler.
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

mbrouillet, 2023-02-22:
> your case (when is more running sessions under the current user)

I have no idea what this is. Either a specific thing of KDE (I see some are related to SDDM which, I thought, handles the login screen. Also possible that this relates to my four virtual desktops ? Just trying to help, but no need to push this further. Only a comment.

ldrahnik, 2023-02-22:
@mbrouillet Attach please log file `/var/log/asus_touchpad_numpad-driver/error.log` or again try to run with DEBUG with disabled service

ldrahnik, 2023-02-22:
@mbrouillet Warning about pip is waiting for the possibility to install driver only for the current user [here](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/96)

ldrahnik, 2023-02-22:
@mbrouillet Curl warning can be ignored (atm will not be even displayed for you as your laptop I have added to static file), created issue https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/97

ldrahnik, 2023-02-22:
@mbrouillet Numlock key can be emulated for example via `xdotool key Num_Lock` but by default will be activated NumberPad only disabled when will be disabled system numlock, you have to change in config this line `sys_numlock_enables_numpad=1` and then should be NumberPad activated when running `xdotool key Num_Lock`

ldrahnik, 2023-02-22:
@mbrouillet Do you use/Ubuntu 22.04 `dconf`? In case yes please install if you do not have and take o look where you change shortcuts, it should be something similar to this: `org.gnome.settings-daemon.plugins.media-keys custom-keybindings`

mbrouillet, 2023-02-22:
> Attach please log files, first one from installed service `/var/log/asus_touchpad_numpad-driver/error.log` and again try to run with DEBUG with disabled service

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo rm /var/log/asus_touchpad_numpad-driver/error.log
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl start asus_touchpad_numpad.service
```
I tried all sorts of tapping on the touchpad... then
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ cat /var/log/asus_touchpad_numpad-driver/error.log
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
marcel@ux582lr:~/asus-touchpad-numpad-driver$ LOG=DEBUG sudo -E ./asus_touchpad.py "ux581l" ""
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
DEBUG:asus-touchpad-numpad-driver:Writting to config file: "<_io.TextIOWrapper name='asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"
```
I tried all sorts of tapping on the touchpad... then Ctrl-C twice
```
^CERROR:root:Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/home/marcel/asus-touchpad-numpad-driver/./asus_touchpad.py", line 1458, in <module>
    listen_touchpad_events()
  File "/home/marcel/asus-touchpad-numpad-driver/./asus_touchpad.py", line 1184, in listen_touchpad_events
    for e in d_t.events():
  File "/home/marcel/.local/lib/python3.10/site-packages/libevdev/device.py", line 536, in events
    ev = self._libevdev.next_event(flags)
  File "/home/marcel/.local/lib/python3.10/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
KeyboardInterrupt
^CException ignored in: <module 'threading' from '/usr/lib/python3.10/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1567, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

mbrouillet, 2023-02-22:
> Numlock key can be emulated for example via `xdotool key Num_Lock` but by default will be activated NumberPad only disabled when will be disabled system numlock, you have to change in config this [...]

I changed the config to 1 in /usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev. **Is that the correct location ?**
Then I restarted the service, installed xdotool and ran (as normal user) `xdotool key Num_Lock` and the touchpad remained a touchpad.

ldrahnik, 2023-02-22:
@mbrouillet Yes, location is correct and log says?

mbrouillet, 2023-02-22:
> Do you use/Ubuntu 22.04 `dconf`? In case yes please install if you do not have and take a look where you change shortcuts, it should be something similar to this: `org.gnome.settings-daemon.plugins.media-keys custom-keybindings`. In case no anyway let me know how can be shorcut added automatically, I could add that to the driver

I am not familiar with dconf. I found the following command and searched for part of the string you indicate, unsuccessfully. I have a process running `/usr/libexec/dconf-service` and the following return data. Does it tell that I am running dconf ?

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ gsettings list-recursively | grep plugins
org.entangle-photo.manager.interface plugins @as []
marcel@ux582lr:~/asus-touchpad-numpad-driver$ gsettings list-recursively | grep media
org.gnome.desktop.media-handling automount true
org.gnome.desktop.media-handling automount-open true
org.gnome.desktop.media-handling autorun-never false
org.gnome.desktop.media-handling autorun-x-content-ignore @as []
org.gnome.desktop.media-handling autorun-x-content-open-folder @as []
org.gnome.desktop.media-handling autorun-x-content-start-app ['x-content/unix-software', 'x-content/ostree-repository']
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

How can I find out _how to add a shortcut_. I'm a little lost here. But very willing to help.

mbrouillet, 2023-02-22:
> @mbrouillet Yes, location is correct and log says?

See above screenshot (`cat /var/log/asus_touchpad_numpad-driver/error.log`) with 7 lines only.

ldrahnik, 2023-02-22:
@mbrouillet Pull latest master, run DEBUG and attach log file, might be problem with implementation of locks in driver - deadlock (two places in code are waiting for the same semaphore)

mbrouillet, 2023-02-22:
> @mbrouillet Please pull latest master, run DEBUG and attach appropriate log file (in case argument is "" local one), might be problem with implementation of locks in driver - deadlock (two places in code are waiting for the same semaphore)

Here are my sequences.
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ git pull
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), 586 bytes | 293.00 KiB/s, done.
From https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver
   7e67d51..46a7621  master     -> origin/master
Updating 7e67d51..46a7621
Fast-forward
 asus_touchpad.py | 12 ++++++++++++
 1 file changed, 12 insertions(+)
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo ./install.sh 
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libevdev2 is already the newest version (1.12.1+dfsg-1).
i2c-tools is already the newest version (4.3-2build1).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput is already the newest version (1.6.3-1build2).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Requirement already satisfied: libevdev in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (0.5)
Requirement already satisfied: evdev in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.21.5)
Requirement already satisfied: inotify in /usr/local/lib/python3.10/dist-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: nose in /usr/local/lib/python3.10/dist-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
Detected laptop: ZenBook Pro Duo UX582LR_UX582LR
Automatically recommended numpad layout: ux581l (associated to UX582LR_UX582LR). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Asus touchpad service enabled
Asus touchpad service started
In system remains config file from previous installation. Do you want replace this with default config file [y/N]y
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).
Install finished
marcel@ux582lr:~/asus-touchpad-numpad-driver$ cat /var/log/asus_touchpad_numpad-driver/error.log
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "enabled" with value: "True"
[... 98 lines cut ...]
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire will be called
[... cut ...]
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

ldrahnik, 2023-02-22:
@mbrouillet Also please pull latest master now and try `sudo python3 ./tests/test_brightness.py`, let it finish succesfully, something around 30s

mbrouillet, 2023-02-22:
> @mbrouillet Also please pull latest master now and try `sudo python3 ./tests/test_brightness.py`, let it finish succesfully, something around 30s

I had pulled from git above (at beginning of message where I say « Here are my sequences »), and apparently am up to date for the above message already.

The brightness test took much more than 30 seconds. Here it is : 
```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo rm /var/log/asus_touchpad_numpad-driver/error.log
rm: cannot remove '/var/log/asus_touchpad_numpad-driver/error.log': No such file or directory
marcel@ux582lr:~/asus-touchpad-numpad-driver$ git pull
Already up to date.
marcel@ux582lr:~/asus-touchpad-numpad-driver$ time sudo python3 ./tests/test_brightness.py
Hight bright
1 0x1
Error: Sending messages failed: Remote I/O error
11 0xb
Error: Sending messages failed: Remote I/O error
81 0x51
Error: Sending messages failed: Remote I/O error
131 0x83
Error: Sending messages failed: Remote I/O error
161 0xa1
Error: Sending messages failed: Remote I/O error
181 0xb5
Error: Sending messages failed: Remote I/O error
231 0xe7
Error: Sending messages failed: Remote I/O error
241 0xf1
Error: Sending messages failed: Remote I/O error
middle bright
21 0x15
Error: Sending messages failed: Remote I/O error
41 0x29
Error: Sending messages failed: Remote I/O error
51 0x33
Error: Sending messages failed: Remote I/O error
61 0x3d
Error: Sending messages failed: Remote I/O error
71 0x47
Error: Sending messages failed: Remote I/O error
91 0x5b
Error: Sending messages failed: Remote I/O error
101 0x65
Error: Sending messages failed: Remote I/O error
111 0x6f
Error: Sending messages failed: Remote I/O error
121 0x79
Error: Sending messages failed: Remote I/O error
141 0x8d
Error: Sending messages failed: Remote I/O error
151 0x97
Error: Sending messages failed: Remote I/O error
171 0xab
Error: Sending messages failed: Remote I/O error
201 0xc9
Error: Sending messages failed: Remote I/O error
low bright
31 0x1f
Error: Sending messages failed: Remote I/O error
Error: Sending messages failed: Remote I/O error

real	3m40.312s
user	0m0.004s
sys	0m0.008s
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

ldrahnik, 2023-02-22:
@mbrouillet Happened at least one time that the backlight of NumberPad was enabled?

mbrouillet, 2023-02-22:
> @mbrouillet Happened at least one time that the backlight of NumberPad was enabled?

No, unfortunately. Let me try again, with the keyboard backlight on to max, if that could matter. Hold on 4 minutes.

ldrahnik, 2023-02-22:
@mbrouillet You do not have to do it.

ldrahnik, 2023-02-22:
@mbrouillet Attach please output of `sudo libinput debug-events`

mbrouillet, 2023-02-22:
> @mbrouillet Attach please output of `sudo libinput debug-events`

```

@mbrouillet Attach please output of sudo libinput debug-events
```
`libinput-bin` is installed, and `libinput10` also. Do I need `libinput-tools` ? or `libinput-dev` ?

ldrahnik, 2023-02-22:
@mbrouillet Let try it works for you without brightness, in file `asus_touchpad.py` comment function `send_value_to_touchpad_via_i2c` like this:

```
def send_value_to_touchpad_via_i2c(value):
    global device_id

    cmd = ["i2ctransfer", "-f", "-y", device_id, "w13@0x15", "0x05", "0x00", "0x3d", "0x03", "0x06", "0x00", "0x07", "0x00", "0x0d", "0x14", "0x03", value, "0xad"]

    #try:
        #subprocess.call(cmd)
    #except subprocess.CalledProcessError as e:
    #    log.error('Error during sending via i2c: \"%s\"', e.output)
```

And then with DEBUG I hope you should be able use NumberPad

ldrahnik, 2023-02-22:
@mbrouillet On Ubuntu `sudo apt-get install libinput-tools`

mbrouillet, 2023-02-22:
> @mbrouillet Attach please output of `sudo libinput debug-events`

I installed the libinput-tools and here is the output. Note that I first moved the mouse, and then understood it was spitting info about the inputs, so at the end of the output below I was tapping the top right corner of the touchpad.

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo libinput debug-events
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event19  DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event20  DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED            Power Button                      seat0 default group4  cap:k
-event260  DEVICE_ADDED            USB AUDIO DEVICE                  seat0 default group5  cap:k
-event5   DEVICE_ADDED            USB2.0 HD UVC WebCam: USB2.0 HD   seat0 default group6  cap:k
-event6   DEVICE_ADDED            USB2.0 HD UVC WebCam: USB2.0 IR   seat0 default group6  cap:k
-event16  DEVICE_ADDED            ELAN1406:00 04F3:3101 Mouse       seat0 default group7  cap:p left scroll-nat scroll-button
-event17  DEVICE_ADDED            ELAN1406:00 04F3:3101 Touchpad    seat0 default group7  cap:pg  size 58x81mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
-event18  DEVICE_ADDED            ELAN1406:00 04F3:3101 Keyboard    seat0 default group7  cap:k
-event7   DEVICE_ADDED            ELAN9008:00 04F3:2C56             seat0 default group8  cap:t  size 332x188mm ntouches 10 calib
-event10  DEVICE_ADDED            ELAN9008:00 04F3:2C56 Stylus      seat0 default group8  cap:T  size 343x194mm calib
-event14  DEVICE_ADDED            ELAN9009:00 04F3:2C23             seat0 default group9  cap:t  size 332x96mm ntouches 5 calib
-event26  DEVICE_ADDED            ELAN9009:00 04F3:2C23 Stylus      seat0 default group9  cap:T  size 343x98mm calib
-event27  DEVICE_ADDED            ELAN9009:00 04F3:2C23 Touchpad    seat0 default group9  cap:pg  size 332x96mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
-event4   DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group10 cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group11 cap:k
-event258  DEVICE_ADDED            M585/M590 Keyboard                seat0 default group12 cap:k
-event259  DEVICE_ADDED            M585/M590 Mouse                   seat0 default group12 cap:p left scroll-nat scroll-button
 event259  POINTER_MOTION          +0.000s	  0.31/  0.31 ( +1.00/ +1.00)
 event259  POINTER_MOTION          +0.000s	  6.88/  6.88 ( +4.00/ +4.00)
 event259  POINTER_MOTION          +0.000s	 18.00/ 20.00 ( +9.00/+10.00)
 event259  POINTER_MOTION          +0.000s	 38.00/ 50.00 (+19.00/+25.00)
 event259  POINTER_MOTION          +0.007s	 74.00/136.00 (+37.00/+68.00)
 event259  POINTER_MOTION          +0.008s	 16.00/ 18.00 ( +8.00/ +9.00)
 event259  POINTER_MOTION          +0.015s	 16.00/ 18.00 ( +8.00/ +9.00)
 event259  POINTER_MOTION          +0.022s	 16.00/ 14.00 ( +8.00/ +7.00)
 event259  POINTER_MOTION          +0.030s	 13.60/  9.72 ( +7.00/ +5.00)
 event259  POINTER_MOTION          +0.037s	 10.57/  8.81 ( +6.00/ +5.00)
 event259  POINTER_MOTION          +0.045s	 11.82/  5.07 ( +7.00/ +3.00)
 event259  POINTER_MOTION          +0.052s	  5.74/  2.87 ( +4.00/ +2.00)
 event259  POINTER_MOTION          +0.060s	  3.46/  2.31 ( +3.00/ +2.00)
 event259  POINTER_MOTION          +0.068s	  2.03/  0.00 ( +2.00/ +0.00)
 event259  POINTER_MOTION          +0.097s	  0.00/  0.94 ( +0.00/ +1.00)
 event259  POINTER_MOTION          +0.135s	  0.60/  0.00 ( +1.00/ +0.00)
 event259  POINTER_MOTION          +0.142s	  0.00/ -0.93 ( +0.00/ -1.00)
 event259  POINTER_MOTION          +0.247s	  0.91/ -0.91 ( +1.00/ -1.00)
 event259  POINTER_MOTION          +0.255s	  3.88/ -3.88 ( +4.00/ -4.00)
 event259  POINTER_MOTION          +0.262s	  9.61/ -9.61 ( +6.00/ -6.00)
 event259  POINTER_MOTION          +0.270s	 17.72/-19.69 ( +9.00/-10.00)
 event259  POINTER_MOTION          +0.277s	  9.68/ -9.68 ( +5.00/ -5.00)
 event259  POINTER_MOTION          +0.285s	 13.44/-19.19 ( +7.00/-10.00)
 event259  POINTER_MOTION          +0.292s	 18.00/-16.00 ( +9.00/ -8.00)
 event259  POINTER_MOTION          +0.300s	 13.89/-11.90 ( +7.00/ -6.00)
 event259  POINTER_MOTION          +0.315s	 17.02/-13.62 (+10.00/ -8.00)
 event259  POINTER_MOTION          +0.322s	 23.00/-17.25 (+12.00/ -9.00)
 event259  POINTER_MOTION          +0.330s	 20.00/-12.00 (+10.00/ -6.00)
 event259  POINTER_MOTION          +0.338s	 20.00/-12.00 (+10.00/ -6.00)
 event259  POINTER_MOTION          +0.345s	 13.60/ -5.83 ( +7.00/ -3.00)
 event259  POINTER_MOTION          +0.352s	 23.46/ -9.78 (+12.00/ -5.00)
 event259  POINTER_MOTION          +0.360s	 32.00/ -8.00 (+16.00/ -4.00)
 event259  POINTER_MOTION          +0.367s	 24.00/ -6.00 (+12.00/ -3.00)
 event259  POINTER_MOTION          +0.375s	 15.68/ -3.92 ( +8.00/ -2.00)
 event259  POINTER_MOTION          +0.382s	 35.28/ -7.84 (+18.00/ -4.00)
 event259  POINTER_MOTION          +0.390s	 22.00/ -4.00 (+11.00/ -2.00)
 event259  POINTER_MOTION          +0.397s	 30.00/ -4.00 (+15.00/ -2.00)
 event259  POINTER_MOTION          +0.405s	 17.85/ -1.98 ( +9.00/ -1.00)
[... cut ...]
 event17  POINTER_MOTION          +7.062s	 -0.72/  0.48 ( -3.00/ +2.00)
 event17  POINTER_MOTION          +7.069s	  0.00/  0.24 ( +0.00/ +1.00)
 event17  POINTER_MOTION          +7.097s	 -0.52/  0.52 ( -2.00/ +2.00)
 event17  POINTER_MOTION          +7.125s	  0.00/  0.44 ( +0.00/ +2.00)
 event17  GESTURE_HOLD_END        +7.174s	1
-event3   KEYBOARD_KEY            +8.946s	*** (-1) pressed
 event3   KEYBOARD_KEY            +10.288s	*** (-1) pressed
^C
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

ldrahnik, 2023-02-22:
@mbrouillet Run command again `sudo libinput debug-events` and which event (on the left) is used when you do something on the Touchpad?

mbrouillet, 2023-02-22:
> Let try it works for you without brightness, in file `asus_touchpad.py` comment function `send_value_to_touchpad_via_i2c`

In the output below, where I write `[Many lines like this]` I was trying to type something on the touchpad, with no success. I placed the focus on another terminal, to make sure I would see if a number got typed. No.

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ git diff asus_touchpad.py
diff --git a/asus_touchpad.py b/asus_touchpad.py
index 57aa886..d918752 100755
--- a/asus_touchpad.py
+++ b/asus_touchpad.py
@@ -136,10 +136,10 @@ def send_value_to_touchpad_via_i2c(value):
 
     cmd = ["i2ctransfer", "-f", "-y", device_id, "w13@0x15", "0x05", "0x00", "0x3d", "0x03", "0x06", "0x00", "0x07", "0x00", "0x0d", "0x14", "0x03", value, "0xad"]
 
-    try:
-        subprocess.call(cmd)
-    except subprocess.CalledProcessError as e:
-        log.error('Error during sending via i2c: \"%s\"', e.output)
+#    try:
+#        subprocess.call(cmd)
+#    except subprocess.CalledProcessError as e:
+#        log.error('Error during sending via i2c: \"%s\"', e.output)
 
 
 def parse_value_from_config(value):
@@ -1470,4 +1470,4 @@ try:
     listen_touchpad_events()
 except:
     logging.exception("Listening touchpad events unexpectedly failed")
-    sys.exit(1)
\ No newline at end of file
+    sys.exit(1)
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo systemctl stop asus_touchpad_numpad.service
marcel@ux582lr:~/asus-touchpad-numpad-driver$ LOG=DEBUG sudo -E ./asus_touchpad.py "ux581l" ""
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:2C23 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 4 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-4/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input48
INFO:asus-touchpad-numpad-driver:Set touchpad id 27 from H: Handlers=mouse4 event27
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
DEBUG:asus-touchpad-numpad-driver:load_all_config_values: config_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:load_all_config_values: config_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:Writting to config file: "<_io.TextIOWrapper name='asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire will be called
[... many lines like this ...]
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire called succesfully
```
Killed it with twice Ctrl-C
```
^CERROR:root:Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/home/marcel/asus-touchpad-numpad-driver/./asus_touchpad.py", line 1470, in <module>
    listen_touchpad_events()
  File "/home/marcel/asus-touchpad-numpad-driver/./asus_touchpad.py", line 1192, in listen_touchpad_events
    for e in d_t.events():
  File "/home/marcel/.local/lib/python3.10/site-packages/libevdev/device.py", line 536, in events
    ev = self._libevdev.next_event(flags)
  File "/home/marcel/.local/lib/python3.10/site-packages/libevdev/_clib.py", line 889, in next_event
    rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
KeyboardInterrupt
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_system_numlock_vs_local: numlock_lock.acquire called succesfully
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire will be called
DEBUG:asus-touchpad-numpad-driver:check_touchpad_status: numlock_lock.acquire called succesfully
^CException ignored in: <module 'threading' from '/usr/lib/python3.10/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1567, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
marcel@ux582lr:~/asus-touchpad-numpad-driver$ cat /var/log/asus_touchpad_numpad-driver/error.log
cat: /var/log/asus_touchpad_numpad-driver/error.log: No such file or directory
marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

mbrouillet, 2023-02-22:
This is the typical sequence when I put my finger for a second or two on the top right corner of the touchpad.

```
-event17  GESTURE_HOLD_BEGIN      +32.317s	1
 event17  POINTER_MOTION          +32.319s	 -1.98/ -0.50 ( -8.00/ -2.00)
 event17  POINTER_MOTION          +32.326s	 -0.32/  0.00 ( -1.00/ +0.00)
 event17  POINTER_MOTION          +32.333s	 -0.32/  0.32 ( -1.00/ +1.00)
 event17  POINTER_MOTION          +32.340s	 -0.68/  0.34 ( -2.00/ +1.00)
 event17  POINTER_MOTION          +32.354s	 -0.32/  0.32 ( -1.00/ +1.00)
 event17  POINTER_MOTION          +32.361s	 -0.29/  0.29 ( -1.00/ +1.00)
 event17  POINTER_MOTION          +32.368s	 -0.68/  0.34 ( -2.00/ +1.00)
 event17  GESTURE_HOLD_END        +32.368s	1 cancelled
 event17  POINTER_MOTION          +32.375s	 -0.34/  0.34 ( -1.00/ +1.00)
 event17  POINTER_MOTION          +32.396s	 -0.59/  0.30 ( -2.00/ +1.00)
 event17  POINTER_MOTION          +32.417s	 -0.47/  0.00 ( -2.00/ +0.00)
 event17  POINTER_MOTION          +32.459s	 -0.41/  0.20 ( -2.00/ +1.00)
 event17  POINTER_MOTION          +32.501s	 -0.35/  0.00 ( -2.00/ +0.00)
 event17  POINTER_MOTION          +32.586s	 -0.31/  0.00 ( -2.00/ +0.00)
 event17  POINTER_MOTION          +32.817s	  0.27/ -0.13 ( +2.00/ -1.00)
 event17  POINTER_MOTION          +32.979s	  0.25/  0.00 ( +2.00/ +0.00)
 event17  POINTER_MOTION          +33.098s	  0.26/  0.00 ( +2.00/ +0.00)
 event17  POINTER_MOTION          +33.119s	  0.38/ -0.19 ( +2.00/ -1.00)
 event17  POINTER_MOTION          +33.140s	  0.52/ -0.52 ( +2.00/ -2.00)
 event17  POINTER_MOTION          +33.182s	  0.49/ -0.74 ( +2.00/ -3.00)
 event17  POINTER_MOTION          +33.211s	 -0.21/  0.43 ( -1.00/ +2.00)
```

ldrahnik, 2023-02-22:
@mbrouillet Please attach entire `/proc/bus/input/devices`, you have more events associated with Touchpad

mbrouillet, 2023-02-22:
> @mbrouillet Please attach entire `cat /proc/bus/input/devices`, you have more events associated with Touchpad

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo cat /proc/bus/input/devices
I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C09:01/PNP0C0D:00/input/input0
U: Uniq=
H: Handlers=event0 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd event3 leds 
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input24
U: Uniq=
H: Handlers=kbd event19 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:13/LNXVIDEO:01/input/input25
U: Uniq=
H: Handlers=kbd event20 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input26
[... 302 lines cut ...]
I: Bus=0005 Vendor=046d Product=b01b Version=0011
N: Name="M585/M590 Mouse"
P: Phys=64:bc:58:f6:52:15
S: Sysfs=/devices/virtual/misc/uhid/0005:046D:B01B.000E/input/input84
U: Uniq=fe:40:1f:32:02:0c
H: Handlers=mouse7 event259 
B: PROP=0
B: EV=17
B: KEY=ffff0000 0 0 0 0
B: REL=1943
B: MSC=10

I: Bus=0003 Vendor=0c76 Product=1711 Version=0100
N: Name="USB AUDIO DEVICE"
P: Phys=usb-0000:00:14.0-12.2/input3
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12.2/1-12.2:1.3/0003:0C76:1711.000F/input/input86
U: Uniq=
H: Handlers=kbd event260 
B: PROP=0
B: EV=13
B: KEY=7800000000 e000000000000 0
B: MSC=10

marcel@ux582lr:~/asus-touchpad-numpad-driver$
```

mbrouillet, 2023-02-22:
Note : « M585/M590 Keyboard » is a bluetooth mouse.
Also, I see some « Asus WMI hotkeys » and I am confused. I installed the asus-wmi-hotkeys-driver and didn't see a difference so I uninstalled it. Should I install it ? I have not understood what it's used for. My Fn keys work well, and if y toggle them with keeping the « Fn » key while pressing F4 or F5 I raise the brightness of the screen. Changing the volume used to work but is broken now. Maybe broken because of the uninstall ?

mbrouillet, 2023-02-22:
> @mbrouillet Does this enable backlight of Touchpad? `$ sudo i2ctransfer -f -y 4 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad` or this `$ sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad`

One does an error, the other doesn't. **But none brings up the backlight on the touchpad**.

```
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo i2ctransfer -f -y 4 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
Error: Sending messages failed: Remote I/O error
marcel@ux582lr:~/asus-touchpad-numpad-driver$ sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
marcel@ux582lr:~/asus-touchpad-numpad-driver$ 
```

mbrouillet, 2023-02-22:
Hold on ! I tried fiddling with the Fn keys, I changed the brightness of the keyboard, and the key that's next to it (F6) and then touched the touchpad, and the touchpad backlight went on !! it is now on. I don't know how to turn it off.

ldrahnik, 2023-02-22:
@mbrouillet Turn it off with `$ sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad`

mbrouillet, 2023-02-22:
> Try turn it off with `$ sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad`

Yes ! Works. Went off.
The odd thing is that the backlight went on far later than the initial two commands you told me. I was able to type your reply, and play with the Fn keys, before it turned on.

mbrouillet, 2023-02-22:
Now, for some reason,
`sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad` turns it on.
`sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad` turns if off.
Both instantly.

ldrahnik, 2023-02-22:
@mbrouillet Touchpad was sleeping probably. I think how it take into account into the code.

ldrahnik, 2023-02-22:
@mbrouillet Workaround until I will commit better solution is to remove the entire driver, pull latest master, do only this change in `asus_touchpad.py`, line 213: `if touchpad_detected == 0 and ("Name=\"ASUE" in line or "Name=\"ELAN" in line) and "Touchpad" in line:` to `if touchpad_detected == 0 and ("ELAN1406" in line) and "Touchpad" in line:` and should works

mbrouillet, 2023-02-22:
> remove the entire driver,
sudo uninstall ?

ldrahnik, 2023-02-22:
@mbrouillet Yes, uninstall, install, use default config

mbrouillet, 2023-02-22:
Keep this diff or not ?

```
diff --git a/asus_touchpad.py b/asus_touchpad.py
index 57aa886..d918752 100755
--- a/asus_touchpad.py
+++ b/asus_touchpad.py
@@ -136,10 +136,10 @@ def send_value_to_touchpad_via_i2c(value):
 
     cmd = ["i2ctransfer", "-f", "-y", device_id, "w13@0x15", "0x05", "0x00", "0x3d", "0x03", "0x06", "0x00", "0x07", "0x00", "0x0d", "0x14", "0x03", value, "0xad"]
 
-    try:
-        subprocess.call(cmd)
-    except subprocess.CalledProcessError as e:
-        log.error('Error during sending via i2c: \"%s\"', e.output)
+#    try:
+#        subprocess.call(cmd)
+#    except subprocess.CalledProcessError as e:
+#        log.error('Error during sending via i2c: \"%s\"', e.output)
 
 
 def parse_value_from_config(value):
@@ -1470,4 +1470,4 @@ try:
     listen_touchpad_events()
 except:
     logging.exception("Listening touchpad events unexpectedly failed")
-    sys.exit(1)
\ No newline at end of file
+    sys.exit(1)
```

ldrahnik, 2023-02-22:
@mbrouillet Do not keep, only workaround change in the code

[51 more replies]