asus-linux-drivers/asus-numberpad-driver #99: [Bug]: Keyboard layout not working on Wayland : G513

Paradox-AT, 2023-02-24:
### Describe the bug

I've installed the driver on an arch system manually, since it was not available in AUR. After installing, the layout is being enabled successfully, but the keys are not working properly.

While debugging, I got the following logs.

Keys when pressed are being detected properly in the logs but just not working properly you type.

### Expected behavior

Keys when pressed should work properly

### Relevant log output

```shell
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE1403:00 04F3:319A Touchpad
ERROR:asus-touchpad-numpad-driver:Getting Device Enabled via xinput failed
Traceback (most recent call last):
  File "/home/paradox/customInstallSrc/asus-touchpad-numpad-driver/./asus_touchpad.py", line 359, in is_device_enabled
    propData = subprocess.check_output(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2023-02-25:
@Paradox-AT Hello, thank you for creating an issue. The problem is the line `unable to find device ASUE1403:00 04F3:319A Touchpad`, could you please attach the entire output of `sudo ./install.sh`, entire log `$ cat /var/log/asus_touchpad_numpad-driver/error.log`, output of the command `$ sudo libinput debug-events` and `$ cat /proc/bus/input/devices`.

ldrahnik, 2023-02-25:
>Desktop
>Program version [e.g. https://github.com/asus-linux-drivers/asus-touchpad-numpad->driver/commit/57c742abc80d1a502523f07e6a2ec7a5bc24b92c]
>Distro [e.g. Arch Linux]
>Kernel [e.g. 5.18.15.arch1-2]

@Paradox-AT Do you use the latest master?

ldrahnik, 2023-02-25:
@Paradox-AT In case you use Wayland, after will be solved this issue I would appreciate a lot test a branch https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/pull/98 with changes not related to this issue.

ldrahnik, 2023-02-28:
Closed for issue owner inactivity. Can be reopened.

closed 2023-02-28 by ldrahnik