asus-linux-drivers/asus-numberpad-driver #104: Wayland support

ldrahnik, 2023-03-09:
1. Feature "NumberPad will be immediately disabled when is disabled Touchpad": atm is not Wayland supported, mentioned error in log (executing of xinput is in try catch block, should not stop the driver from working), max 9 times is executed xinput, then will be feature (and of course executing xinput) disabled. Mentioned topic on Unix page describes only how get configuration via `libinput list-devices` but not way got get a status enabled/disabled. Or then is device not listed? When you disable Touchpad via special key for example or directly using /sys/device/.. file named enabled, is then Touchpad listed via command `$ sudo libinput list-devices` or not?

2. What is equivalent for this command for wayland: `cmd = ["xinput", "set-prop", touchpad_name, 'libinput Tapping Enabled', str(value)]`

3. How to get remapped keys in Wayland. For example install French keyboard. I installed `French Bepo`. Atm is implemented:

```
import Xlib.display
import Xlib.X
import Xlib.XK
```

```
def get_keycode_of_ascii_char(char):
    display = Xlib.display.Display()
    keysym = Xlib.XK.string_to_keysym(char)
    keycode = display.keysym_to_keycode(keysym) - 8
    return keycode
```

On French keyboard has to be Unicode character send via not SHIFT, LEFT CTRL, U + numbers but SHIFT, LEFT CTRL, S + numbers. Does it works under Wayland? Or what should be used? In case the same way works check can be: print character in layout "%" using French keyboard layout enabled.

Sporknife, 2024-01-07:
Works for me.

ldrahnik, 2024-01-07:
@Sporknife Could you be more explicit, please?

Sporknife, 2024-01-07:
It works for me on opensuse kde+wayland.

ldrahnik, 2024-01-07:
@Sporknife Do you mean point 3? Please try to install the French keyboard and unicodeNumberPad layout and let me know whether works "%" and others, thank you.

Sporknife, 2024-01-07:
wdym by "unicode NumberPad"?

ldrahnik, 2024-01-07:
Select during installing layout with `-unicode` in the end, so keys are sent via Unicode shortcut (left shift + left ctrl + U (on French keyboard it is V, that is the problem) and numbers.

ldrahnik, 2024-01-08:
@Sporknife Result?

Sporknife, 2024-01-08:
@ldrahnik Can you walk me through the steps? From what I've understood they go like this.
1. ``bash install.sh -unicode``
2. Change keyboard to french
3. Test if it works.

ldrahnik, 2024-01-12:
@Sporknife ping

adamant-pwn, 2024-03-07:
I'm using Hyprland on Arch Linux.

Installing this as a global service via [asus-numberpad-driver-gx551-git](https://aur.archlinux.org/packages/asus-numberpad-driver-gx551-git) fails for some reason, even if I try to do it in tty (similar to #144). However, installing it as a user service from git via `bash install.sh` does work.

In tty, it works fully as expected. But on Hyprland:
- Switching from touchpad to numberpad doesn't disable touchpad, so it registers the number press, but also registers it as a mouse click, likely moving the cursor or removing focus from the text field;
- Numberpad is automatically turned off if I press any regular keyboard button.

adamant-pwn, 2024-03-07:
Here's the error log of doing the following:
1. Switch the numberpad on.
2. Try to use all buttons in order `0123456789/*-+.<enter>` (it also registers by laptop as LMB clicks).
3. Switch the numberpad off.
4. Switch the numberpad on and press "a" (it switches off on its own).

<details>
<summary>/var/conf/asus-numberpad-driver/error.log</summary>

```sh
2024-03-07 20:47:50,858 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-07 20:47:50,888 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-07 20:47:50,888 INFO     Set keyboard 5 from H: Handlers=sysrq kbd leds event5
2024-03-07 20:47:50,904 INFO     Detecting touchpad from string: "N: Name="ASUE140F:00 04F3:32A2 Touchpad""
2024-03-07 20:47:50,905 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE140F:00/0018:04F3:32A2.0004/input/input26
2024-03-07 20:47:50,905 INFO     Set touchpad id 9 from H: Handlers=event9 mouse1
2024-03-07 20:47:50,905 INFO     Touchpad min-max: x 0-1710, y 0-2925
2024-03-07 20:47:50,905 INFO     Numpad min-max: x 20-1690, y 0-2925
2024-03-07 20:48:49,056 INFO     Touched numlock key (not top_right_icon) in time: 1709840929.0566313
2024-03-07 20:48:50,302 INFO     The numpad numlock was pressed longer than the activation time: 1.2456622123718262
2024-03-07 20:48:50,302 INFO     Activation time: 1.0
2024-03-07 20:48:50,302 INFO     Un-touched with NumberPad activation numlock key (not top_right_icon) in time: 1709840930.3024247
2024-03-07 20:48:50,341 INFO     System numlock activated
2024-03-07 20:48:50,341 INFO     Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140F:00 04F3:32A2 Touchpad
2024-03-07 20:48:50,394 INFO     Setting up for config file key: "enabled" with value: "True"
2024-03-07 20:48:50,394 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-03-07 20:48:58,056 INFO     Pressed numpad key
2024-03-07 20:48:58,056 INFO     KEY_KP0:82
2024-03-07 20:48:58,056 INFO     Unpressed numpad key
2024-03-07 20:48:58,056 INFO     KEY_KP0:82
2024-03-07 20:49:01,097 INFO     Pressed numpad key
2024-03-07 20:49:01,097 INFO     KEY_KP1:79
2024-03-07 20:49:01,097 INFO     Unpressed numpad key
2024-03-07 20:49:01,097 INFO     KEY_KP1:79
2024-03-07 20:49:01,953 INFO     Pressed numpad key
2024-03-07 20:49:01,953 INFO     KEY_KP2:80
2024-03-07 20:49:01,953 INFO     Unpressed numpad key
2024-03-07 20:49:01,953 INFO     KEY_KP2:80
2024-03-07 20:49:04,173 INFO     Pressed numpad key
2024-03-07 20:49:04,173 INFO     KEY_KP3:81
2024-03-07 20:49:04,173 INFO     Unpressed numpad key
2024-03-07 20:49:04,173 INFO     KEY_KP3:81
2024-03-07 20:49:04,803 INFO     Pressed numpad key
2024-03-07 20:49:04,803 INFO     KEY_KP4:75
2024-03-07 20:49:04,803 INFO     Unpressed numpad key
2024-03-07 20:49:04,803 INFO     KEY_KP4:75
2024-03-07 20:49:05,399 INFO     Pressed numpad key
2024-03-07 20:49:05,399 INFO     KEY_KP5:76
2024-03-07 20:49:05,399 INFO     Unpressed numpad key
2024-03-07 20:49:05,399 INFO     KEY_KP5:76
2024-03-07 20:49:06,680 INFO     Pressed numpad key
2024-03-07 20:49:06,680 INFO     KEY_KP6:77
2024-03-07 20:49:06,680 INFO     Unpressed numpad key
2024-03-07 20:49:06,680 INFO     KEY_KP6:77
2024-03-07 20:49:07,940 INFO     Pressed numpad key
2024-03-07 20:49:07,940 INFO     KEY_KP7:71
2024-03-07 20:49:07,940 INFO     Unpressed numpad key
2024-03-07 20:49:07,940 INFO     KEY_KP7:71
2024-03-07 20:49:08,810 INFO     Pressed numpad key
2024-03-07 20:49:08,810 INFO     KEY_KP8:72
2024-03-07 20:49:08,810 INFO     Unpressed numpad key
2024-03-07 20:49:08,810 INFO     KEY_KP8:72
2024-03-07 20:49:10,104 INFO     Pressed numpad key
2024-03-07 20:49:10,104 INFO     KEY_KP9:73
2024-03-07 20:49:10,104 INFO     Unpressed numpad key
2024-03-07 20:49:10,104 INFO     KEY_KP9:73
2024-03-07 20:49:13,172 INFO     Pressed numpad key
[... 29 lines cut ...]
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140F:00 04F3:32A2 Touchpad
2024-03-07 20:49:28,040 INFO     Setting up for config file key: "enabled" with value: "False"
2024-03-07 20:49:28,040 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-03-07 20:50:32,974 INFO     Touched numlock key (not top_right_icon) in time: 1709841032.9748936
2024-03-07 20:50:34,384 INFO     The numpad numlock was pressed longer than the activation time: 1.4100697040557861
2024-03-07 20:50:34,385 INFO     Activation time: 1.0
2024-03-07 20:50:34,385 INFO     Un-touched with NumberPad activation numlock key (not top_right_icon) in time: 1709841034.3850563
2024-03-07 20:50:34,411 INFO     System numlock activated
2024-03-07 20:50:34,411 INFO     Numpad activated
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
unable to find device ASUE140F:00 04F3:32A2 Touchpad
2024-03-07 20:50:34,428 INFO     Setting up for config file key: "enabled" with value: "True"
[... cut ...]
unable to find device ASUE140F:00 04F3:32A2 Touchpad
2024-03-07 20:50:35,607 INFO     Setting up for config file key: "enabled" with value: "False"
2024-03-07 20:50:35,607 INFO     check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=numberpad_dev path=/usr/share/asus-numberpad-driver pathname=/usr/share/asus-numberpad-driver/numberpad_dev wd=1 >
2024-03-07 20:50:35,707 INFO     Numpad deactivated
```

</details>

adamant-pwn, 2024-03-07:
And as for [asus-numberpad-driver-gx551-git](https://aur.archlinux.org/packages/asus-numberpad-driver-gx551-git) installation, it gives the following:
```sh
[adamant@zephyrus-duo-16 ~]$ systemctl status asus_numberpad_driver
× asus_numberpad_driver.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/system/asus_numberpad_driver.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2024-03-07 20:57:59 CET; 49s ago
   Duration: 566us
    Process: 812 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py gx551 /usr/share/asus-numberpad-driver (code=exited, status=203/EXEC)
   Main PID: 812 (code=exited, status=203/EXEC)
        CPU: 433us

Mar 07 20:57:59 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Scheduled restart job, restart counter is at 5.
Mar 07 20:57:59 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Start request repeated too quickly.
Mar 07 20:57:59 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Failed with result 'exit-code'.
Mar 07 20:57:59 zephyrus-duo-16 systemd[1]: Failed to start Asus NumberPad Driver.
```

adamant-pwn, 2024-03-07:
On some further inspection, the command from the global service
```sh
/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py gx551 /usr/share/asus-numberpad-driver
```
fails because `.env` doesn't exist in `/usr/share/asus-numberpad-driver` on my machine for some reason.

ldrahnik, 2024-03-07:
What is output of `$ libinput debug-events` because touchpad device was not loaded `unable to find device ASUE140F:00 04F3:32A2 Touchpad`

ldrahnik, 2024-03-07:
@kamack38 Could you check it please?

adamant-pwn, 2024-03-07:
```sh

[adamant@zephyrus-duo-16 ~/asus-numberpad-driver]$ xinput
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:14                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:14            	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer-gestures:14            	id=8	[slave  pointer  (2)]
⎜   ↳ xwayland-touch:14                       	id=10	[slave  pointer  (2)]
⎜   ↳ xwayland-tablet stylus:14               	id=11	[slave  pointer  (2)]
⎜   ↳ xwayland-tablet eraser:14               	id=12	[slave  pointer  (2)]
⎜   ↳ xwayland-tablet cursor:14               	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:14                    	id=9	[slave  keyboard (3)]
```

ldrahnik, 2024-03-07:
@adamant-pwn Now is in driver implemented disabling tapping (when is NumberPad activated) using `$ xinput` & `$ gsettings` & `$ synclient` and that is reason why it does not work for you I guess. How is disabled tap-to-click in hyprland? Could you provide install log? Which distro do you use? KDE based?

kamack38, 2024-03-07:
> @kamack38 Could you check it please?

I'll fix the package on Saturday. As to the Wayland driver I'm using Hyprland too but I'll still check it out then.

ldrahnik, 2024-03-07:
@adamant-pwn Wait, you can enable/disable tap-to-click using xinput? And only one problem was the device names are different because of wayland compositor?

ldrahnik, 2024-03-07:
@adamant-pwn Please, what is output of `$ xinput list-props <device_of_touchpad - maybe xwayland-touch?>` and `$ xinput list-props <Virtual core XTEST keyboard> and `$ xinput list-props <xwayland-keyboard>`

ldrahnik, 2024-03-07:
@adamant-pwn Do you have `$ gsettings`? Try this:

```
$ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' false
# test whether you can tap-to-click touchpad, no? It works
$ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' true
```

adamant-pwn, 2024-03-07:
I have gsettings, but I can still tap-to-click after using that command.

ldrahnik, 2024-03-07:
@adamant-pwn Is your distro gnome based?

adamant-pwn, 2024-03-07:
I use Hyprland window manager on Arch Linux, I don't have any desktop environment.

ldrahnik, 2024-03-07:
@kamack38 Do not you use the same setup? Arch with Hyprland? How do you disable tap-to-click? Mentioned `gsettings`?

kamack38, 2024-03-07:
> @kamack38 Do not you use the same setup? Arch with Hyprland? How do you disable touchpad's `tap-to-click`? By mentioned `gsettings`?

What do you mean by `tap-to-click`?

ldrahnik, 2024-03-07:
@kamack38 By default is driver installed with `enabled_touchpad_pointer=3`. Do you use `3`? And then when is NumberPad activated is disabled touchpad's left-click (tap-to-click device libinput option - in my case - x11)

kamack38, 2024-03-09:
```
❯ G_MESSAGES_DEBUG=all LOG=DEBUG python3 /usr/share/asus-numberpad-driver/numberpad.py g533 /usr/share/asus-numberpad-driver
2024-03-09 18:57:40,170 INFO     Detecting keyboard from string: "N: Name="Asus Keyboard""
2024-03-09 18:57:40,216 INFO     Detecting touchpad from string: "N: Name="ASUE1403:00 04F3:319A Touchpad""
2024-03-09 18:57:40,216 INFO     Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.000C/input/input34
2024-03-09 18:57:40,216 INFO     Set touchpad id 17 from H: Handlers=event17 mouse1
2024-03-09 18:57:40,216 INFO     Detecting keyboard from string: "N: Name="ASUE1403:00 04F3:319A Keyboard""
2024-03-09 18:57:40,216 INFO     Set keyboard 19 from H: Handlers=sysrq kbd leds event19
2024-03-09 18:57:40,246 ERROR    Can't open the I2C bus connection (id: 0)
```
After fixing the package dependencies, it doesn't work for me on Wayland

ldrahnik, 2024-03-09:
@kamack38 It does not look like problem with dbus but with permissions problem during accessing I2c device as user - permissions should be added automatically when laptop boot for user [file](https://github.com/asus-linux-drivers/asus-numberpad-driver/blob/master/install_user_groups.sh)

ldrahnik, 2024-03-15:
@adamant-pwn Does arch package with changes by @kamack38 works for you?

kamack38, 2024-03-30:
After reading [this](https://wiki.archlinux.org/title/Keyboard_input) Arch Wiki article, it occurred to me that if we figure out how to send keysyms we would be able to avoid any Xorg and Wayland dependency.

ldrahnik, 2024-04-02:
@kamack38 I do not know what do you mean.

> At the protocol level, wayland has a wl_keyboard.keymap event. This event contains a file descriptor to the keymap and a format classifier. Right now, only one format is defined: "xkb". So a wayland client will receive an XKB-compatible keymap and can use libxkbcommon to interpret that to get the right glyph on the screen, etc.

I tried to implement a way for Wayland. The branch is attached, look at that, please. It looks like it works under Ubuntu 22.04 partially, when I move in the settings keyboard layout French BEPO to the first position, then I receive via hook new keymap, also to the client compositor sends the current layout BEPO as active correctly and others as inactive, and for example for 'u' character usually printed under 'u' with keycode 22 driver send correctly 31 `s` as it is correct for French BEPO BUT when I change keyboard layout via Super+Space I do not receive any event with new keymap for changed active keyboard via hook so it still sends wrong character, I do not know why.

Everything should be tested and still it is better then nothing. But as I do not use wayland yet I developed that in virtual box with Ubuntu22.04.

adamant-pwn, 2024-04-18:
Hi, sorry for the delayed response! AUR package [asus-numberpad-driver-gx551-git](https://aur.archlinux.org/packages/asus-numberpad-driver-gx551-git) still doesn't work for me:
```
[adamant@zephyrus-duo-16 ~]$ systemctl status asus_numberpad_driver
× asus_numberpad_driver.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/system/asus_numberpad_driver.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2024-04-18 14:24:28 CEST; 1min 45s ago
   Duration: 2ms
    Process: 797 ExecStart=python3 /usr/share/asus-numberpad-driver/numberpad.py gx551 /usr/share/asus-numberpad-driver (code=exited, status=209/STDOUT)
   Main PID: 797 (code=exited, status=209/STDOUT)
        CPU: 3ms

Apr 18 14:24:28 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Scheduled restart job, restart counter is at 5.
Apr 18 14:24:28 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Start request repeated too quickly.
Apr 18 14:24:28 zephyrus-duo-16 systemd[1]: asus_numberpad_driver.service: Failed with result 'exit-code'.
Apr 18 14:24:28 zephyrus-duo-16 systemd[1]: Failed to start Asus NumberPad Driver.
```

ldrahnik, 2024-04-18:
@adamant-pwn 

> Also when I try to execute the command separately, I get:

> [adamant@zephyrus-duo-16 ~]$ python3 /usr/share/asus-numberpad-driver/numberpad.py gx551 /usr/share/asus- numberpad-driver
> 2024-04-18 14:29:24,003 ERROR    Numpad layout *.py from dir layouts is required as first argument. Re-run inst

Directly it should work now as I committed the missing standard layout `gx551`. Were exist only `gx551-unicode` and `gx551-non-unicode` variant.

ldrahnik, 2024-04-18:
@adamant-pwn 

> Hi, sorry for the delayed response! AUR package [asus-numberpad-driver-gx551-git](https://aur.archlinux.org/packages/asus-numberpad-driver-gx551-git) still doesn't work for me:
> ``` [...]

Without an error log we unfortunately can do nothing as we only see return code. Thank you for posting. Np for delay.

ldrahnik, 2024-05-02:
Can be reopened.

closed 2024-05-02 by ldrahnik