asus-linux-drivers/asus-numberpad-driver #105: [Feature Request] Short Description

esmaily, 2023-03-09:
### Short Description

Please this model in support list Asus Zenbook `UM3402YA` also two and three gestures actions such as minize ,maximaize,switch between apps and volume up and down

ldrahnik, 2023-03-10:
@esmaily Thank you for creating an issue. But that was very short and I am not sure what you mean. Does NumberPad driver works for model `Asus Zenbook UM3402YA` correctly? Is NumberLayout associated correctly? Do you mean add laptop only to list for layout automatic recommendation? What is output of command `$ sudo dmidecode -s system-product-name`? Layout which I use too should be fit for your laptop too, that means [up5401ea.py](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/blob/master/numpad_layouts/up5401ea.py)

> three gestures actions such as minize ,maximaize,switch between apps and volume up and down
I do not understand, is it somehow associated with this repository? If not and it can be associated to any repository in this org please create separated issue there, otherwise does not fit any repository for issue, you can write me e-mail (mentioned in org) and I could try help you but I promise nothing.

ldrahnik, 2023-03-10:
@esmaily I am sorry, this driver take care only about NumberPad funkcionality and as it looks like nothing else is associated to this driver so I have to close this issue. In case volume up down does not work I recommend you look ať WMI driver in this org And in aby problems create issue there. For defining own  touchpad gestures are already developed projects, look around in internet, I do not use any So I can not recommend.

closed 2023-03-10 by ldrahnik

esmaily, 2023-03-10:
@ldrahnik thanks