asus-linux-drivers/asus-numberpad-driver #108: [Bug]: Not working on UM5401QA

lyollix, 2023-03-10:
### Describe the bug

Installed but nothing works

[error.log](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/files/10944516/error.log)

### Expected behavior

I believe when touching the right corner of the touchpad a numeric keypad would light up and I would be able to type in numbers. Wherever I tap, long-press, double-tap, triple-tap on the top of the keypad, nothing alike happens.

### Relevant log output

```shell
sudo bash ./install.sh --user
`
groupadd: group 'uinput' already exists
KERNEL=="uinput", GROUP="uinput", MODE:="0660"
driver will run under user lyollix
apt is here
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Уже установлен пакет ibus самой новой версии (1.5.26-4).
Уже установлен пакет libevdev2 самой новой версии (1.12.1+dfsg-1).
Уже установлен пакет i2c-tools самой новой версии (4.3-2build1).
Уже установлен пакет python3-dev самой новой версии (3.10.6-1~22.04).
Уже установлен пакет python3-pip самой новой версии (22.0.2+dfsg-1ubuntu0.2).
Обновлено 0 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Уже установлен пакет xinput самой новой версии (1.6.3-1build2).
Обновлено 0 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: libevdev in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (0.5)
Requirement already satisfied: evdev in /home/lyollix/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /home/lyollix/.local/lib/python3.10/site-packages (from -r requirements.txt (line 3)) (1.24.2)
Requirement already satisfied: inotify in /home/lyollix/.local/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: python-xlib in /home/lyollix/.local/lib/python3.10/site-packages (from -r requirements.txt (line 5)) (0.33)
Requirement already satisfied: nose in /home/lyollix/.local/lib/python3.10/site-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
Requirement already satisfied: six>=1.10.0 in /usr/lib/python3/dist-packages (from python-xlib->-r requirements.txt (line 5)) (1.16.0)
Testing interface i2c-0 : failed
Testing interface i2c-1 : sucess
Detected laptop: Zenbook UM5401QAB_UM5401QA
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 52030    0 52030    0     0   209k      0 --:--:-- --:--:-- --:--:--  209k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  249k    0  249k    0     0   822k      0 --:--:-- --:--:-- --:--:--  824k
Automatically recommended numpad layout: up5401ea (associated to UM3402YA-OLED059W). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues).
Asus touchpad numpad service enabled
Asus touchpad numpad service started
Install finished
`
```

### Desktop

Kubuntu 22.04 LTS

ldrahnik, 2023-03-10:
@lyollix Hello, thank you for creating an issue. Could you please run this command `$  getfacl /dev/input/eventxy`, it looks like your user has no permissions for group input. Mentioned command will print group under which run events on your laptop.

lyollix, 2023-03-10:
getfacl /dev/input/eventxy
getfacl: /dev/input/eventxy: No such file or directory

ldrahnik, 2023-03-10:
@lyollix `xy` means please use random number

ldrahnik, 2023-03-10:
@lyollix Print which groups has your user via `$ groups lyollix`

lyollix, 2023-03-10:
groups lyollix
lyollix : lyollix adm cdrom sudo dip plugdev input lpadmin lxd sambashare i2c uinput

ldrahnik, 2023-03-10:
@lyollix Try uninstall via `$ sudo bash ./uninstall.sh --user` and let me know driver works as sudo via `sudo bash ./install.sh` (without `--user`)

lyollix, 2023-03-10:
doesn't work, same error again in error.log

lyollix, 2023-03-10:
I have a fresh kubuntu install, a couple of hours ago

ldrahnik, 2023-03-10:
@lyollix In case you did not reboot your laptop after the first updates & upgrades of libs I recommend you do it now. The error is weird, at this moment I do not know what is the problem. But I will think about that. Meanwhile please try google.

lyollix, 2023-03-10:
it works, just no backlight

lyollix, 2023-03-10:
@ldrahnik 
`
i2cdetect -l
i2c-0   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   smbus           SMBus PIIX4 adapter port 0 at 0b00      SMBus adapter
i2c-3   smbus           SMBus PIIX4 adapter port 2 at 0b00      SMBus adapter
i2c-4   smbus           SMBus PIIX4 adapter port 1 at 0b20      SMBus adapter
i2c-5   i2c             AMDGPU DM i2c hw bus 0                  I2C adapter
i2c-6   i2c             AMDGPU DM i2c hw bus 1                  I2C adapter
i2c-7   i2c             AMDGPU DM i2c hw bus 2                  I2C adapter
i2c-8   i2c             AMDGPU DM i2c hw bus 3                  I2C adapter
i2c-9   i2c             AMDGPU DM aux hw bus 0                  I2C adapter
i2c-10  i2c             AMDGPU DM aux hw bus 2                  I2C adapter
i2c-11  i2c             AMDGPU DM aux hw bus 3                  I2C adapter
`
`
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
`
it don't on backlight

ldrahnik, 2023-03-10:
@lyollix What was the problem? About blacklight try another I2C - number 0 or different values, test script is in folder test which use values from entire register 0 - 255. `0x01` should enable but when are used some values then is NumberPad blacklight somehow blocked and even values `0x01` can not enable it

lyollix, 2023-03-11:
The error "Permission denied" went away on its own after a reboot.

Somehow strange, with the help of the test, I determined that the inclusion is 0x60, the deactivation is 0x61; fixed the driver, everything worked fine, but after a reboot it stopped working and again determined by experience that now the on/off is 0x01/0x00; :)

Thanks for the help!

closed 2023-03-11 by lyollix

ldrahnik, 2023-03-11:
@lyollix Great it works. I will look at mentioned values by you `0x60`, `0x61` and if they can affect activating in my laptop too I will try to figure out how use them so this problem will no-one else has to solve. Create another issue in case something does not work properly. If you find driver useful put a star to repository :)