asus-linux-drivers/asus-numberpad-driver #109: Brightness values

tasmirz, 2023-03-12:
### Short Description

Is there any particular reason to use these brightness values?
```
    "0x41",
    "0x42",
    "0x43",
    "0x44",
    "0x45",
    "0x46",
    "0x47",
    "0x48"
```
As far as I have tested, Brightness is actually num%16 starting from `0x10`. 
both
`"0x10","0x11","0x12","0x13","0x14","0x15","0x16","0x17","0x18","0x19","0x1a","0x1b","0x1c","0x1d","0x1e","0x1f"`
and 
`"0x20","0x21","0x22","0x23","0x24","0x25","0x26","0x27","0x28","0x29","0x2a","0x2b","0x2c","0x2d","0x2e","0x2f"`
produces the same output.
Can you add the last values to (starting from 9-f) to allow greater control?

ldrahnik, 2023-03-12:
@tasmirz Thank you for creating an issue and opening a discussion about this. Could you mention laptop model and touchpad product id and used touchpad driver? Equal to this example `input: ASUE120A:00 04F3:319B Touchpad`.
Values used in layouts work for my laptop and as I have no way how I would check for others are copied across others layouts as defaults. Looks like different values are associated with different Touchpad ID's (I hope at least), might differ or might not. Collecting of these values should start. I will probably add block to README.MD help is needed as I do not see other options (concact developers who develop this driver for Windows though Asus is not possible as Asus does not support Linux and I do not know anyone personally. Via support I already asked. No help)

tasmirz, 2023-03-13:
Laptop Model : `ASUS EXPERTBOOK B3302FEA_B5302FEA 1.0`
Product ID and touchpad driver `ASUE140C:00 04F3:3145 Touchpad`
And thanks for responding.

ldrahnik, 2023-03-13:
Sorry I do not understand which values you propose, could you please write them one more time into array like is mine: ```backlight_levels = [
    "0x41",
    "0x42",
    "0x43",
    "0x44",
    "0x45",
    "0x46",
    "0x47",
    "0x48"
]
```

tasmirz, 2023-03-13:
You can try this one.
```
backlight_levels = [
     "0x10","0x11","0x12","0x13","0x14","0x15","0x16","0x17","0x18","0x19","0x1a","0x1b","0x1c","0x1d","0x1e","0x1f"
 ]
```
What I meant was, 
The above array and this one below would produce the same result.
```
backlight_levels = [
"0x20","0x21","0x22","0x23","0x24","0x25","0x26","0x27","0x28","0x29","0x2a","0x2b","0x2c","0x2d","0x2e","0x2f"
]
```
So, If you even start from` "0x40"` to `"0x4f"` that would also produce the same output.  
Here `"0x10"` or `"0x20"` both outputs max brightness. and `"0x1f"` and `"0x2f"` outputs the min brightness.

ldrahnik, 2023-03-13:
@tasmirz Thank you. I committed association between different brightness levels and touchpad id's, e.g. `up5401ea-31B9.py`, `up5401ea-3145.py`.

ldrahnik, 2023-03-13:
@tasmirz Let me know it works so I can close this issue :)

tasmirz, 2023-03-16:
It works. Thank you for the fast patch and sorry for late response.

closed 2023-03-16 by ldrahnik