asus-linux-drivers/asus-numberpad-driver #110: Nothing happens after (successful?) installation

sanjibukai, 2023-03-16:
### Describe the bug

Hello,
I'm running a zenbook UX582 on Fedora 37 and I installed that driver.. After installing all the dependencies it seems I got the driver installed correctly as I don't get any error.
But after installing nothing particular is happening, when I touch on the top of my touchpad (tapping the icons trying to enable the numpad), the cursor is just moving..
Here is what I did:
```
git clone https://github.com/asus-linux-drivers/asus-numberpad-driver
cd asus-numberpad-driver
sudo bash ./install.sh
```
And here is the output:
```
driver will run under user root
Requirement already satisfied: libevdev in /usr/lib/python3.11/site-packages (from -r requirements.txt (line 1)) (0.11)
Requirement already satisfied: evdev in /usr/local/lib64/python3.11/site-packages (from -r requirements.txt (line 2)) (1.6.1)
Requirement already satisfied: numpy in /usr/lib64/python3.11/site-packages (from -r requirements.txt (line 3)) (1.22.0)
Requirement already satisfied: inotify in /usr/local/lib/python3.11/site-packages (from -r requirements.txt (line 4)) (0.2.10)
Requirement already satisfied: python-xlib in /usr/local/lib/python3.11/site-packages (from -r requirements.txt (line 5)) (0.33)
Requirement already satisfied: nose in /usr/local/lib/python3.11/site-packages (from inotify->-r requirements.txt (line 4)) (1.3.7)
Requirement already satisfied: six>=1.10.0 in /usr/lib/python3.11/site-packages (from python-xlib->-r requirements.txt (line 5)) (1.16.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Testing interface i2c-0 : failed
Testing interface i2c-1 : sucess
Detected laptop: ZenBook Pro Duo UX582ZM_UX582ZM
Automatically recommended numpad layout: ux581l (associated to UX582ZM-OLED032W). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout ux581l
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
io.elementary.calculator here
Toggling script for calculator app gnome-calculator has been installed.
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad@root.service → /etc/systemd/system/asus_touchpad_numpad@.service.
Asus touchpad numpad service enabled
Asus touchpad numpad service started
Install finished
```
I chose `y` on every questions.

What I'm supposed to do?

Thank you very much!

### Expected behavior

After a successful installation, I was expecting to enable/disable the numpad using the icon.

### Desktop

- Program version: 9fd27d8
- Fedora 37
- Kernel 6.1.18

ldrahnik, 2023-03-16:
@sanjibukai Hello, could you please attach entire log file of installed driver: ` /var/log/asus_touchpad_numpad-driver/error.log` and also run driver directly without installing with enabled DEBUG as is described in [Troubleshoting](https://github.com/asus-linux-drivers/asus-numberpad-driver#troubleshooting) block in README.

sanjibukai, 2023-03-16:
> could you please attach entire log file of installed driver: ` /var/log/asus_touchpad_numpad-driver/error.log`

Hello @ldrahnik  and thanks.
Here is the output of the `error.log` and indeed, it seems to have some errors..
```
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:4067 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input20
INFO:asus-touchpad-numpad-driver:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:4067 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input20
INFO:asus-touchpad-numpad-driver:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
INFO:asus-touchpad-numpad-driver:System numlock activated
INFO:asus-touchpad-numpad-driver:Numpad activated
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "enabled" with value: "True"
Error: Sending messages failed: Remote I/O error
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:4067 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input20
INFO:asus-touchpad-numpad-driver:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
Unable to connect to X server
ERROR:asus-touchpad-numpad-driver:Getting Device Enabled via xinput failed
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 406, in is_device_enabled
    propData = subprocess.check_output(cmd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xinput', '--list-props', 'ELAN9009:00 04F3:4067 Touchpad']' returned non-zero exit status 1.
INFO:asus-touchpad-numpad-driver:System numlock activated
INFO:asus-touchpad-numpad-driver:Numpad activated
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "enabled" with value: "True"
Unable to connect to X server
ERROR:asus-touchpad-numpad-driver:b''
Error: Sending messages failed: Remote I/O error
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "enabled" with value: "False"
Unable to connect to X server
ERROR:asus-touchpad-numpad-driver:b''
Error: Sending messages failed: Remote I/O error
INFO:asus-touchpad-numpad-driver:Numpad deactivated
Unable to connect to X server
ERROR:asus-touchpad-numpad-driver:Getting Device Enabled via xinput failed
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 406, in is_device_enabled
    propData = subprocess.check_output(cmd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xinput', '--list-props', 'ELAN9009:00 04F3:4067 Touchpad']' returned non-zero exit status 1.
Unable to connect to X server
ERROR:asus-touchpad-numpad-driver:Getting Device Enabled via xinput failed
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 406, in is_device_enabled
    propData = subprocess.check_output(cmd)
[... 69 lines cut ...]
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 406, in is_device_enabled
    propData = subprocess.check_output(cmd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xinput', '--list-props', 'ELAN9009:00 04F3:4067 Touchpad']' returned non-zero exit status 1.
INFO:asus-touchpad-numpad-driver:Getting Device Enabled via xinput disabled because failed more then: "9" times in row

```

sanjibukai, 2023-03-16:
I also ran with the debug mode by doing the following:
```
sudo systemctl stop asus_touchpad_numpad@root.service
cd asus-numberpad-driver
LOG=DEBUG sudo -E ./asus_touchpad.py "ux581l" ""
```
And here is the output:
```
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ELAN9009:00 04F3:4067 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input20
INFO:asus-touchpad-numpad-driver:Set touchpad id 15 from H: Handlers=mouse4 event15
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3984, y 0-1152
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 80-3904, y 200-1072
DEBUG:asus-touchpad-numpad-driver:Writting to config file: "<_io.TextIOWrapper name='asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"
```
It seems it still running as the command didn't return (I stopped doing `ctrl+c`) but nothing happens..
I tried to use the touchpad to see if additional log would be outputted but nothing else appears.

ldrahnik, 2023-03-16:
@sanjibukai Please run command `$ cat /proc/bus/input/devices` and post output here

sanjibukai, 2023-03-16:
@ldrahnik Hi again..
Here is the output (fairly long, I hope it's okay, otherwise tell me if there is a better way to share outputs (like pastebin etc.)

```
I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:09/PNP0C09:01/PNP0C0D:01/input/input0
U: Uniq=
H: Handlers=event0 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd leds event3 
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0018 Vendor=04f3 Product=4067 Version=0100
N: Name="ELAN9009:00 04F3:4067"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input14
U: Uniq=
H: Handlers=mouse2 event11 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
B: ABS=3273800000000003
B: MSC=20

I: Bus=0018 Vendor=04f3 Product=4067 Version=0100
N: Name="ELAN9009:00 04F3:4067 UNKNOWN"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-2/i2c-ELAN9009:00/0018:04F3:4067.0003/input/input15
U: Uniq=
H: Handlers=event12 
B: PROP=0
B: EV=9
B: ABS=10000000000

I: Bus=0018 Vendor=04f3 Product=4067 Version=0100
N: Name="ELAN9009:00 04F3:4067 UNKNOWN"
[... 269 lines cut ...]
B: KEY=100000 0 0 0

I: Bus=0005 Vendor=054c Product=0df0 Version=0113
N: Name="WH-1000XM5 (AVRCP)"
P: Phys=98:43:fa:94:ff:8d
S: Sysfs=/devices/virtual/input/input50
U: Uniq=
H: Handlers=kbd event257 
B: PROP=0
B: EV=100007
B: KEY=2fc800 145200000000 0 10300 49e800000c00 e16800000000f f810000010000ffc
B: REL=0

I: Bus=0005 Vendor=046d Product=b034 Version=0003
N: Name="MX Master 3S Mouse"
P: Phys=98:43:fa:94:ff:8d
S: Sysfs=/devices/virtual/misc/uhid/0005:046D:B034.0008/input/input51
U: Uniq=d3:27:97:f3:b7:f8
H: Handlers=mouse7 event258 
B: PROP=0
B: EV=17
B: KEY=ffff0000 0 0 0 0
B: REL=1943
B: MSC=10
```

ldrahnik, 2023-03-16:
@sanjibukai You have 2 touchpad devices and this driver was trying to connect to the wrong one. Try `git pull`. Latest master should be OK for your laptop. Let me know so I can close an issue.

ldrahnik, 2023-03-16:
@sanjibukai In case brightness change/activation /deactivation will not work for you but keys you will be able to press, please look at this thread, similar laptop to yours (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/95), we need to get values which change via I2C the brightness for you. The way how to get these values is by brute forcing all possible values to control the registr (1-255) file `tests/test_brightness.py`, also read, in that case, `Readme.md` file text in troubleshoting and text above Touchpad models.

sanjibukai, 2023-03-16:
@ldrahnik Wow!
It works!
Seems I need to tap and hold a few seconds (I guess that's the intended behavior in order to not activate it when using the touchpad I guess, I just never used it on windows lol)
The top right "button" (calculator or numpad icon) is switching on the light and then I'm able to input numbers..
And it seems the top left "button" (triangle with dots) is changing the dimming..

I have two questions though:
- It seems it's turning off after a moment. Is it possible to keep it on?
- Also, when installing I saw some prompts about the Calculator App, is there a way to launch the calculator app from the touchpad? I'm not sure but I remember having read something like that was possible in windows at least.. Do you know what's the correct sequence or gesture to trigger the calculator?

Many thanks!

ldrahnik, 2023-03-16:
@sanjibukai Default activation is press calc icon at least for 1s, configurable. Also is possible to activate NumberPad with a slide gesture with one finger from calc icon on top right corner to the center (30% width & height atleast), then is NumberPad activated immediately.

- NumberPad auto disabling due to inactivity is by default 1 minute so I guess what is turning off is Touchpad and is going to sleep mode. How to confirm it, when you touch the key then is backlight woke up at the same level and you do not have to NumberPad activate again. The time when it goes to sleep mode for touchpad is configurable in linux.
- Please read `Readme.md` about this fist

sanjibukai, 2023-03-16:
@ldrahnik Thank you for these detailed explanations!
Indeed swiping from the top right corner toward the middle is switching on right away.
Regarding the sleep, after a minute or so, only the lights (of the numpad) is turning off, and when I touch the touchpad, the lights does not come back but the touchpad is working as usual.

And, regarding the Calculator, when I do `xdotool key XF86Calculator` nothing happens..
I tried with controlling the volume e.g. `xdotool key XF86AudioRaiseVolume` and that correctly triggered the increase of the volume (with the overlay of Gnome etc.)

When I tried to see the input using `sudo libinput debug-events`, I'm able to switch the numpad with the swipe to the right from the calculator icon, but I don't see the XF86Calculator being triggered:
This is what I'm seeing:
```
...
 event17  POINTER_MOTION          +1.068s	 -0.23/ -0.46 ( -1.00/ -2.00)
 event17  POINTER_MOTION          +1.096s	 -0.21/ -0.42 ( -1.00/ -2.00)
 event17  GESTURE_HOLD_BEGIN      +2.495s	1
 event17  POINTER_MOTION          +2.503s	 -3.25/ -0.89 (-11.00/ -3.00)
 event17  POINTER_MOTION          +2.510s	 -1.36/ -0.34 ( -4.00/ -1.00)
 event17  GESTURE_HOLD_END        +2.510s	1 cancelled
 event17  POINTER_MOTION          +2.517s	 -1.36/ -0.34 ( -4.00/ -1.00)
 event17  POINTER_MOTION          +2.524s	 -1.36/  0.00 ( -4.00/ +0.00)
...
```
Using an actual media key displays correctly the following for example:
```
-event3   KEYBOARD_KEY            +4294967.295s	KEY_VOLUMEUP (115) pressed
 event3   KEYBOARD_KEY            +0.049s	KEY_VOLUMEUP (115) released
```

Regarding the xdotool command I can tell I don't have any shortcut setup..
But OTOH, it seems the Calculator key is also not triggering (so I don't know how I can setup a custom shortcut in Gnome settings without triggering the key.. I tried doing the following in the shortcut dialog settings ` leep 5 ; xdotool key XF86Calculator` (and switching back into the dialog so the triggered key is captured for the shortcut) but it doesn't worked..

sanjibukai, 2023-03-16:
Something interesting.. I was able to see some existing custom shortcuts (I guess generated from your install script)..
The command is `bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh `
I looked into that file and here is the content:
```
#!/bin/bash

if [[ $(ps aux | grep io.elementary.calculator | grep -v grep) ]]; then
	flatpak kill io.elementary.calculator &>/dev/null &
else
	flatpak run io.elementary.calculator &>/dev/null &
fi
```
Now it makes sebes that this was not working since I don't have that program (even less as a flatpak)..

I replaced it with the following:
```
#!/bin/bash

if [[ $(ps aux | grep gnome-calculator | grep -v grep) ]]; then
	killall gnome-calculator  &>/dev/null &
else
	gnome-calculator &>/dev/null &
fi
```

That script now is working, but the shortcut is still not triggering unfortunately..

sanjibukai, 2023-03-16:
@ldrahnik Do you know if it's possible to trigger the numpad from the commandline?

ldrahnik, 2023-03-16:
@sanjibukai From your install script I see following line: `io.elementary.calculator here` (that is why was selected elementary calculator - is prioritized atm, in [this](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/83) feature request is planned during install script pick up program when is more calculators installed - not implemented yet). How I detect in install script elementary calculator is installed and flatpack aswell? With this line `if [[ $(type flatpak 2>/dev/null && flatpak list | grep io.elementary.calculator 2>/dev/null) ]]; then`. It does not work for you?

sanjibukai, 2023-03-16:
@ldrahnik Hello and thanks..
Sorry for the confusion, but I somehow fixed the command.. But nothing is triggered actually...
In my first message just above, I put the output of the command you told me to run (`xdotool key XF86Calculator` and `libinput debug-events`) and it seems the XF86Calculator key is never triggered..
I'm trying to trigger by swiping starting from the top right corner (calculator icon) towards the left..
And around the third of the width the numpad is switching on (so the numpad works) but there is nothing triggered (on libinput nothing shows up)

Also, do you know if it's possible to trigger programmatically the numpad using CLI?

Many thanks!

ldrahnik, 2023-03-17:
@sanjibukai

- enable config option `sys_numlock_enables_numpad = 1` and `numpad_disables_sys_numlock = 1` (disabled both by default), then will NumberPad activated/disabled according to system numlock, how press system numlock (source can be numlock physical key even on the same laptop or external keyboard or simulated via `xdotool key Num_Lock`

- directly just change `enabled` in appropriate config file:

```
# enable NumberPad via command line
sudo sed -i "s/enabled.*/enabled = 1/g" asus_touchpad_numpad_dev
# disable
sudo sed -i "s/enabled.*/enabled = 0/g" asus_touchpad_numpad_dev
```

Tested on latest master.

ldrahnik, 2023-03-18:
@sanjibukai

**About toggling script for calculator**

Please, try the latest master, I hope I fixed all bugs regarding to toggling calcualtor script. I tried a clean installation for running under root/user and looks like it works. Remember every time when manipulating with keybindings is required at least log out. Also, I hope i have not missed any point you asked etc. Thank you for letting me know.

sanjibukai, 2023-03-19:
Hello @ldrahnik 
I don't know why but activating the numpad is never triggering the XF86Calculator key event.
But indeed with the correct configuration, now I'm able to enable the numpad by doing `xdotool key Num_Lock` which somehow solve my problem.
I just bound one of the physical key to Num_Lock so now it's working (even if I lost one physical key now).

sanjibukai, 2023-03-19:
Can you confirm me if activating the numpad (swiping from the calculator icon at the right toward the left) SHOULD send a NumLock event or even a XF86Calculator event?
Also, last question, I saw in the configuration file there is also a configuration from swiping from the left (three connected circles icon) toward the right... Does this action should trigger something?
If not, is it possible to bind a key to that action?
Thanks.

ldrahnik, 2023-03-19:
@sanjibukai Event `XF86Calculator` I have never seen and it works for me.

ldrahnik, 2023-03-19:
@sanjibukai Did you reboot laptop? Remove all shorcuts, set empty values to Calculator static & stativ as suggested in FAQ on README nos, look for sure you have no doubled this key `XF86Calculator` somewhere else via dconf, reboot.

sanjibukai, 2023-03-19:
Thanks (I've missed that in the README.. I'm going to try)
Quick question, when I hchange the config file, how can I make the settings to be updated?
I tried to stop and restart the service but still not taken into account..
I guess restarting will help..

sanjibukai, 2023-03-19:
Just checked and indeed this time, swiping from right is sending a KEY_EVENT (the num_lock key)..
I remember a couple of days ago it was not the case (log above).

But now I tried with the left side, and indeed there is still no KEY_EVENT triggered.
I did reset the gsettings as you can see:
```
$gsettings set org.gnome.settings-daemon.plugins.media-keys calculator [\'\']
$ gsettings set org.gnome.settings-daemon.plugins.media-keys calculator-static [\'\']
$gsettings get org.gnome.settings-daemon.plugins.media-keys calculator-static
['']
$gsettings get org.gnome.settings-daemon.plugins.media-keys calculator
['']
```
Doing one of the following launch correctly the calculator:
```
xdotool key XF86Calculator 
sudo bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh
# -> Here I modified the script to manage gnome-calculator
```
But the left swipe is still doing nothing..

As there is nothing shown in `sudo libinput debug-events` I wonder if it's because simply nothing is triggered?

(Again many thanks for the follow up)

ldrahnik, 2023-03-19:
@sanjibukai 

>Quick question, when I hchange the config file, how can I make the settings to be updated?
I tried to stop and restart the service but still not taken into account..
I guess restarting will help..

How do you change the config file? Which value is not changed immediately? In the driver is implemented `inotify` which should load config changes immediately. (`git pull` latest master is necessary). Example above works for me:

```
# installed as 
$ sudo bash ./install.sh

# enabling NumberPad via command line
# debugging in repository
sudo sed -i "s/enabled = 0/enabled = 1/g" asus_touchpad_numpad_dev
# installed config file
sudo sed -i "s/enabled = 0/enabled = 1/g" /usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev

# disabling (the same)
sudo sed -i "s/enabled = 1/enabled = 0/g" asus_touchpad_numpad_dev
sudo sed -i "s/enabled = 1/enabled = 0/g" /usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev

# installed as --user
$ sudo bash ./install.sh --user (the same as above but without sudo)

sed -i "s/enabled = 0/enabled = 1/g" asus_touchpad_numpad_dev
...
```

When you install the driver again, is installed correct script for the calculator? I mean for gnome. In case you do not have installed `flatpak` and `io.elementary.calculator` under `flatpak`. Should be. Try latest master.

For running the calculator toggle script you should run bash script as non sudo in both cases (installed as root and even as user):
```
bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh
```

>Just checked and indeed this time, swiping from right is sending a KEY_EVENT (the num_lock key)..
I remember a couple of days ago it was not the case (log above).

NumLock Key is not send when is system NumLock already enabled during activation.

ldrahnik, 2023-03-19:
@sanjibukai More about changing values in config for example when is connected external keyboard, is installed by default udev script which uses `sed` aswell. Tested now and it works. Not connected external keyboard, you can activate NumberPad with `xdotool key Num_Lock`, with connected external keyboard you can not. [Link](https://github.com/asus-linux-drivers/asus-numberpad-driver#external-keyboard-configuration) to doc.

ldrahnik, 2023-03-21:
@sanjibukai ping

ldrahnik, 2023-03-24:
@sanjibukai Can be reopened.

closed 2023-03-24 by ldrahnik