asus-linux-drivers/asus-numberpad-driver #113: [Bug]: UX363EA Doesn't stay off after turning off

Crylia, 2023-03-20:
### Describe the bug

When I try to turn off the numbpad(holding for 1 sec top right) it turns itself on again

### Expected behavior

It should stay off

### Desktop

- Program version V2.72
- Distro Pop!_OS
- Kernel 6.2.0-76060200

ldrahnik, 2023-03-21:
@Crylia Hello, thank you for creating an issue. I will let you know when will be fixed.

Could you please provide these information `$ egrep -B1 -A5 "ASUE|ELAN" /proc/bus/input/devices | grep -B1 -A5 Touchpad, whether default levels of backlight work for your model.. for the [table](https://github.com/asus-linux-drivers/asus-numberpad-driver#table-columns) in file `Readme.md`.

Crylia, 2023-03-21:
Yes it worked, thanks you very much.

Here is the output of the command if you are still interested (after the update)

```
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Touchpad"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input22
U: Uniq=
H: Handlers=mouse1 event17 
B: PROP=5
```

closed 2023-03-21 by Crylia