asus-linux-drivers/asus-numberpad-driver #115: [Bug]:

satva2, 2023-04-20:
### Describe the bug

When Installing in fedora 38, something to do with pip or numpy it seems, is breaking sddm. I'm new to Linux basically, so I'm really not sure. 

### Expected behavior

I installed it in the past with no problems on an older distro, but it might of been with an older version. I'm using a G533zx

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

kamack38, 2023-04-20:
First of all don't edit the issue add comments instead. Second of all please look at existing issues before filing a new one. As you can see #104 hasn't been closed, so it means Wayland is not currently supported. And please be more descriptive in your future issues. Say what happened and provide a log.

ldrahnik, 2023-04-20:
@kamack38 has right. Title & description is at this moment missing for futher help. Driver shoud work on wayland but with limizations mentioned in README.md and issue mentioned above. So depends what is your problem.

satva2, 2023-04-20:
Alright.. thanks for the reply and good work. Appreciate it.

closed 2023-04-20 by ldrahnik