asus-linux-drivers/asus-numberpad-driver #118: [Bug]: Black screen after login when systemctl enabled

isokosan, 2023-06-03:
### Describe the bug

I installed the dev version by mistake and encountered this issue. I then uninstalled and reinstalled the stable version 2.7.3, but the build is still broken. In both cases I've installed the driver on the user as suggested with the --user flag. Here is what happens:
 
When I run `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "/usr/share/asus_touchpad_numpad-driver/"` to test the drivers the drivers work but I do get errors in the console.
 
When I enable the systemctl service and reboot, there is only a black screen after the login menu. Perhaps that is related to the error I get I posted here or perhaps not. When I disable the service in recovery mode I am again able to log in and then use the touchpad by starting the service manually. 
 
I'm happy to provide more info if necessary. 

Thanks!

### Expected behavior

The enabled systemctl service should not break the login process and the session should start.

### Relevant log output

```shell
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input22
INFO:asus-touchpad-numpad-driver:Set touchpad id 17 from H: Handlers=mouse1 event17
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-2238
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-2158
DEBUG:asus-touchpad-numpad-driver:Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"
Exception in thread Thread-1 (check_system_numlock_status):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
Exception in thread Thread-2 (check_touchpad_status_endless_cycle):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
TypeError: __main__.check_system_numlock_status() argument after * must be an iterable, not function
Exception in thread Thread-3 (check_numpad_automatical_disable_due_inactivity):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
Exception in thread Thread-4 (check_config_values_changes):
Traceback (most recent call last):
TypeError: __main__.check_touchpad_status_endless_cycle() argument after * must be an iterable, not function
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self._target(*self._args, **self._kwargs)
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
TypeError: __main__.check_numpad_automatical_disable_due_inactivity() argument after * must be an iterable, not function
    self._target(*self._args, **self._kwargs)
TypeError: __main__.check_config_values_changes() argument after * must be an iterable, not function
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 9 y 1052
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 998
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3932 y 1877
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3906 y 1870
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 911
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 902
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 976
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 0 y 1078
```

### Desktop

- Program version `2.7.3`
- Distro: `Ubuntu 22.04.2 LTS`
- Kernel `5.19.0-43-generic`
- Layout: `up5401ea`

ldrahnik, 2023-06-03:
@isokosan Hello, at first thank you for creating an issue, and sorry for the problems. In the latest master branch mentioned error should not occur, the problem has been fixed in a previously opened issue. Can you try it? And please try to install the driver as root as well. These two use cases. Then please attach logs from driver as you did and systemd booting where should be written why service did not started, in case still did not.

isokosan, 2023-06-03:
Hi @ldrahnik and thanks for the quick response on a Saturday!

I pulled the latest from the master branch and reinstalled. I do not get the errors with "must be an iterable" anymore, however the login process is still broken. When I enable the systemctl service I cannot get past a black screen.
 
The service logs from the failed boot don't give much info:
```
-- Boot a78e24071ee54519bcf73e297f50fd9a --
Jun 03 10:23:11 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:16 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:16 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:16 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.856s CPU time.
Jun 03 10:23:16 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 1.
Jun 03 10:23:16 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:16 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.856s CPU time.
Jun 03 10:23:16 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:21 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:21 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:21 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.856s CPU time.
Jun 03 10:23:22 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 2.
Jun 03 10:23:22 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:22 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.856s CPU time.
Jun 03 10:23:22 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.852s CPU time.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 3.
Jun 03 10:23:27 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.852s CPU time.
Jun 03 10:23:27 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:27 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.859s CPU time.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 4.
Jun 03 10:23:28 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.859s CPU time.
Jun 03 10:23:28 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.852s CPU time.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 5.
Jun 03 10:23:28 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.852s CPU time.
Jun 03 10:23:28 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Main process exited, code=exited, status=1/FAILURE
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Failed with result 'exit-code'.
Jun 03 10:23:28 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.857s CPU time.
Jun 03 10:23:29 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Scheduled restart job, restart counter is at 6.
Jun 03 10:23:29 deniz-zenbook systemd[1]: Stopped Asus Touchpad to Numpad Handler.
Jun 03 10:23:29 deniz-zenbook systemd[1]: asus_touchpad_numpad@deniz.service: Consumed 1.857s CPU time.
Jun 03 10:23:29 deniz-zenbook systemd[1]: Started Asus Touchpad to Numpad Handler.
```

And this is the boot with the black screen:

isokosan, 2023-06-03:
Installed the latest master branch as user and root and in both cases the previous error is resolved:
```
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140D:00 04F3:31B9 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE140D:00/0018:04F3:31B9.0001/input/input22
INFO:asus-touchpad-numpad-driver:Set touchpad id 17 from H: Handlers=mouse1 event17
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-2238
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-2158
DEBUG:asus-touchpad-numpad-driver: Writting to config file: "<_io.TextIOWrapper name='/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev' mode='w' encoding='UTF-8'>"
``` 

However the black screen after login when systemctl is enabled persists. I'm getting an output of the latest journalctl logs from the boot and the service and posting them in a bit.

isokosan, 2023-06-03:
So I have the journalctl logs below from a failed boot with the service enabled, and another with the service disabled attached.

It's probably unrelated but I want to also note that I'm using x11 and not wayland and I haven't tried this in wayland. Also I have displaylink drivers installed for a USB monitor, I also think its unrelated but might be an incompatibility there.
 
Thanks again for your time.

[journalctl-failed-boot.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11642507/journalctl-failed-boot.log)
[journalctl-no-numpad-boot.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11642508/journalctl-no-numpad-boot.log)

ldrahnik, 2023-06-03:
@isokosan At this moment I have no idea why it does not work. Anyway in case boot with enabled drivers service does not work properly  will not work recovering from sleep mode as well. Can you confirm?

isokosan, 2023-06-03:
I'm testing the sleep mode now enabling the service please give me 2 minutes. In the meantime I found the crash report also attaching it here maybe it holds information.

```Jun 03 13:36:50 deniz-zenbook whoopsie-upload-all[1380]: INFO:root:/var/crash/_usr_share_asus_touchpad_numpad-driver_asus_touchpad.py.1000.crash already marked for upload, skipping
`
[crash.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11642522/crash.log)

ldrahnik, 2023-06-03:
@isokosan Do you use only one display? I mean x.11 system environment variable `echo $DISPLAY`, usually `:0`.

isokosan, 2023-06-03:
I get `:1`
I have displaylink installed as I said and have 1 hdmi 1 usd monitor attached. There are some `evdi` lines in the crash log this might be related to displaylink but I have no idea.
The boot issue happens without the monitors attached as well.

ldrahnik, 2023-06-03:
```Traceback:
 Traceback (most recent call last):
   File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 302, in <module>
     with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
 PermissionError: [Errno 13] Permission denied: '/dev/input/event3'```

isokosan, 2023-06-03:
The logs I sent were from installating with the --user flag. I can try a clean root install and we check if the same error occurs or a different issue is coming up then. 

`event3` is owned by root in group `input`

Do I also need to put my user in the `input` group and not just `uinput`?

ldrahnik, 2023-06-03:
@isokosan Please try as root

ldrahnik, 2023-06-03:
@isokosan For user variant permissions should be solved in install script, weird, can you check users groups? (`usermod -a -G "i2c,input,uinput" $RUN_UNDER_USER` [19; install.sh])

isokosan, 2023-06-03:
My user did not have the input or i2c groups. Only the `uinput` was added.
I ran the line manually now and will report if its working now (with --user install again)

isokosan, 2023-06-03:
Unfortunately the issue persist, and I cannot spend more time on the issue. I will try to come back to it when I have time again.

ldrahnik, 2023-06-03:
@isokosan What I would try also is to change the display `Environment="DISPLAY=:0"` variable in `asus_touchpad.X11.service` to discovered correct number of your running x11 instance. Is it hardcored atm to `:0` or better first replace content of `asus_touchpad.X11.service` with content of service without x11 `asus_touchpad.service` to confirm issue is in `x11`.

ldrahnik, 2023-06-03:
@isokosan 
> @ldrahnik What I would try also is to change the display Environment="DISPLAY=:0" variable in asus_touchpad.X11.service to discovered correct number of your running x11 instance.

`git pull` latest master.

isokosan, 2023-06-04:
I pulled the latest changes, ran uninstall and installed again as user and ran into the same issue. I tried as root and ran into the same issue again. I don't have anymore time to spend on this unfortunately so I will not be using the numpad for now - however, I've added journalctl log from the boot and a python crash again.

Also my users groups: `adm cdrom sudo dip plugdev input lpadmin lxd sambashare i2c docker uinput`
[py_crash.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11646525/py_crash.log)
[boot.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11646526/boot.log)

ldrahnik, 2023-06-04:
@isokosan In log Is written user which run thé script has these roles: `UserGroups: adm cdrom dip docker lpadmin lxd plugdev sambashare sudo`. Can ve reopened.

closed 2023-06-04 by ldrahnik

isokosan, 2023-06-05:
That's weird as my current users groups output is `deniz adm cdrom sudo dip plugdev input lpadmin lxd sambashare i2c docker uinput`. The root users groups output is `root : root`. When I have time I'll have a look into it again to see which user is running the python script and why the groups are showing up differently. Thank you again for your time and your quick responses!