asus-linux-drivers/asus-numberpad-driver #119: [Bug]: Unresponsive keyboard and trackpad

athithya12, 2023-06-07:
### Describe the bug

I installed the dev version successfully. Post-reboot, I was able to login successfully. After logging in, the screen freezes and the keyboard and the trackpad becomes unresponsive. I also tried plugging in a mouse, but no luck.

### Expected behavior

The boot should continue normally instead of freezing.

### Relevant log output

```shell
Unable to
```

### Desktop

- Distro [Fedora]
- Kernel [Fedora 38]

ldrahnik, 2023-06-07:
@athithya12 Hello, at first thank you for creating an issue. When you run driver manually vith LOG=DEBUG as is described in Troubleshoting heading in README.md, it works as expected? Strach loga 

When you run

ldrahnik, 2023-06-08:
@athithya12 Can be reopened.

closed 2023-06-08 by ldrahnik