asus-linux-drivers/asus-numberpad-driver #121: installation issue

mariogolf, 2023-06-21:
Dear devs,

when I try to install in an Asus ExpertBook B7402, runnign Ubuntu 23.04 with KDE, via 

> sudo bash ./install.sh --user 
or

> sudo bash ./install.sh

I get this messagge:

> error: externally-managed-environment
× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    See /usr/share/doc/python3.11/README.venv for more information.
note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
pip dependencies via file requirements.txt cannot be loaded correctly.

I installed python3-full package, without success.

How can I fix it?

Thanks!
Mario

ldrahnik, 2023-06-21:
@mariogolf Hello, thank you for creating an issue. Are you able manually install on Ubuntu 23.04 these packages via pip?

```
libevdev
numpy
inotify
python-xlib
```

Try one by one. Looks like problem is one of them `pip dependencies via file requirements.txt cannot be loaded correctly.`.

ldrahnik, 2023-06-21:
@mariogolf Do you use wayland or x.11?

mariogolf, 2023-06-21:
@ldrahnik Thank you for your answer.

This I get:

> # pip install libevdev

Same error previously showed, but there is a new last line:
> hint: See PEP 668 for the detailed specification.

Any suggests?

mariogolf, 2023-06-21:
> @mariogolf Do you use wayland or x.11?

X11

ldrahnik, 2023-06-21:
@mariogolf Why do you use `pip`? Is that equivalent for `python3 -m pip`?

ldrahnik, 2023-06-22:
@mariogolf Please try the latest master. Refactored to using apt repo packages. More details for example [here](https://www.omgubuntu.co.uk/2023/04/pip-install-error-externally-managed-environment-fix).

ldrahnik, 2023-06-22:
@mariogolf Can be reopened.

closed 2023-06-22 by ldrahnik

mariogolf, 2023-06-23:
> @mariogolf Please try the latest master. Refactored to using apt repo packages instead of pip. More details for example [here](https://www.omgubuntu.co.uk/2023/04/pip-install-error-externally-managed-environment-fix).

@ldrahnik Thanks for link: I installed via pipx but, when I run `install-sh --user`, I get the same error also if requirements.txt is empty...
pip still doesn't work...
No fix I found.

Any suggests?
Many thanks for all

ldrahnik, 2023-06-23:
@mariogolf Do not install `pipx`. Just download And install latest master branch of this driver and let me know.

mariogolf, 2023-06-23:
> @mariogolf Do not install `pipx`.
> 
> > Please try the latest master.
> 
> Just download and install latest master branch of this driver and let me know.

New driver installation fixed the problem, thank you!

ldrahnik, 2023-06-23:
@mariogolf Put a star to repository if you like and do not hesitate create new issues when a problem occurs. Glad to help you.