asus-linux-drivers/asus-numberpad-driver #123: Cannot login after installation

maniacx, 2023-06-24:
### Describe the bug

Downloaded from the release package 2.7.6
Installed dependencies.
Switched to X11
Ran installer . 
Choose format up5401ea
Choose y for button linking
Choose y to reboot

After reboot on login screen entered password, and I shows screen with mouse cursur but not moving forward.
Force powerdown by holding POWER button
Power again, same thing.

Booted liveusb mounted my drive and removed systemd service and rebooted , logged in and ran unintstaller

First time installer, so I dont know if the problem was in earlier release.

### Expected behavior

Should  not prevent, the system from booting/logging in, even is there is an error with asus-numberpad-driver

### Desktop

- Program version [2.7.6]
- Distro [Fedora 38]
- Kernel [6.3.8-200.fc38.x86_64]
- Asus Viwobook M3401QC

ldrahnik, 2023-06-24:
@maniacx Hello, thank you for letting me know. Unfortunately I can do nothing more without booting logs then try myself and I have no problem with booting. Elementary OS Lóki Ubuntu 22.04.1.

ldrahnik, 2023-06-25:
@maniacx Can be reopened.

Tahinli, 2023-06-26:
Okay i'm in Pop OS which is gnome and x11.

i installed program with sudo bash ./install, didn't use --user. then rebooted
Login screen is shown but after login, I saw black screen.
Switched to tty2 (ctrl + alt + f2) and desktop is available without logging in or starting gdm.
tty3-4-5-6 are normal and wait for user login as it should be.

in tty2 service is running and numpad in touchpad is working.

I commented standart input line in service, then rebooted. and black screen problem is gone.

Tahinli, 2023-06-26:
I will test in Fedora 38 like @maniacx . I will turn back.

ldrahnik, 2023-06-26:
@Tahinli Well, nice. But did you check driver's log? Needs something to be refactored?

Tahinli, 2023-06-26:
> @Tahinli Well, nice. But did you check driver's log? No errors during using? Needs something to be refactored?

here is output of /var/log/asus_touchpad_nump-driver/error.log

[a (1).txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11874952/a.1.txt)

ldrahnik, 2023-06-26:
@Tahinli Looks like as I thought, unfortunatelly this change will not be enough because this driver has to be connected to standard input (X11 libs and var XDISPLAY) that is reason why now fails xinput, gsettings and reading environment variables I think will fail too (`os.environ.get`). Can you confirm on the same distro when you run Numberpad from terminal with not commented out StandartInput row these errors are not displayed? Because xinput has to be instaled etc.. thank you a lot

ldrahnik, 2023-06-26:
@Tahinli We need compare driver error log from installed driver with commented out row with StandardInput and driver which is running inside terminal like this `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" ""`

Tahinli, 2023-06-26:
these two is coming from ~/.local/share/xorg/

maybe it can be useful 

[Xorg.0.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11875015/Xorg.0.log)
[Xorg.1.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11875017/Xorg.1.log)

after getting this outputs i recorded a video and run driver from terminal, here is the video
https://www.veed.io/view/0bf96679-0ed3-481f-bf1a-1e32e778161a?sharingWidget=true&panel=
after that i did
cat /var/log/asus_touchpad_numpad-driver/error.log | tee Desktop/b.txt
[b.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11875109/b.txt)

> @Tahinli Looks like as I thought, unfortunatelly this change will not be enough because this driver has to be connected to standard input (difference between stylus repo, X11 libs and var XDISPLAY) that is reason why now fails xinput, gsettings and reading environment variables I think will fail too (`os.environ.get`). Can you confirm on the same distro when you run Numberpad from terminal these errors are not displayed in log? Because xinput has to be instaled etc.. thank you a lot

Tahinli, 2023-06-26:
> @Tahinli We need compare driver error log from installed driver with commented out row with StandardInput and driver which is running inside terminal like this `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" ""` on the same gnome distro
> 
> My tip is: Unfortunatelly this change will not be enough because this driver has to be connected to standard input (difference between stylus repo, X11 libs and var XDISPLAY) that is reason why now fails xinput, gsettings and reading environment variables I think will fail too (`os.environ.get`)

sorry for late btw, i was trying to compress video.

I will try this now.

Tahinli, 2023-06-26:
Okay i did : LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" > logs.txt 2>&1

and here the logs, i pressed every button i think.

[loggy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11875570/loggy.txt)

Tahinli, 2023-06-27:
Btw is it about X11 or algorithm itself, i don't know but. Program has 2 things to improve. 

1) When numpad is open, and pressing some numbers program still trying to click.

2) If i press numbers quickly, it doesn't print for a second then after second it prints what i press(If i pressed 3 times, quickly, waits nearly 1 second then print all 3 immediately.). So it's like keeping data in itself for a second then release. I don't know why.

> This problems happen when driver works as a service.
> If I open driver manually, there is no problem.
> That's why, i think problem is not about your algorithm but there are some kind of bugs

At the end it's still working as a service but a bit uncomfortable.

ldrahnik, 2023-06-27:
@Tahinli Y, without connection to StandardInput exists associated bugs as I wrote. As workaround can be changed this value in config file `enabled_touchpad_pointer=2` when will ve allowed use only left, middle, right button pointer buttons or you can disable it entirely. It happens because gsettings is then not able disable tap to click property for Touchpad when activating & activate when disabling.

Tahinli, 2023-06-27:
> @Tahinli
> 
> > 1. When numpad is open, and pressing some numbers, program still trying to click (printing numbers and clicking at the same time).
> 
> Y, without connection to StandardInput exists associated bugs. As workaround can be changed this value in config file `enabled_touchpad_pointer=2` when will ve allowed use only left, middle, right pointer buttons or you can disable it entirely (more in doc in README.md). It happens because gsettings is then not able disable tap to click property for Touchpad when activating & back activate when disabling.

I don't how but in this [implementation(Rust)](https://github.com/iamkroot/asus-numpad). When i press numbers on touchpad, automatically my cursor is gone then when i swipe (without need to close numpad), cursor comes back.

ldrahnik, 2023-06-27:
> I don't how but in this [implementation(Rust)](https://github.com/iamkroot/asus-numpad). When i press numbers on touchpad, automatically my cursor is gone then when i swipe (without need to close numpad), cursor comes back.

Try config value `enabled_touchpad_pointer=1`

ldrahnik, 2023-06-27:
@Tahinli Increase config value `distance_to_move_only_pointer=400` (in px, Is limited by sizes of key area, then is cursor released, when Is crossed border to another key)

Tahinli, 2023-06-27:
Btw, nowadays i don't have to use my computer much, so if you want any help for debugging, i can help (Can try different distros, config, even we can break system). But after that when i start to use computer, there will be some files, maybe documents, so i might not be very open for every test. Until that day, maybe we can solve X11 issues. If you have any idea ?

Tahinli, 2023-06-27:
> @Tahinli
> 
> > > It worked as you said, but needs a bit toleration. Because if i even move really really slightly (we are human :D) it immediately shows cursor.
> 
> > Increase/try to play with config value distance_to_move_only_pointer=400 (in px, Is limited by sizes of key area, then is cursor released anyway;; even the value is higher)
> 
> Should be default value incresead? To which one?

I didn't test much, i was trying to fix fedora with driver. even i wasn't be able to switch tty3-4, driver is completely breaking.

now i fixed with live usb, and i'm going to try to start driver manually. when i fix i try configs

Tahinli, 2023-06-27:
okay in fedora 38 workstation (logged in with x11), there is no numpad backlight.  and when try to close with top right button, it closes but can not be opened again until restart service.

> it happens when we disable standartinput in service file.

Tahinli, 2023-06-27:
/var/log/asusnumpad error
[aFed.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886368/aFed.txt)

my service file
[asus_touchpad_numpad@.service.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886369/asus_touchpad_numpad%40.service.txt)

~/.local/share/xorg/Xorg.0.log
[Xorg.0.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886372/Xorg.0.log)

manually started with LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" and it crashed while i was trying to change brightness (actually there is no backlight)
[floggy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886370/floggy.txt)

manually started with LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" again and it crashed when i tried to close with upper right button
[floggyy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886371/floggyy.txt)

Tahinli, 2023-06-27:
do you want me to do anything on fedora, because i will leave it, if we don't need anything

ldrahnik, 2023-06-27:
@Tahinli I do not have much time now think how solve this black screen problem on Gnome. In case you have I would recommend find some service on Gnome which also requires X11 look how is written, I would do the same if I would have time.

> /var/log/asusnumpad error
[aFed.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886368/aFed.txt)

The error log says what is wrong, is missing `numpy` module which should be installed via pip inside `install.sh`, why has not been module installed? `ModuleNotFoundError: No module named 'numpy'`, also is missing `ModuleNotFoundError: No module named 'Xlib'` and `FileNotFoundError: [Errno 2] No such file or directory: 'xinput'`. Looks like file `requirements` was empty during installation? (for missing Xlib & numpy module). For `xinput`, Fedora does not have this package `dnf -y install xinput` which is in install script? Can you try install it?

This error `(process:4568): dconf-WARNING **: 21:28:34.148: failed to commit changes to dconf: Error spawning command line “dbus-launch --autolaunch=5939fe6a09ee4be38ac2a53a5489d3a3 --binary-syntax --close-stderr”: Child process exited with code 1
Exception in thread Thread-1 (check_system_numlock_status):` is I guess because driver is not connected to `StandardInput`, line was commented out as I see in posted service? right?

This error `ERROR:root:Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 1694, in <module>
    listen_touchpad_events()
  File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 1441, in listen_touchpad_events
    increase_brightness()
  File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 584, in increase_brightness
    if (brightness + 1) >= len(backlight_levels):` Happened because was not caught exception when was not found `xinput`, `git pull` latest master, fixed in code, thank you!

> manually started with LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" again and it crashed when i tried to close with upper right button
[floggyy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886371/floggyy.txt)
This latest report should be fixed by previous fix when was not found `xinput`.

ldrahnik, 2023-06-27:
@Tahinli Can you post me here output of command `systemctl --type=target` run inside Gnome environment?

Tahinli, 2023-06-27:
> @Tahinli
> 
> > /var/log/asusnumpad error
> > [aFed.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886368/aFed.txt)
> 
> The error log says what is wrong, is missing `numpy` module which should be installed via pip inside `install.sh`, why has not been module installed? `ModuleNotFoundError: No module named 'numpy'`, also is missing `ModuleNotFoundError: No module named 'Xlib'` and `FileNotFoundError: [Errno 2] No such file or directory: 'xinput'`. Looks like file `requirements` was empty during installation? (for missing Xlib & numpy module). For missing `xinput`, Fedora does not have this package `dnf -y install xinput` which is in install script? Can you try install it?
> 
> This error `(process:4568): dconf-WARNING **: 21:28:34.148: failed to commit changes to dconf: Error spawning command line “dbus-launch --autolaunch=5939fe6a09ee4be38ac2a53a5489d3a3 --binary-syntax --close-stderr”: Child process exited with code 1 Exception in thread Thread-1 (check_system_numlock_status):` is I guess because driver is not connected to `StandardInput`, line was commented out as I see in posted service? right?
> 
> This error `ERROR:root:Listening touchpad events unexpectedly failed Traceback (most recent call last): File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 1694, in <module> listen_touchpad_events() File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 1441, in listen_touchpad_events increase_brightness() File "/home/tahinli/asus-numberpad-driver/./asus_touchpad.py", line 584, in increase_brightness if (brightness + 1) >= len(backlight_levels):` Happened because was not caught exception when was not found `xinput`, `git pull` latest master, fixed in code, thank you!
> 
> > manually started with LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" "" again and it crashed when i tried to close with upper right button
> > [floggyy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886371/floggyy.txt)
> > This latest report should be fixed by previous fix when was not found `xinput`.
> 
> Unfortunatelly do not have much time think how elegantly solve problem with `StandardInput` now - so how solve this black screen problem on Gnome. In case you have I would recommend find some service on Gnome which also requires X11 look how is written, I would do the same if I would have time.

interesting thing is i have numpy but didn't have xinput. I installed it now.

Tahinli, 2023-06-27:
> @Tahinli Can you post me please here output of command `systemctl --type=target` run inside Gnome environment?

sure it's
[target.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11887385/target.txt)

ldrahnik, 2023-06-27:
@Tahinli These commands should print `x11` string, is it true also on Fedora?

```
session_id=$(loginctl | grep <root|user_name> | head -1 | awk '{print $1}') # change line <root|user_name> to root OR user_name
wayland_or_x11=$(loginctl show-session $session_id -p Type --value)
echo $wayland_or_x11
```

Tahinli, 2023-06-27:
we can use 

> echo $XDG_SESSION_TYPE

if you want me to check, i am in wayland or x11.

I'm in x11

ldrahnik, 2023-06-27:
@Tahinli I asked because then should be run this block of code with installing `xinput`

```
    if [ "$wayland_or_x11" = "x11" ]; then
        dnf -y install xinput
    fi
```

Tahinli, 2023-06-27:
> > @Tahinli
> > > /var/log/asusnumpad error
> > > [aFed.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11886368/aFed.txt)
> > 
> > 
> > The error log says what is wrong, is missing `numpy` module which should be installed via pip inside `install.sh`, why has not been module installed? `ModuleNotFoundError: No module named 'numpy'`, also is missing `ModuleNotFoundError: No module named 'Xlib'` and `FileNotFoundError: [Errno 2] No such file or directory: 'xinput'`. Looks like file `requirements` was empty during installation? (for missing Xlib & numpy module). For missing `xinput`, Fedora does not have this package `dnf -y install xinput` which is in install script? Can you try install it?
> 
> interesting thing is i had numpy but didn't have xinput. I installed it now.

after i installed xinput and restart my backlight is here and do not crash when i press backlight button or open/close button.

Tahinli, 2023-06-27:
> @Tahinli I asked because then should be run this block of code with installing `xinput`
> 
> ```
>     if [ "$wayland_or_x11" = "x11" ]; then
>         dnf -y install xinput
>     fi
> ```

I had similar thing that maybe you can use, https://github.com/Tahinli/Shells/blob/main/chad.sh

ldrahnik, 2023-06-27:
@Tahinli 

>> we can use

>echo $XDG_SESSION_TYPE

>if you want me to check, i am in wayland or x11.

I think we can not because when install script is called as sudo then is variable `$XDG_SESSION_TYPE empty

Tahinli, 2023-06-27:
> @Tahinli
> 
> > > we can use
> 
> > echo $XDG_SESSION_TYPE
> 
> > if you want me to check, i am in wayland or x11.
> 
> I think we can not because when install script is called as sudo then is variable `$XDG_SESSION_TYPE` empty

Are you sure, because now i just tried in Arch and Fedora

>sudo echo $XDG_SESSION_TYPE

and they both print x11 to me

ldrahnik, 2023-06-27:
@Tahinli 

```
# this works because sudo sets the environment variable SUDO_USER to the original username
session_id=$(loginctl | grep $SUDO_USER | head -1 | awk '{print $1}')
wayland_or_x11=$(loginctl show-session $session_id -p Type --value)
```

If you are able to simplify these lines using one variable I will merge it

Tahinli, 2023-06-27:
> @Tahinli
>
>
> The error log says what is wrong, is missing `numpy` module which should be installed via pip inside `install.sh`, why has not been module installed? `ModuleNotFoundError: No module named [...]

btw i realised that, fedora hasn't got pip by default, maybe that's why it has some installation issue.
Because while installing it didn't install numpy, i2c, libedev, inotify and i installed them with dnf install python3-numpy etc.

ldrahnik, 2023-06-27:
> btw i realised that, fedora hasn't got pip by default, maybe that's why it has some installation issue.
Because while installing it didn't install numpy, i2c, libedev, inotify and i installed them with dnf install python3-numpy etc.

The same, I will merge PR in case you will change pip packages to dnf packages, the same I has to reimplement for Ubuntu23.04 where is preferred to use apt.

Tahinli, 2023-06-27:
> > btw i realised that, fedora hasn't got pip by default, maybe that's why it has some installation issue.
> > Because while installing it didn't install numpy, i2c, libedev, inotify and i installed them with dnf install python3-numpy etc.
> 
> The same, I will merge PR in case you will change pip packages to dnf packages, the same I had to reimplement for Ubuntu23.04 where is preferred to use apt.
> 
> Is required check names of packages as I did for apt if exist (`echo 'apt is here' && apt -y install ibus libevdev2 i2c-tools python3-dev python3-libevdev python3-numpy python3-xlib python3-pyinotify`) for Fedora under dnf and remove these lines, that is all:
> 
> ```
>     runuser -u $RUN_UNDER_USER -- python3 -m pip install -r requirements.txt
> 
>     if [[ $? != 0 ]]; then
>         echo "pip dependencies via file requirements.txt cannot be loaded correctly."
>         exit 1
>     fi
> ```

Okay, there is a special day tomorrow in my country. So when i have time i will prepare and test with freshly installed fedora 38 then make pull request.

daurmur, 2023-06-28:
Hello!
I encountered the same problem. After I removed the service file from /etc/systemd/system I wanted to run uninstall.sh, but it produced the following output:
`
Driver will be stopped and uninstalled for user
root
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
Failed to stop asus_touchpad_numpad@root.service: Unit asus_touchpad_numpad@root.service not loaded.
asus_touchpad_numpad.service cannot be stopped correctly...
`
Do I need to do something to uninstall this driver completely to return to the state before the installation? What should I delete too?

ldrahnik, 2023-06-28:
@daurmur Only run uninstall script with `--user` or without, depends on how was driver installed (with, or without `--user`. Dependency packages will not be installed but everything else will be

maniacx, 2023-06-28:
> Hello!
> I encountered the same problem. After I removed the service file from /etc/systemd/system I wanted to run uninstall.sh, but it produced the following output:
> `
> Driver will be stopped and [...]

Yes. I had the same issue.
Due to failure of stopping the systemd service the script just exits with code1 instead  continuing and removing all other files.

So I had to read installer and uninstaller script and find out which files were installed and remove them all manually.

A suggestion to the dev would be not to exit on failure of stopping  asus_touchpad_numpad@root.service as user may have manually stopped it or even removed it due to a bug.
Instead of exiting, just report to the user, that it could be stopped and , just continue removing other files that were installed.

ldrahnik, 2023-06-28:
@maniacx @daurmur `git pull` latest master

Tahinli, 2023-06-29:
> > @Tahinli
> > > > we can use
> > 
> > 
> > > echo $XDG_SESSION_TYPE
> > 
> > 
> > > if you want me to check, i am in wayland or x11.
> > 
> > 
> > I think we can not because when install script is called as sudo then is variable `$XDG_SESSION_TYPE` empty
> 
> Are you sure, because now i just tried in Arch and Fedora
> 
> > sudo echo $XDG_SESSION_TYPE
> 
> and they both print x11 to me
> 
> here is fedora
> 
> [image]

you were right, $XDG_SESSION_TYPE returns empty, if we are in sudo mod. when i did 

> sudo echo $XDG_SESSION_TYPE

it just runs echo with sudo privilege, i couldn't realize that yesterday.

but i changed pip packets with dnf for fedora and pacman for arch, i will control them tomorrow.

> loginctl show-session $(loginctl | grep $(logname) | awk '{print $1}') -p Type

in addition also this command shows which one (x11-wayland) we are using. and also works in non sudo mod as a bonus. maybe it can be useful. but your code 

> session_id=$(loginctl | grep $SUDO_USER | head -1 | awk '{print $1}')
> wayland_or_x11=$(loginctl show-session $session_id -p Type --value)

is much more simple then other code, because we have to parse output correctly for other one.

ldrahnik, 2023-06-29:
@Tahinli Are you sure with `python3-inotify` package instead of `python3-pynothify` for Fedora? It works?(dnf)

ldrahnik, 2023-06-29:
@Tahinli Can you please (maybe one more time) confirm, the service with `StandardInput=tty-force` is a problem (black screen) and does not matter what program is executed inside a service like this on Gnome?

Tahinli, 2023-06-29:
> @Tahinli Can you please (maybe one more time) confirm, the service with `StandardInput=tty-force` is a problem (black screen) and does not matter what program is executed inside a service like this on Gnome?

for x11, seems yes.

Tahinli, 2023-06-29:
> @Tahinli Are you sure with `python3-inotify` package instead of `python3-pyinotify` for Fedora? It works?(dnf)

yes. for arch it's pyinotify but fedora its just inotify.

in addition dnf detection has problem(that is why it didn't installed python libraries at the first time), i solved it and will make pull request.

Tahinli, 2023-06-29:
btw i just did my very first 2 pull requests to you. and i don't know should i keep my forks :D

Tahinli, 2023-06-29:
and in wayland swipe left icon do not open calculator. i checked logs, it detect swipe action but there is no calculator. if you want i can open new issue for it.

ldrahnik, 2023-06-29:
@Tahinli You can delete the fork when is pull request merged, then is source code merged into the repository

ldrahnik, 2023-06-29:
@Tahinli 
> and in wayland swipe left icon do not open calculator. i checked logs, it detect swipe action but there is no calculator. if you want i can open new issue for it.

Yeah, you can open issue. In every distro should be send `KEY_CALC` key by default (can be changed to another key) and then it is up to user set up script which open calculator, install script handles only adding script for toggling calculator via `gsettings` and asks for permissions do that. So maybe was `gsettings` missing on that distribution? But it may be bug.

Tahinli, 2023-06-29:
> > @Tahinli Can you please (maybe one more time) confirm, the service with `StandardInput=tty-force` is a problem (black screen) and does not matter what program is executed inside a service like this on Gnome?
> 
> for x11, seems yes. i dont know many program but, asus stylus and numpad drivers both has problem with standartinput in x11. in wayland it does not left us in black screen but make opening slower than normal (very slow actually but at the end it opens)

making opening slower is not standartinput issue, i analyzed it wrong, there should be another problem.

ldrahnik, 2023-06-29:
@Tahinli 

> also different keyboard layouts still act different. printing comma instead of dot for example.

On wayland or X11?

Tahinli, 2023-06-29:
> @Tahinli
> 
> > also different keyboard layouts still act different. printing comma instead of dot for example.
> 
> On wayland or X11?

both

ldrahnik, 2023-06-29:
>> @Tahinli

>>> also different keyboard layouts still act different. printing comma instead of dot for example.

>> On wayland or X11?

> both

Then feel free create another issue, bug logs are required. Without that I can do nothing.

ldrahnik, 2023-06-29:
@Tahinli I will not gonna spam more this issue thread, the issue thread should keep the discussion to the title. You can email me if you want, my email is public. In case you will have time debug problem about black screen we can continue here, I need to know if on Gnome starts service with empty command bash for example `ExecStart`  (without driver) but with StandardInput and with these variables Environment="DISPLAY=$XDISPLAY" Environment="XAUTHORITY=$XAUTHORITY". Just what is minimal bug reproduce example when it starts failing during booting.

Tahinli, 2023-06-29:
> @Tahinli I will not gonna spam more this issue thread, the issue thread should keep the discussion to the title. You can email me if you want, my email is public. In case you will have time debug problem about black screen we can continue here, I need to know if on Gnome starts service with `ExecStart` as empty command bash for example (so without driver code) but with `StandardInput=tty-force` and with these variables Environment="DISPLAY=$XDISPLAY" Environment="XAUTHORITY=$XAUTHORITY". Just we need find out what is minimal bug reproduce example when it starts failing during booting.

you are definitely right about staying in subject for this thread. I commented about another problems a lot. Sorry for that.

I'm going to help with standartinput problem, when i'm ready.

closed 2023-06-29 by ldrahnik

maniacx, 2023-07-02:
@ldrahnik 
Just an update.
I had to reformat my drive as I was re-installing hackintosh, so decide to install Ubuntu 23.04 instead of fedora 38 (after installing hackintosh).
Tried the numpad driver. Installation gave me some message about systemd cannot be started and my device couldnt be detected. However after reboot, the numpad worked correctly on X11 Ubuntu 23.04.
Thanks.
I am happy with the driver. However I just saved the message while installing just for your reference. I didnt use the release, I just use git clone. Last commit below

```
commit 568bd88120c65ef06425993e717ac52075b45f28 (HEAD -> master, origin/master, origin/HEAD)
Author: Lukáš Drahník <ldrahnik@gmail.com>
Date:   Sun Jul 2 14:55:20 2023 +0200

    #128 adding shortcut clean up when installing and uninstalling (is required reboot or logout)

commit 95c326a23ae4b3c81f1f3f77017cba3124155e0b (tag: v2.8.0)
Author: Lukáš Drahník <ldrahnik@gmail.com>
Date:   Sat Jul 1 16:50:38 2023 +0200

    Bump to v2.8.0
```

Installation messages
```
KERNEL=="uinput", GROUP="uinput", MODE:="0660"
driver will run under user maniacx
x11 is DETECTED
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ibus is already the newest version (1.5.28-3).
libevdev2 is already the newest version (1.13.0+dfsg-1).
i2c-tools is already the newest version (4.3-2build3).
python3-dev is already the newest version (3.11.2-1).
python3-libevdev is already the newest version (0.5-3).
python3-numpy is already the newest version (1:1.24.2-1).
python3-xlib is already the newest version (0.33-2).
python3-pyinotify is already the newest version (0.9.6-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput is already the newest version (1.6.3-1build2).
xinput set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Testing interface i2c-0 : sucess
Detected laptop: Vivobook_ASUSLaptop M3401QC_M3401QC
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 52107    0 52107    0     0  41378      0 --:--:--  0:00:01 --:--:-- 41387
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 84271    0 84271    0     0  77084      0 --:--:--  0:00:01 --:--:-- 77100
Could not automatically detect numpad layout for your laptop. Please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues).

Select your model keypad layout:
1) b7402.py	       5) gx551.py	     9) up5401ea.py
2) e210ma.py	       6) gx701.py	    10) ux433fa.py
3) g513.py	       7) up5401ea-3145.py  11) ux581l.py
4) g533.py	       8) up5401ea-31B9.py  12) Quit
Please enter your choice 9
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
X11 is detected
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]N
Systemctl daemon realod called succesfully
Created symlink /etc/systemd/system/graphical.target.wants/asus_touchpad_numpad@maniacx.service → /etc/systemd/system/asus_touchpad_numpad@.service.
Asus touchpad numpad service enabled
Asus touchpad numpad service started
Install process requested to succesfull finish atleast log out or reboot
Without that driver might not work properly
Do you want reboot now? [y/N]N
Install finished
```

ldrahnik, 2023-07-02:
@maniacx Thank you for letting me know. Super.

ldrahnik, 2023-07-02:
@maniacx I can add your laptop to table for autodetection `Could not automatically detect numpad layout for your laptop.`. Everything what I need is output of this command `$ sudo dmidecode -s system-product-name`

maniacx, 2023-07-02:
Ok
```
sudo dmidecode -s system-product-name
[sudo] password for maniacx: 
Vivobook_ASUSLaptop M3401QC_M3401QC
```

ldrahnik, 2023-07-02:
@maniacx Added, please only check backlight levels are OK (hold top left icon when is NumberPad activated) because already exist 2 layouts which differ (I do not know why) only in these levels. Brightness should be increased step by step from lowest to full backlight touchpad brightness.

[1 more reply]