asus-linux-drivers/asus-numberpad-driver #125: Keyboard Layout Problem

Tahinli, 2023-06-29:
Normally i use Turkish Q keyboard. when i press "dot" in numpad, it prints "comma"
If i switch to the English keyboard. there is no problem.

Here are the logs:

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11905411/error.log)

ldrahnik, 2023-06-29:
@Tahinli Replicated, thank you for report

ldrahnik, 2023-06-29:
@Tahinli Every layout at this moment relies on the fact that `KEY_KPxyz` keys are equal in all layouts and are not rebinded. Workaround untill will be solved is to use "." instead of EV_KEY.KEY_KPDOT inside layout file which is used.

ldrahnik, 2023-06-29:
@Tahinli `git pull` latest master

ldrahnik, 2023-06-29:
@Tahinli Because send one key event is less then send unicode number of the same key via Shift+Ctrl+U+number (5 key events vs 1)

Tahinli, 2023-06-29:
> @Tahinli Because send one key event is less then send unicode number of the same key via Shift+Ctrl+U+number (5 key events minimum vs 1)

Makes sense, thanks

Tahinli, 2023-06-29:
it crashes, when i print "+", "." , "-", "*", "/", "="

i will try to capture logs

Tahinli, 2023-06-29:
here my service file
[asus_touchpad_numpad@.service.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11906543/asus_touchpad_numpad%40.service.txt)

ldrahnik, 2023-06-29:
@Tahinli wrong variable name is there, `xdisplay` vs `display`, do you want make PR?

Tahinli, 2023-06-29:
> @Tahinli wrong variable name is there, `xdisplay` vs `display`, do you want make PR?

i would be very happy.

Tahinli, 2023-06-29:
strange thing is fedora didn't show me black screen even standartinput is not commented

ldrahnik, 2023-06-29:
@Tahinli Maybe this misstyped variable name, bug created by me in version 2.7.4 is responsible for that blackscreen.

Tahinli, 2023-06-29:
> @Tahinli wrong variable name is there, `xdisplay` vs `display`, do you want make PR?

maybe i did wrong but, can you check

 https://github.com/Tahinli/asus-numberpad-driver/commit/123f16f106ebb5933c994dec0c2534d629280af4

don't you mean that.

ldrahnik, 2023-06-29:
@Tahinli 
> maybe i did wrong but, can you check

Is now display variable filled ok to service?

> don't you mean that.

> because still xdisplay is not valid

I do not know, I will wait until you will analyze what has to be in service to fail during booting.

ldrahnik, 2023-06-29:
@Tahinli Change even this please

`cat asus_touchpad.X11.service | CONFIG_FILE_DIR="$CONFIG_FILE_DIR/" USER=$RUN_UNDER_USER LAYOUT=$model XDISPLAY=$xdisplay XAUTHORITY=$xauthority envsubst '$LAYOUT $XAUTHORITY $CONFIG_FILE_DIR' > /etc/systemd/system/asus_touchpad_numpad@.service`

to 

`cat asus_touchpad.X11.service | CONFIG_FILE_DIR="$CONFIG_FILE_DIR/" USER=$RUN_UNDER_USER LAYOUT=$model XDISPLAY=$xdisplay XAUTHORITY=$xauthority envsubst '$LAYOUT $XAUTHORITY $XDISPLAY $CONFIG_FILE_DIR' > /etc/systemd/system/asus_touchpad_numpad@.service`

Tahinli, 2023-06-29:
oaky display worked. I'm happy that, also black screen is resolved. I will try with fresh install again for black screen.

ldrahnik, 2023-06-29:
@Tahinli 
> display worked. I'm happy that, also black screen is resolved. I will try with fresh install again for black screen.

Nice, let me know.

Tahinli, 2023-06-29:
> @Tahinli
> 
> > display worked. I'm happy that, also black screen is resolved. I will try with fresh install again for black screen.
> 
> Nice, let me know.

happy news. no black screen

ldrahnik, 2023-06-30:
@Tahinli So now is driver log clear? No error there? You could now try IMHO best option of config value `enabled_touchpad_pointer=3`, does it work? Does it also fixed problem with toggling calculator by swiping top left icon?

Tahinli, 2023-06-30:
> @Tahinli So now is driver log clear? No error there? You could now try IMHO best option of config value `enabled_touchpad_pointer=3`, does it work? Does it also fixed problem with toggling calculator by swiping top left icon?

I'm not available now. Aproximately after 8 hours, i can check. Funny thing is calculator was working afaik. I dont know why but even without update, after fresh install there were no calculator or slow opening issue. When i said calculator wasnt working and wayland was opening slowly, there should have been another thing. Like i said lastly there were no issue. I will come with logs and much more verified information when i am available

Tahinli, 2023-06-30:
Wayland Fresh install Fedora, I installed stylus driver first, then reboot, installed numpad, then reboot. Here the logs.
When i swiped for calculator, it crashed
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11923194/error.log)

Tahinli, 2023-06-30:
Last few days i installed fedora maybe over 10 times. how every setup can show different results, i will lost my mind

ldrahnik, 2023-06-30:
@Tahinli Head up. Error in the log does make sense. It was the first attempt on Wayland? I will react tomorrow.

Tahinli, 2023-06-30:
ldrahnik@gmail.com this is your mail right, if it is, can i use it to communicate with you instead of spamming unrelated things under issues.

I sent a mail 6/6/23. it was about stylus pen and didn't know what to do that time (later i solved, it was hardware issue and now  there is no problem), you didn't send any mail after that, you could be busy no problem, is this your mail right ?

ldrahnik, 2023-06-30:
@Tahinli Error from the last attached log should be fixed,`git pull` latest master.

Tahinli, 2023-06-30:
> @Tahinli Error from the last attached log should be fixed,`git pull` latest master and confirm please.

i didn't try it with fresh install but, 

> git pull
> sudo ./install

then reboot

no crash but no calculator

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11923695/error.log)

ldrahnik, 2023-07-01:
@Tahinli the same error as is displayed in log ```(process:3619): dconf-WARNING **: 02:17:34.262: failed to commit changes to dconf: Cannot autolaunch D-Bus without X11 $DISPLAY`` is shown when you run in terminal this command for example?

```
           (sudo) gsettings set org.gnome.settings-daemon.plugins.media-keys calculator-static [\'\']
```

Tahinli, 2023-07-01:
still it's not openingi and here the logs
[loggy.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11925558/loggy.txt)

ldrahnik, 2023-07-01:
@Tahinli I need output of mentioned command from your terminal. Sorry for confusion.

Tahinli, 2023-07-01:
no output
[image]

ldrahnik, 2023-07-01:
@Tahinli Please reinstall (uninstall, then install) the driver, reboot And do not(!!) even activate NumberPad and please attach driver log `/var/log/asus_touchpad_numpad-driver/error.log`

ldrahnik, 2023-07-01:
@Tahinli Then please activate / deactivate and attach log again whether error shown up.

Tahinli, 2023-07-01:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11925821/error.log)

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11925823/error.log)

ldrahnik, 2023-07-01:
@Tahinli 

And `gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click true` and `gsettings set org.gnome.desktop.peripherals.touchpad tap-to-click false` throw error it terminal or not? What is in variable DISPLAY inside your terminal `echo $DISPLAY`?

Tahinli, 2023-07-01:
[image]

ldrahnik, 2023-07-01:
@Tahinli Looks like driver detected wayland, so was used service without variable $DISPLAY and then gsetting could not work because had no XDISPLAY but terminal has this variable.  I am confused. Do you use wayland? Does have this change effect? So you can not tap to click via touchpad? On wayland?

Tahinli, 2023-07-01:
I'm in wayland, normally i use tap to click but now it's disable probably because of command that you send.

ldrahnik, 2023-07-02:
@Tahinli please comment out these lines `set_touchpad_prop_tap_to_click(0|1)` in activate and deactivate functions and reinstall, so this function will not be used. For check my prediction is correct :)

Tahinli, 2023-07-02:
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11929357/error.log)
[errorrrr.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11929358/errorrrr.log)

no calculator still

ldrahnik, 2023-07-02:
@Tahinli Attached error has nothing to do with toggling calculator, for debugging toggling calculator try manually run script for toggling first located in directory where is NumberPad installed: `bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh`, then check key `XF86Calculator` is send always when by using top left icon on activated Numberpad with running `libinput debug-events` listener on background, you should see ` ``event23  KEYBOARD_KEY            +12.282s	KEY_CALC (140) pressed
 event23  KEYBOARD_KEY            +12.282s	KEY_CALC (140) released```, after that check via `dconf` block of added custom shortcuts - should be there only 1 NumberPad record for toggling calculator open dconf -> path is `/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings`. Then it should work.

ldrahnik, 2023-07-02:
@Tahinli This record you should have there

[image]

Tahinli, 2023-07-04:
> @Tahinli Nice, error missed when are lines commented out. But attached error has nothing to do with toggling calculator, for debugging toggling calculator try manually run script for toggling first located in directory where is NumberPad installed: `bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh`, then check key `XF86Calculator` is send always when you do slide gesture from top left icon on activated Numberpad with running `libinput debug-events` listener on background, you should see
> 
> ```
> event23  KEYBOARD_KEY            +12.282s	KEY_CALC (140) pressed
>  event23  KEYBOARD_KEY            +12.282s	KEY_CALC (140) released\
> ```
> 
> after that check via `dconf` block of added custom shortcuts - should be there only 1 NumberPad record for toggling calculator open dconf -> path is `/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings`. Then it should work.

Firstly sorry for late, i have many exams.
> bash /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh
it gave no output and no calculator.
[image]

and swiping top left button triggers key_calc as you said, so there should be another problem instead of your driver.

here is my dconf read output.
[image]

ldrahnik, 2023-07-05:
@Tahinli Which calculator Fedora uses? Is it `gnome-calculator`?  You can cat the installed script (`cat /usr/share/asus_touchpad_numpad-driver/scripts/calculator_toggle.sh`):

`#!/bin/bash

if [[ $(pgrep -lf gnome-calculator) ]]; then
	killall gnome-calculator &>/dev/null &
else
	gnome-calculator &>/dev/null &
fi`

and debug.

ldrahnik, 2023-07-10:
@Tahinli Had you time try it?

Tahinli, 2023-07-10:
> @Tahinli Had you time try it?

Tomorrow i hope

Tahinli, 2023-07-11:
[image]

ldrahnik, 2023-07-11:
@Tahinli In install script `install.sh` is condition for this, code below has to be true in your laptop in case Elementary script is installed. Test it please:

```
if [[ $(type flatpak 2>/dev/null && flatpak list | grep io.elementary.calculator 2>/dev/null) ]]; then
            echo "io.elementary.calculator here"
```

Tahinli, 2023-07-11:
> @Tahinli In install script `install.sh` is condition for this, code below has to be true in your laptop in case Elementary script is installed. Test it please:
>
> ```
> if [[ $(type flatpak 2>/dev/null [...]

I wrote this in shell script (and added fi to end of the line)

it returned 

> io.elementary.calculator here

Tahinli, 2023-07-11:
when i did 

> flatpak run io.elementary.calculator

it says not installed.

ldrahnik, 2023-07-11:
@Tahinli `flatpak list | grep io.elementary.calculator` output is?

Tahinli, 2023-07-11:
I just have few apps

[image]

ldrahnik, 2023-07-11:
@Tahinli Will check tommorow, late night here.

ldrahnik, 2023-07-12:
@Tahinli Do the same as me please in your terminal and post screenshot.

[image]

`

Tahinli, 2023-07-12:
Sorry, fedora broke itself first then broke me again, i had to reinstall. i'm in wayland gnome again. i don't know if it's important

[image]

ldrahnik, 2023-07-12:
@Tahinli I close this issue as last unresolved gsettings error is in discussion in [this issue](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/130)  and for shortcuts looks like nothing wrong in driver code.

closed 2023-07-12 by ldrahnik