asus-linux-drivers/asus-numberpad-driver #127: [Bug]: since install, windows/super L to lock screen does not work anymore

beankylla, 2023-06-30:
### Describe the bug

since install, windows/super L to lock screen does not work anymore

### Expected behavior

windows + L should lock computer

### Relevant log output

```shell
no clue what log is relevant here.

Here is the log of sudo libinput debug-events

-event2   KEYBOARD_KEY            +0.000s	KEY_LEFTMETA (125) pressed
 event2   KEYBOARD_KEY            +0.261s	*** (-1) pressed
l event2   KEYBOARD_KEY            +0.342s	*** (-1) released
 event2   KEYBOARD_KEY            +0.561s	KEY_LEFTMETA (125) released
 event2   KEYBOARD_KEY            +2.578s	*** (-1) pressed
 event2   KEYBOARD_KEY            +2.728s	*** (-1) pressed
^C
```

### Desktop

- Program version [2.7.8]
- Distro [Fedora 38 Gnome]
- Kernel [Kernel: 6.3.8-200.fc38.x86_64 ]

ldrahnik, 2023-06-30:
@beankylla Hello there, thank you for creating ani issue. Does I understand well when is driver uninstalled then shortcut again works?

beankylla, 2023-06-30:
Hi!
i also notice that my hotkeys for screen brightness sound are not working anymore. The keyboard brightness one still works.
i tried to look in this file:

/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev

but did not find anything that looks like the calculator option

[main]```

numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 60
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_x_ratio = 0.3
top_left_icon_slide_func_activation_y_ratio = 0.3
top_right_icon_slide_func_activation_x_ratio = 0.3
top_right_icon_slide_func_activation_y_ratio = 0.3
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 0
default_backlight_level = 0x01
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
```

ldrahnik, 2023-06-30:
@beankylla By settings I mean your distribution settings, not driver config file. In case you answered Y to this question:

```Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]```

during installation of this driver (`install.sh` script) was created record inside custom shortcuts. Try to remove them too.

I recommend use `dconf` tool and path is `/org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/`, remove these records:

[image]

beankylla, 2023-06-30:
Thanks a lot for the detailed explanations.

I reinstalled with calculator
removed the entries as instructed
uninstalled
rebooted

still does not work :/

beankylla, 2023-06-30:
and now i have my happy numpad and hotkeys working i'll call it a day thanks! :+1:

closed 2023-06-30 by ldrahnik