asus-linux-drivers/asus-numberpad-driver #129: [Bug]: Asus numberpad driver service preventing login

fededileo, 2023-07-02:
### Describe the bug

Hi! I'm running Fedora 38 on my UX425QA. I installed the driver just like the description says. However, after first reboot post-installation causes the system to freeze when logging in. Input password, enter, freeze. Even trying to open terminal without logging in causes freezing. I solved it by running a live F38 iso and deleting the asus numberpad service located in `/etc/systemd/system`. I'm not an advanced Linux user nor a programmer so probably couldn't give you much more information than this. 
Keep on your work, this driver is very useful and i'd be glad to donate if the issue gets solved :)

### Expected behavior

I expected the driver to work.

### Desktop

- Distro [e.g. Fedora Workstation 38]
- Kernel [e.g. 6.3.8-200.fc38.x86_64]

ldrahnik, 2023-07-02:
@fededileo Hello, thank you for creating an issue and for your kind words. Could you please attach here output of log of this driver located here `cat /var/log/asus_touchpad_numpad-driver/error.log`? And then please post the output of the installation script? You can immediately uninstall then before you will do restart.

ldrahnik, 2023-07-02:
@fededileo Also please let me know what version do you use. Latest `git pull master`?

YanaiKlugman, 2023-07-03:
@ldrahnik Hello, I've also used this most recent version of the driver on an Asus Zephyrus gx501vik using the GX701 layout. While the driver does work, and the correct key does turn on the backlit numberpad, after entering my password I'm met with a black screen. 

The only way I've found to fix this is by booting in recovery mode and running the uninstall script as root. I'm on the latest Ubuntu 23.04.

ldrahnik, 2023-07-03:
@YanaiKlugman 

> I've looked for any log files in /var/log/ but the uninstall script seems to remove them.

Ok, please attach the content of the driver error log from quick using between install/uninstalling. Please also attach entire output from running the install script.

YanaiKlugman, 2023-07-03:
./install.sh output:
[install_output.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11936946/install_output.txt)

error log from /var/log/asus-touchpad_numpad-driver/
[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11936956/error.log)

ldrahnik, 2023-07-03:
@YanaiKlugman Please install driver, then activate NumberPad, do something, click all keys for example and then uninstall and attach config file. Everything looks ok up to now. 

// I have added autodetection for your laptop numberpad layout - for that use latest driver version

fededileo, 2023-07-03:
> @fededileo Also please let me know what version of driver do you use. Latest `git pull master`?

I did `git clone`, didn't download stable.

ldrahnik, 2023-07-03:
@YanaiKlugman

> time after the Ubuntu login screen I see only my cursor and a black/grey screen. 

Which distribution do you use? Ubuntu?

ldrahnik, 2023-07-03:
@fededileo Please install driver (attach entire output from installation), activate and use for some time, and then attach here error file `cat /var/log/asus_touchpad_numpad-driver/error.log` and uninstall before reboot.

YanaiKlugman, 2023-07-03:
@ldrahnik I'm using the latest Ubuntu release 23.04 with Xorg (non-wayland) 

Doesn't uninstalling remove the error log? I'll try your suggestion tomorrow and update.

ldrahnik, 2023-07-04:
@YanaiKlugman

>  Doesn't uninstalling remove the error log?

It does.

YanaiKlugman, 2023-07-04:
Just cloned the updated repo and ran the install script as user right now. The numperpad works perfectly. 
Here is the /var/log/asus.../error.log file before uninstalling. I opened the default gnome calculator and used the numpad randomly for a bit. I'll reboot now to see if I encounter the same problem and update you in a few moments. 

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11946491/error.log)

YanaiKlugman, 2023-07-04:
[image]

Unfortunately same bug still occurs

ldrahnik, 2023-07-04:
@YanaiKlugman I need to see logs from booting then.

ldrahnik, 2023-07-04:
@YanaiKlugman Using `journactl`, `man journalctl` page or google command for your specific distribution

ldrahnik, 2023-07-04:
@YanaiKlugman Also please attach `cat /etc/systemd/system/asus_touchpad_numpad@.service`

YanaiKlugman, 2023-07-04:
Here is the output from my previous boot (current boot is after uninstalling the driver)
Output from: "journalctl -b -1 | less"
[boot_log.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/11946989/boot_log.txt)

For the .service file I'll need to install again, just a moment.

YanaiKlugman, 2023-07-04:
@ldrahnik 
  /etc/systemd/system                                                                                                                                                      
❯ cat asus_touchpad_numpad@.service      
[Unit]
Description=Asus Touchpad to Numpad Handler
BindsTo=graphical.target

[Service]
User=%i
Type=idle
ExecStart=/usr/bin/env python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py gx701 /usr/share/asus_touchpad_numpad-driver/
StandardInput=tty-force
StandardOutput=append:/var/log/asus_touchpad_numpad-driver/error.log
StandardError=append:/var/log/asus_touchpad_numpad-driver/error.log
TimeoutSec=5
Restart=on-failure
Environment="DISPLAY=:1"
Environment="XAUTHORITY=/run/user/1000/gdm/Xauthority"

[Install]
WantedBy=graphical.target%

ldrahnik, 2023-07-04:
@YanaiKlugman 

> `WantedBy=graphical.target%`

Percent in the end should not be there, only misclicked when attaching here I guess?

> `Environment="DISPLAY=:1"`

Do you use external monitor? This might be problem, in case you are logging to another DISPLAY then to was installed this driver.

YanaiKlugman, 2023-07-04:
The laptop is connected to an external monitor with HDMI, but the primary display is the built-in screen. I'll try again without the second monitor. 

The % may be a misclick but I'll double check.

YanaiKlugman, 2023-07-04:
@ldrahnik 
A band-aid solution could be to add a daemon that uninstalls the driver before rebooting/powering off and installs it again on boot. But, I'm not sure how practical that would be.

ldrahnik, 2023-07-04:
@YanaiKlugman Unfortunately I do not see any error in the booting log. Everything looks ok. Workaround is after installation disable service manually `systemctl disable asus service name` and add a script that will start this service via startup apps for example.

ldrahnik, 2023-07-04:
@YanaiKlugman Somewhere in system (startup apps for example) put this line `$ sudo systemctl start asus_touchpad_numpad@root.service` and after installation run `$ sudo systemctl disable asus_touchpad_numpad@root.service` in case driver was installed as root. It should solve your issue.

ldrahnik, 2023-07-04:
@YanaiKlugman On Ubuntu23 does not exist `XAUTORITY` here `/home/$user/.Xauthority` anymore?

ldrahnik, 2023-07-04:
@YanaiKlugman Please install and replace posted systemd here `/etc/systemd/system/asus_touchpad_numpad@.service` with this:

```
[Unit]
Description=Asus Touchpad to Numpad Handler
After=display-manager.service

[Service]
User=%i
Type=idle
ExecStart=/usr/bin/env python3 /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py $LAYOUT $CONFIG_FILE_DIR
StandardInput=tty-force
StandardOutput=append:/var/log/asus_touchpad_numpad-driver/error.log
StandardError=append:/var/log/asus_touchpad_numpad-driver/error.log
TimeoutSec=5
Restart=on-failure
Environment="DISPLAY=:0"
Environment="XAUTHORITY=/run/user/1000/gdm/Xauthority"

[Install]
WantedBy=graphical.target
```

and reboot but a black screen might be also.

YanaiKlugman, 2023-07-04:
@ldrahnik 
I don't understand exactly what you mean, should the systemctl start/disable be added to a startup script? 

Also I don't see any .Xauthority file in my home folder

ldrahnik, 2023-07-04:
@YanaiKlugman Only start systemd this driver service should be added to the startup script, disable service is necessary do only after installing one time, about this, I will add to install script part which will ask question something like `Do you want automatically start driver via systemd service each reboot? [Y/n]  (Auto starting can cause black screen, recommended is via script or manually start service each login yourself)

ldrahnik, 2023-07-04:
@YanaiKlugman Also please try posted changed service, might help but expect the worst, again black screen.

YanaiKlugman, 2023-07-04:
@ldrahnik I'll try it out and update you

ldrahnik, 2023-07-06:
@YanaiKlugman ping :)

YanaiKlugman, 2023-07-06:
Haven't had a chance quite yet, didn't forget 🫡

ldrahnik, 2023-07-22:
Issue continues here https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/134

closed 2023-07-22 by ldrahnik