asus-linux-drivers/asus-numberpad-driver #130: [Bug]: touchpad click still triggered with numpad entry, when "enabled_touchpad_pointer = 3"

galaxy52, 2023-07-10:
### Describe the bug

When using any button on the numpad (other than the top right/left), a mouse tap-click is entered. This often results in clicking away from text I'm working on, or double-clicking/highlighting/replacing numbers I'm entering. Basically: numpad is activated -> number is entered -> mouse click is triggered.

Not sure if this is relevant, but I believe I have the service running [under? with? on?] X11. From readme.md: "touchpad tap-to-click is disabled/enabled using gnome via gsettings and for xinput for X11 with this order priority" 

### Expected behavior

From my understanding, "enabled_touchpad_pointer = 3" should basically lead to the following characteristics while numpad is activated:
a) pointer can be moved with touchpad (this is working as expected);
b) mouse can be clicked by physical button press of the touchpad (this is working as expected);
c) mouse click should _not_ be activated when tap-clicking a number on the numpad (this is _not_ working as expected)

### Relevant log output

```shell
As a beginner, not sure what should go here. Please let me know if I've missed anything.
```

### Desktop

- Program version: v2.8.9
- Distro: Ubuntu 22.04.2 LTS x86_64
- Kernel: 5.19.0-46-generic

ldrahnik, 2023-07-10:
@galaxy52 Thello there, thank you for creating an issue. Please attach the driver log `cat /var/log/asus_touchpad_numpad-driver/error.log` and output from installation.

galaxy52, 2023-07-10:
Sure thing - I've attached the output after using git clone, `cd asus-numberpad-driver`, and `sudo bash ./install.sh` (installed as root; no --user flag). I selected yes (Y) for every option provided. I've also attached the error.log after trying the feature that has a bug.

Thanks!
[output.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12008793/output.txt)
[errorlog.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12008794/errorlog.txt)

A couple random things:
- I did a full uninstall, delete asus-numberpad-driver from home, git clone, and reinstall, in order to get the output of a full install. I noticed this time around that it _didn't_ ask me if I wanted to use X11, I think it was, for automatically starting the service. I'm not sure if that's relevant.
- I haven't been able to get the calculator app to open with the top left button even though I selected to enable it while installing. However, I assume that this would be for a separate issue, and wouldn't affect the current issue.

ldrahnik, 2023-07-10:
@galaxy52 

> I haven't been able to get the calculator app to open with the top left button even though I selected to enable it while installing. However, I assume that this would be for a separate issue, and wouldn't affect the current issue.

About this please read this [thread](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/125). In metioned thread is in last messages described how to debug what is wrong. Or might not work when is missing reboot after installation, is necessary do that.

ldrahnik, 2023-07-10:
@galaxy52 About the main problem described in the issue, when using Wayland then unfortunately `xinput` and `gsettings` for disabling taps on touchpad does not work so recommended solution is change in config file this value `enabled_touchpad_pointer = 3` to `enabled_touchpad_pointer = 1`.

ldrahnik, 2023-07-10:
@galaxy52 

> ./install.sh: command substitution: line 221: syntax error near unexpected token `('
./install.sh: command substitution: line 221: `Disabling Touchpad (e.g. Fn+special key) disables NumberPad aswell'

Error in the message when is Wayland detected fixed in latest master.

ldrahnik, 2023-07-10:
@galaxy52 

> WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Error during installation because is used `apt` and not `apt-get` fixed in latest master. Thank you for report.

ldrahnik, 2023-07-10:
@galaxy52 

> I did a full uninstall, delete asus-numberpad-driver from home, git clone, and reinstall, in order to get the output of a full install. I noticed this time around that it didn't ask me if I wanted to use X11, I think it was, for automatically starting the service. I'm not sure if that's relevant.

That means you first time installed with using `X11` and attached files correspods to using `Wayland`

ldrahnik, 2023-07-10:
@galaxy52 

> About the main problem described in the issue, when using Wayland then unfortunately xinput and gsettings for disabling taps on touchpad does not work so recommended solution is change in config file this value enabled_touchpad_pointer = 3 to enabled_touchpad_pointer = 1.

Or I can try with your assist debug this problem more.

I would need output of these command from terminal:

```
$ echo $XDG_SESSION_TYPE
$ echo $DISPLAY
# does command below disable tap and next one enable?
$ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' false
$ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' true
```

galaxy52, 2023-07-10:
> @galaxy52
> 
> > About the main problem described in the issue, when using Wayland then unfortunately xinput and gsettings for disabling taps on touchpad does not work so recommended workaround is change in config file this value enabled_touchpad_pointer = 3 to enabled_touchpad_pointer = 1.
> 
> Or I can try with your assist debug this problem more and maybe fix it for you and even others.
> 
> I would need output of these command from your terminal:
> 
> ```
> $ echo $XDG_SESSION_TYPE
> $ echo $DISPLAY
> # does command below disable tap on touchpad and next one enable?
> $ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' false
> $ gsettings set 'org.gnome.desktop.peripherals.touchpad' 'tap-to-click' true
> ```

Sure. I'm not sure whether my installation is using X11 or Wayland, but I assume that since my output file indicated Wayland, that's how it's running currently. I completed the below with "Windowing System" on wayland (according to my Settings/About tab). Here are the outputs, with `enabled_touchpad_pointer = 3`, as default:

`echo $XDG_SESSION_TYPE` returns `wayland`

`echo $DISPLAY` returns `:0`

and yes, the two commands accordingly enable and disable the touchpad tap click. The physical button click and moving the pointer both still function unaffected by the commands.

ldrahnik, 2023-07-10:
@galaxy52 And output of this command?

```
$ echo $(/usr/bin/xauth info | grep Authority | awk '{print $3}')
```

galaxy52, 2023-07-10:
@ldrahnik that returns `/run/user/1000/.mutter-Xwaylandauth.KRBS71`.

ldrahnik, 2023-07-11:
@galaxy52 Stop driver systemctl service `$ systemctl start asus_touchpad_numpad@root.service` and then run in your terminal in the root of cloned repository `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea"`. Now does work `enabled_touchpad_pointer = 3`?

galaxy52, 2023-07-11:
> @galaxy52 Stop driver systemctl service `$ systemctl start asus_touchpad_numpad@root.service` and then run in your terminal in the root of cloned repository `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea"`. Now does work `enabled_touchpad_pointer = 3`? I mean is disabled tapping when is touched the key during activated NumberPad? Attach output, please.

@ldrahnik It's giving me some really weird behavior. 

First attempt: I ran the commands, and initially after activating the numberpad, it would deactivate (LED and any responsiveness) a moment later, returning to normal touchpad behavior. I managed to enable the numberpad and quickly touch a number, which actually kept the numberpad enabled. At this point, when I _deactivate_ the numberpad, it _reactivates_ (LED and responsiveness) quickly (its behavior had inverted). In its activated state, `enabled_touchpad_pointer = 3` actually seems to be working. However, the numberpad inputs are doubling (e.g. outputting 557788//++.. and so on). I unfortunately couldn't get a copy/paste of the output correctly, and ended up rebooting.

Second attempt: when the reboot completed, the numberpad was stuck on, but same behavior, unable to keep numberpad deactivated. I ran the commands again, demonstrated the behavior, and copy/pasted the full output into the attached file. The numberpad ended up self-deactivating after the 60 second period and stays that way. However, my tap-click seems permanently disabled, with the touchpad only moving the pointer and clicking with the physical button press.

Update; just now (having saved and closed the output file already), the touchpad/numpad seems to have almost _all_ expected behavior - I can enable/disable normally and `enabled_touchpad_pointer = 3` is working as expected - however, my tap-click is permanently disabled, regardless of the state of the numberpad. I'll see if this behavior remains the same after another reboot.

[output_debug.txt ](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12010224/output_debug.txt)

galaxy52, 2023-07-11:
The last mentioned behavior seems to remain after rebooting. All normal function except for tap-click being permanently disabled.

ldrahnik, 2023-07-11:
@galaxy52 Sorry, late night in my country and I copy pasted start indstead od stop, of course stop via `stop`. Sorry.

galaxy52, 2023-07-11:
> So driver was running two times at the same time.

Ah that makes sense, it's all good haha. I ran the uninstall script as sudo (no --user flag), deleted the directory, rebooted, used git clone, cd into directory, and ran the command. Output file attached below. 

As far as I can tell, almost everything works fine. Tap click is disabled only when numberpad is enabled (`enabled_touchpad_pointer = 3` is working as expected), numberpad activates/deactivates as normal etc. 

I'm noticing, however, when the numberpad is activated and I make a physical mouse click, the numberpad will also send an input of whichever button my finger was on. In other words, this is the opposite of the original issue; the numberpad is not disabled upon physical touchpad click. This usually results in a number being unintentionally inputted when trying to use the physical click to switch tabs or something. I'm not sure if this issue was present before, since I don't think I tested for it earlier. I assume that what _should_ happen is that the physical mouse click and tap input should be mutually exclusive, so that only one can happen at a time. 

[output.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12014331/output.txt)

ldrahnik, 2023-07-11:
@galaxy52 

> I'm noticing, however, when the numberpad is activated and I make a physical mouse click, the numberpad will also send an input of whichever button my finger was on. In other words, this is the [...]

Create another issue please if you want make change in configuration this. Atm enabling/disabling physical clicks depends on `enabled_touchpad_pointer` value and has no separated own config value.

galaxy52, 2023-07-12:
> @galaxy52
> 
> > I'm noticing, however, when the numberpad is activated and I make a physical mouse click, the numberpad will also send an input of whichever button my finger was on. In other words, this is the opposite of the original issue; the numberpad is not disabled upon physical touchpad click. This usually results in a number being unintentionally inputted when trying to use the physical click to switch tabs or something. I'm not sure if this issue was present before, since I don't think I tested for it earlier. I assume that what should happen is that the physical mouse click and tap input should be mutually exclusive, so that only one can happen at a time.
> 
> Create another issue please if you want make change in configuration. Atm enabling/disabling physical clicks depends on `enabled_touchpad_pointer` value and has no separated own config value.

Ah I actually have no clue how the driver really works (I'm still very much a beginner at linux haha), but I can create another issue if the problem persists after this troubleshooting, if that helps.

galaxy52, 2023-07-12:
@ldrahnik Just making sure I understand your instructions, here's what I did. Let me know if I missed anything:

``sudo bash ./uninstall.sh``
 deleted the asus-numberpad-driver directory
``reboot``
``git clone https://github.com/asus-linux-drivers/asus-numberpad-driver``
using the file explorer, navigated to asus-numberpad-driver 
opened ``install.sh`` with Text Editor
clicked on (empty) line 29 and pasted in
> ``wayland_or_x11 = "x11"``

clicked save
``cd asus-numberpad-driver``
``sudo bash ./install.sh`` - attached install output below (answered "Y" to all options, including reboot) (looks like line 29 was not recognized?)

[install_output.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12023317/install_output.txt)

At this point, I'm actually not sure what you want me to do from here. Should I demonstrate using the touchpad then take and attach ``/var/log/asus_touchpad_numpad-driver/error.log``? 

[errorlog.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12023316/errorlog.txt)

[image]

ldrahnik, 2023-07-12:
@galaxy52 Yeah, bash do not allow space between equal character so please change to `wayland_or_x11 
="x11"`, then reboot, everything Y during install and test whether Is disabled tap to click when is NumberPad activated

galaxy52, 2023-07-13:
@ldrahnik 
I took the exact same steps as my previous message, with the only difference being replacing ``wayland_or_x11 = "x11"`` with ``wayland_or_x11="x11"`` (no spaces). The numberpad/touchpad behavior seems to be exactly the same as the initial issue (i.e. ``enabled_touchpad_pointer = 3`` isn't working as expected). Install output and error log after testing are below:

[install_output.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12034934/install_output.txt)
[errorlog.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12034935/errorlog.txt)
[image]

ldrahnik, 2023-07-14:
@galaxy52

> I'm noticing, however, when the numberpad is activated and I make a physical mouse click, the numberpad will also send an input of whichever button my finger was on. In other words, this is the [...]

> Create another issue please if you want make change in configuration. Atm enabling/disabling physical clicks depends on enabled_touchpad_pointer value and has no separated own config value.

I have added to the project `enabled_touchpad_pointer = 4` which is equal to `3` with one difference it does not allow taps and even pointer button clicks since is NumberPad activated until is deactivated. `git pull` latest master and you can try, disable installed systemctl service so will not be duplicated instance of driver running and then test via `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea"`

ldrahnik, 2023-07-14:
@galaxy52

> I'm noticing, however, when the numberpad is activated and I make a physical mouse click, the numberpad will also send an input of whichever button my finger was on. In other words, this is the [...]

> Create another issue please if you want make change in configuration. Atm enabling/disabling physical clicks depends on enabled_touchpad_pointer value and has no separated own config value.

`git pull` latest master, stop systemctl service, run as `LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea"` and please let me know I committed the feature that you requested.

ldrahnik, 2023-07-15:
@galaxy52 

> git pull latest master, stop systemctl service (in case driver is already installed), run as LOG=DEBUG sudo -E ./asus_touchpad.py "up5401ea" and please let me know I committed the feature that you requested.

Pushed a better version of this feature.

galaxy52, 2023-07-15:
@ldrahnik I actually had it uninstalled at the moment so I just used ``git clone`` and did _not_ run ``install.sh`` - I assume that achieves the same result.

After cloning running the debug command, ``enabled_touchpad_pointer = 3`` seemed to be working as expected, plus I was able to use the physical click without inputting a number while the numberpad was active. All seems good. Debug transcript below.

I'm not sure if it's very important, but when I proceeded to actually run the installation (as root, no --user tag), my requested feature worked, but the issue of ``enabled_touchpad_pointer = 3`` not working was still there.

[debuglog.txt](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12062200/debuglog.txt)

ldrahnik, 2023-07-15:
@galaxy52 Unfortunatelly I had no time looks futher neither idea what you could try on your laptop atm.

ldrahnik, 2023-07-22:
@galaxy52 Issue continues here #134

closed 2023-07-22 by ldrahnik