asus-linux-drivers/asus-numberpad-driver #132: [Bug]: no backlight on UM5302TA

guilhem, 2023-07-15:
### Describe the bug

When activated, numpad doesn't light on.
Control doesn't work.

### Expected behavior

Numpad light on and can be controled

### Relevant log output

```shell
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0059194564819336
INFO:asus-touchpad-numpad-driver:Activation time: 1.0
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:0
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "brightness" with value: "0x41"
INFO:asus-touchpad-numpad-driver:check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=asus_touchpad_numpad_dev path=/usr/share/asus_touchpad_numpad-driver pathname=/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev wd=1 >
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0067079067230225
INFO:asus-touchpad-numpad-driver:Activation time: 1.0
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:1
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "brightness" with value: "0x42"
INFO:asus-touchpad-numpad-driver:check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=asus_touchpad_numpad_dev path=/usr/share/asus_touchpad_numpad-driver pathname=/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev wd=1 >
INFO:asus-touchpad-numpad-driver:The top_left_icon was pressed longer than the activation time: 1.0067501068115234
INFO:asus-touchpad-numpad-driver:Activation time: 1.0
INFO:asus-touchpad-numpad-driver:Increased brightness of backlight to
INFO:asus-touchpad-numpad-driver:2
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "brightness" with value: "0x43"
INFO:asus-touchpad-numpad-driver:check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=asus_touchpad_numpad_dev path=/usr/share/asus_touchpad_numpad-driver pathname=/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev wd=1 >
```
```

### Desktop

- Program version c052f0e
- Distro Ubuntu 23.04
- Kernel 6.4.3-x64v3-xanmod1

ldrahnik, 2023-07-15:
@guilhem Hello there, please read [this block](https://github.com/asus-linux-drivers/asus-numberpad-driver#faq) in `Readme.md` and let me know about the result. You do not have to worry about hardware, by trying these values you can not break the hardware device. Thank you for creating an issue.

guilhem, 2023-07-15:
Wow indeed

activation worked with
```bash
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x60 0xad
```

deactivation
```bash
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x61 0xad
```

ldrahnik, 2023-07-15:
@guilhem Please post me entire output from installation.

ldrahnik, 2023-07-15:
@guilhem And entire output of `/var/log/asus_touchpad_numpad-driver/error.log`. Thank you.

guilhem, 2023-07-15:
install.sh:
<details>

```
groupadd: group 'uinput' already exists
KERNEL=="uinput", GROUP="uinput", MODE:="0660"
driver will run under user guilhem
wayland is DETECTED
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ibus is already the newest version (1.5.28-3).
libevdev2 is already the newest version (1.13.0+dfsg-1).
i2c-tools is already the newest version (4.3-2build3).
python3-dev is already the newest version (3.11.2-1).
python3-libevdev is already the newest version (0.5-3).
python3-numpy is already the newest version (1:1.24.2-1).
python3-xlib is already the newest version (0.33-2).
python3-pyinotify is already the newest version (0.9.6-2).
The following packages were automatically installed and are no longer required:
  ethtool liblc3-0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Testing interface i2c-0 : failed
Testing interface i2c-1 : sucess
Detected laptop: Zenbook UM5302TA_UM5302TA
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 52107    0 52107    0     0   143k      0 --:--:-- --:--:-- --:--:--  144k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  257k    0  257k    0     0   711k      0 --:--:-- --:--:-- --:--:--  712k
Automatically recommended numpad layout: up5401ea (associated to UM3402YA-OLED059W). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout up5401ea
Selected key layout specified to up5401ea-31B9 by touchpad ID 31B9
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
Wayland is detected, unfortunatelly you will not be able use feature: Disabling Touchpad (e.g. Fn+special key) disables NumberPad aswell, at this moment is supported only X11
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
grep: asus_touchpad_numpad_dev: No such file or directory
In system remains config file from previous installation. Do you want replace this with default config file [y/N] 
gsettings is here
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
Found already existing ducplicated shortcut for toggling calculator, will be removed
io.elementary.calculator here
expected value:
  , /org/gnome/settings-daemon/plugins/media-keys/custom-keybindings/custom1']
  ^                                                                           
Toggling script for calculator app io.elementary.calculator has been installed.
Systemctl daemon realod called succesfully
Asus touchpad numpad service enabled
Asus touchpad numpad service started
Install process requested to succesfull finish atleast log out or reboot
Without that driver might not work properly
Do you want reboot now? [y/N]
Install finished
```

</details>

[error.log](https://github.com/asus-linux-drivers/asus-numberpad-driver/files/12062107/error.log)

ldrahnik, 2023-07-15:
@guilhem Please try reboot as is written in [issue mentioned in FAQ](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/108#issuecomment-1464735275). Let me know it then works with `0x01` and `0x00`.

guilhem, 2023-07-15:
> @guilhem Please try reboot as is written in [issue mentioned in FAQ](https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/108#issuecomment-1464735275). Let me know it then works with `0x01` and `0x00`.

@ldrahnik nope, still the same :man_shrugging:

ldrahnik, 2023-07-15:
@guilhem Try change working backlight values in appropriated layout file in dir `numpad_layouts`, then install driver and if will this difference persist I will try update driver to reflect it. IMHO it might be somehow associated with AMD kernel modules? Only tip.

guilhem, 2023-07-15:
@ldrahnik Do I have to edit this?
```python
backlight_levels = [
    "0x41",
    "0x42",
    "0x43",
    "0x44",
    "0x45",
    "0x46",
    "0x47",
    "0x48"
]
```
to

```python
backlight_levels = [
    "0x60",
    "0x61"
]
```

But how to have level control?

ldrahnik, 2023-07-15:
@guilhem Activate using `0x60` and then in terminal manually try every value until you will change brightness level. But I guess it will be in one row I would try be you 62,63,64,65 first.

guilhem, 2023-07-15:
@ldrahnik I had some really strange behavior.

I don't just reboot, I shutdown the laptop.
When wakeup, 0x60 (on) 0x61 (off) didn't work anymore.

but 0x01 (on) 0x00 (off) do!

I was suprised, I tested 0x60/61 and it did worked, and 0x01/00 don't work again anymore.

guilhem, 2023-07-15:
it really look like 0x60 / 61 is used to change a way the touchpad is working.

After enabling 0x60, I'm able to use 0x00/01 02/03 ... 0e/0f
but after setting 0x61, it doesn't work anymore.

ldrahnik, 2023-07-15:
@guilhem It looks like `0x60` and `0x61` can be imagined easily as a switch on the thread with another switch `0x01` and `0x00` on the same thread. That means does not matter in which order is called `0x01` and `0x60` but have to be for brightness up called both of them. I am right? So I will add every NumberPad activation calling also `0x60` for sure and then we can close this issue I guess.

closed 2023-07-15 by ldrahnik

guilhem, 2023-07-15:
it works :)