asus-linux-drivers/asus-numberpad-driver #133: [Bug]: When the driver is clicked on the ubuntu login screen, three numbers will be displayed

Zijian050820, 2023-07-19:
### Describe the bug

When the driver is clicked on the ubuntu login screen, three numbers will be displayed

### Expected behavior

When the driver is clicked on the ubuntu login screen, three numbers will be displayed

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. ubuntu]
- Kernel [e.g. 5.18.15]

Zijian050820, 2023-07-19:
When using numberpad on the ubuntu login screen, three identical numbers appear on the number screen once clicked

ldrahnik, 2023-07-19:
@Zijian050820 Hello there! Thank you for letting me know. Unfortunatelly without logs (driver log primary) and description about used environment (Ubuntu version, wayland/X11, keyboard layout, ..) I will have to close this issue as u unfounded.

ldrahnik, 2023-07-20:
Can be reopened.

closed 2023-07-20 by ldrahnik