asus-linux-drivers/asus-numberpad-driver #135: [Bug]: Not working after install

fededileo, 2023-07-30:
### Describe the bug

Hi. A couple of weeks ago I opened an issue about the driver preventing login on Fedora 38. It's me again! However this time it's not preventing login. Nothing happens instead when trying to enable numberpad. Here you have an output from `var/log/asus_touchpad_numpad-driver/error.log`.
I installed the driver as user by downloading 2.9.2 from stable branch 

Nice work, I'm glad you're answering every request and bug.

### Expected behavior

Normal behaviour

### Relevant log output

```shell
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Traceback (most recent call last):
  File "/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py", line 299, in <module>
    with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/input/event2'
```

### Desktop

- Program version [2.9.2]
- Distro [Fedora Workstation 38]
- Kernel [6.4.6-200.fc38.x86_64]

ldrahnik, 2023-07-31:
@fededileo Hello there, could you please post groups of your user? It only looks like your user do not have required groups `uinput` and `input`

ldrahnik, 2023-07-31:
@fededileo This time you installed as `--user`? And previously without `--user` flag?

fededileo, 2023-07-31:
Hi, I made a quick Google search about users and groups (remember, not a programmer lol). I'll paste the output of `cat /etc/group` down here, guess this is what you're looking for.

```
[federico@zenbook ~]$ cat /etc/group
root:x:0:
bin:x:1:
daemon:x:2:
sys:x:3:
adm:x:4:
tty:x:5:
disk:x:6:
lp:x:7:
mem:x:8:
kmem:x:9:
wheel:x:10:federico
cdrom:x:11:
mail:x:12:
man:x:15:
dialout:x:18:
floppy:x:19:
games:x:20:
tape:x:33:
video:x:39:
ftp:x:50:
lock:x:54:
audio:x:63:
users:x:100:
nobody:x:65534:
dbus:x:81:
apache:x:48:
utmp:x:22:
utempter:x:35:
tss:x:59:
input:x:104:
kvm:x:36:qemu
render:x:105:
sgx:x:106:
systemd-journal:x:190:
systemd-network:x:192:
systemd-oom:x:999:
systemd-resolve:x:193:
printadmin:x:998:
qemu:x:107:
polkitd:x:997:
avahi:x:70:
dip:x:40:
geoclue:x:996:
nm-openconnect:x:995:
colord:x:994:
usbmuxd:x:113:
gluster:x:993:
rtkit:x:172:
pipewire:x:992:
brlapi:x:991:
chrony:x:990:
dnsmasq:x:989:
rpc:x:32:
saslauth:x:76:saslauth
rpcuser:x:29:
openvpn:x:988:
nm-openvpn:x:987:
unbound:x:986:
libvirt:x:985:
abrt:x:173:
flatpak:x:984:
gdm:x:42:
gnome-initial-setup:x:983:
vboxsf:x:982:
sshd:x:74:
power:x:981:
tcpdump:x:72:
plocate:x:980:
gamemode:x:979:
systemd-coredump:x:978:
systemd-timesync:x:977:
federico:x:1000:
uinput:x:1001:
```

On the other hand, yes. I installed the driver with the `--user` suffix. Last time I didn't.

:)

ldrahnik, 2023-07-31:
@fedelileo Please post output of cmd `groups`

fededileo, 2023-07-31:
Here you have :)
`[federico@zenbook ~]$ groups
federico wheel`

fededileo, 2023-07-31:
Just out of curiosity added my user federico to the groups `input` and `uinput` and reinstalled the driver but nothing happened. Log file is intact so it's still the same file you saw. I'll paste here all output of the install script, perhaps there's something interesting in there.

`
[federico@zenbook asus-numberpad-driver-2.9.2]$ sudo bash ./install.sh --user
[sudo] contraseña para federico: 
groupadd: el grupo «uinput» ya existe
KERNEL=="uinput", GROUP="uinput", MODE:="0660"
usermod: el grupo «i2c» no existe
driver will run under user federico
wayland is DETECTED
dnf is here
Última comprobación de caducidad de metadatos hecha hace 0:46:32, el lun 31 jul 2023 16:36:05.
El paquete ibus-1.5.28-6.fc38.x86_64 ya está instalado.
El paquete libevdev-1.13.1-1.fc38.x86_64 ya está instalado.
El paquete i2c-tools-4.3-5.fc38.x86_64 ya está instalado.
El paquete python3-devel-3.11.4-1.fc38.x86_64 ya está instalado.
El paquete python3-libevdev-0.11-4.fc38.noarch ya está instalado.
El paquete python3-numpy-1:1.24.4-1.fc38.x86_64 ya está instalado.
El paquete python3-inotify-0.9.6-29.fc38.noarch ya está instalado.
El paquete python3-xlib-0.33-2.fc38.noarch ya está instalado.
Dependencias resueltas.
Nada por hacer.
¡Listo!
Testing interface i2c-0 : sucess
Detected laptop: ZenBook UX425QA_UM425QA
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 52135    0 52135    0     0  23356      0 --:--:--  0:00:02 --:--:-- 23368
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  247k    0  247k    0     0  90749      0 --:--:--  0:00:02 --:--:-- 90728
Automatically recommended numpad layout: up5401ea (associated to B9400CEA-KC0519R). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout up5401ea
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
Wayland is detected, unfortunatelly you will not be able use feature: Disabling Touchpad (e.g. Fn+special key) disables NumberPad aswell, at this moment is supported only X11
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
./install.sh: línea 338: elemento inesperado `;', se esperaba un operador binario condicional
[federico@zenbook asus-numberpad-driver-2.9.2]$ 
`

:)

fededileo, 2023-08-03:
Hi. Here you have the test result. Backlight turns on on two brightness level. Second is darker than the first one just by a little bit. I've seen it go darker on Windows but that's irrelevant at the moment.

`
[federico@zenbook tests]$ sudo python3 test_brightness.py
[sudo] contraseña para federico: 
0x01
0x41
Tested value of registr: 2 (hex: 0x2)
Press Enter to continue...
`

Hope you'll fix it soon :)

fededileo, 2023-08-03:
I've got even more interesting data. I was fiddling around with systemctl and wrote `systemctl enable asus_touchpad_numpad@federico.service`. Rebooted and guess what? Login freeze just like the other day. So right now i'm deleting  the service on a live iso so I can boot again, but this time I got the error log from the `var/log` directory.

`
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 2 from H: Handlers=sysrq kbd leds event2
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140A:00 04F3:3134 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0003/input/input17
INFO:asus-touchpad-numpad-driver:Set touchpad id 5 from H: Handlers=mouse1 event5
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-1972
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-1892
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3958 y 89
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1691032538.116181
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3948 y 118
INFO:asus-touchpad-numpad-driver:Un-touched without NumberPad activation top_right_icon (numlock) in time: 1691032538.840921
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3987 y 95
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1691032539.5136707
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3959 y 108
INFO:asus-touchpad-numpad-driver:Un-touched without NumberPad activation top_right_icon (numlock) in time: 1691032540.3947384
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3869 y 172
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1691032540.4791424
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3831 y 157
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.0731558799743652
INFO:asus-touchpad-numpad-driver:Activation time: 1.0
INFO:asus-touchpad-numpad-driver:Un-touched with NumberPad activation top_right_icon (numlock) in time: 1691032541.5525556
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 254 y 179
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 245 y 227
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1675 y 955
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1668 y 846
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2105 y 906
INFO:asus-touchpad-numpad-driver:Pressed touchpad pointer button
INFO:asus-touchpad-numpad-driver:Released touchpad pointer button
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x -1 y -1
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3971 y 29
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1691032545.3788047
INFO:asus-touchpad-numpad-driver:Pressed touchpad pointer button
INFO:asus-touchpad-numpad-driver:Released touchpad pointer button
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 285 y 0
INFO:asus-touchpad-numpad-driver:Pressed touchpad pointer button
INFO:asus-touchpad-numpad-driver:Released touchpad pointer button
INFO:asus-touchpad-numpad-driver:Pressed touchpad pointer button
INFO:asus-touchpad-numpad-driver:Released touchpad pointer button
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2264 y 757
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2091 y 1043
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 2 from H: Handlers=sysrq kbd leds event2
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140A:00 04F3:3134 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0003/input/input17
INFO:asus-touchpad-numpad-driver:Set touchpad id 5 from H: Handlers=mouse1 event5
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-1972
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-1892
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3897 y 188
[... 199 lines cut ...]
  File "/usr/lib/python3.11/site-packages/libevdev/device.py", line 764, in create_uinput_device
    d._uinput = UinputDevice(self._libevdev, uinput_fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/libevdev/_clib.py", line 964, in __init__
[... cut ...]
  File "/usr/lib/python3.11/site-packages/libevdev/device.py", line 764, in create_uinput_device
    d._uinput = UinputDevice(self._libevdev, uinput_fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/libevdev/_clib.py", line 964, in __init__
    raise OSError(-rc, os.strerror(-rc))
PermissionError: [Errno 13] Permission denied`

I wish I wasn't so annoying but if I help you to get this working we'll be helping many people to use their numberpads, hope you'll understand :)

ldrahnik, 2023-08-03:
@fededileo Sorry, unfortunatelly I am on vacation.

Try to run command `groups` and please do a screenshot of command output.

What should solve your problem Is command
```
usermod -a -G "i2c,input,uinput" <here put your username on laptop>
```

and reboot.

fededileo, 2023-08-03:
Hi, I apologize for the inconvenience. Please feel free to answer whenever you can. 

Here's the output of `groups federico`
`
[federico@zenbook ~]$ groups federico
federico : federico wheel input uinput
`

And here's the output of `usermod...`

`
[federico@zenbook ~]$ usermod -a -G "i2c,input,uinput" federico
usermod: el grupo «i2c» no existe 
`

So the group `i2c` doesn't exist.

:)

ldrahnik, 2023-08-05:
@fededileo Hello, I am back so I can respond more regularly. 

I read your posts and it still looks like you have only wrongly configured your user permissions to input,uinput & i2c, your user has to have access to input devices, so usually is necessary add user to groups `input, uniput and i2c`, in case does not exist `i2c` group I would create the group by command and add your user to group. To check your laptop is configured with these names of groups you can check easily like this (you should see something similar):

```
ldrahnik@Zenbook-UP5401EA:/sys/class/i2c-dev$ find /dev -group i2c
/dev/i2c-14
/dev/i2c-13
/dev/i2c-12
/dev/i2c-11
/dev/i2c-10
/dev/i2c-9
/dev/i2c-8
/dev/i2c-7
/dev/i2c-6
/dev/i2c-5
/dev/i2c-4
/dev/i2c-3
/dev/i2c-2
/dev/i2c-1
/dev/i2c-0
ldrahnik@Zenbook-UP5401EA:/sys/class/i2c-dev$ find /dev -group input
/dev/input/event23
/dev/input/event22
/dev/input/event21
/dev/input/event20
/dev/input/event19
/dev/input/event18
/dev/input/event17
/dev/input/event13
/dev/input/mouse3
/dev/input/event12
/dev/input/mouse2
/dev/input/event11
/dev/input/event16
/dev/input/event10
/dev/input/event9
/dev/input/event8
/dev/input/mouse1
/dev/input/event7
/dev/input/event6
/dev/input/event5
/dev/input/mouse0
/dev/input/event4
/dev/input/event15
/dev/input/event14
/dev/input/event3
/dev/input/event2
/dev/input/event1
/dev/input/event0
/dev/input/mice
ldrahnik@Zenbook-UP5401EA:/sys/class/i2c-dev$ find /dev -group uinput
/dev/uinput
```

ldrahnik, 2023-08-05:
@fededileo Try to download and install the latest master branch or execute:

```
$ groupadd "input"
$ groupadd "uinput"
$ groupadd "i2c"

# and then
usermod -a -G "i2c,input,uinput" <your username>
```

closed 2023-08-08 by ldrahnik

fededileo, 2023-08-14:
Hi, it has been quite a while since my last update, didn't use my laptop since then.
This is the output you asked me to check.

`[federico@zenbook ~]$ /sys/class/i2c-dev$ find /dev -group i2c
bash: /sys/class/i2c-dev$: No existe el fichero o el directorio `

Translation: file or directory doesn't exist

:)

ldrahnik, 2023-08-15:
@fededileo And driver works for you since my recommended commands or not? State Is without any change?

fededileo, 2023-08-15:
@ldrahnik Hi! Unfortunately it didn't work. I ran the group-adding commands and installed stable version and it prevented login, booted on live ISO to delete service although I forgot to save the log. Then proceeded to install dev version and rebooted but same thing happened. This is the error log output:

`INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 2 from H: Handlers=sysrq kbd leds event2
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE140A:00 04F3:3134 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0004/input/input17
INFO:asus-touchpad-numpad-driver:Set touchpad id 5 from H: Handlers=mouse1 event5
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-1972
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-1892`

I don't know whether it is because you updated the script or because something new is happening because of the new `i2c` group but the installer asked me if I wanted to enable calculator and asked for reboot. In the past it didn't.

:)

ldrahnik, 2023-08-15:
@fededileo Unfortunatelly in might happens, warning during installing for autostart systemctl service says aswell the same. And driver except this works or not?

fededileo, 2023-08-15:
@ldrahnik I'm not sure if I understand what you said, I assume you're using a translator. I'm not using one but not a native english speaker here so apologies.
I couldn't test the driver since it prevented login.

fededileo, 2023-08-15:
@ldrahnik I don't recall seeing that question during install. IIRC questions were: 

1) use suggested layout
2) use calculator gesture
3) reboot

:)

ldrahnik, 2023-08-15:
@fededileo Does make sense in case you use Wayland. I will add this question for Wayland aswell, probably this night. Up to that time you can disable systemctl service via `sudo systemctl disable Asus...` and test driver.

ldrahnik, 2023-08-16:
@fededileo `git pull` latest master

fededileo, 2023-08-18:
Hi @ldrahnik, sorry for my late response. I uninstalled the driver but now my keyboard shortcuts are not working and `gnome-control-center` crashes when I go to the keyboard shortcuts menu item :(

ldrahnik, 2023-08-18:
@fededileo Can you please attach here the error log of `gnome-control-center`. And did you test the driver with my previously changed code? I mean with manual starting via `sudo systemctl start asus_touchpad_numpad@<your username or root>.service`

fededileo, 2023-08-19:
Hi @ldrahnik , this is getting longer than I thought. However, this is not a complaint, it's a recognition for your work. Thank you.
Where can I find the gcc logs?
I'll try reinstalling the driver in a couple of hours.

ldrahnik, 2023-08-19:
@fededileo Unfortunately I do not use that tool, I would google as you and I do not recommend call it gcc as gcc is usually called in Linux community GNU Compiler Collection.