asus-linux-drivers/asus-numberpad-driver #136: [Bug]: Error during installation on Pop!_OS

pvojnisek, 2023-08-02:
### Describe the bug

Thank you for your effort to develop this great driver!
During installation I got an error message. I copied the full installation log bellow.

### Expected behavior

successful install

### Relevant log output

```shell
$ sudo bash ./install.sh --user
[sudo] password for peter: 
groupadd: group 'uinput' already exists
KERNEL=="uinput", GROUP="uinput", MODE:="0660"
driver will run under user peter
x11 is DETECTED
apt is here
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ibus is already the newest version (1.5.26-4).
libevdev2 is already the newest version (1.12.1+dfsg-1).
python3-pyinotify is already the newest version (0.9.6-1.3).
python3-pyinotify set to manually installed.
i2c-tools is already the newest version (4.3-2build1).
python3-xlib is already the newest version (0.29-1).
python3-xlib set to manually installed.
python3-dev is already the newest version (3.10.6-1~22.04).
The following additional packages will be installed:
  libgfortran5 liblapack3
Suggested packages:
  python-libevdev-doc gfortran python-numpy-doc
The following NEW packages will be installed:
  libgfortran5 liblapack3 python3-libevdev python3-numpy
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,868 kB of archives.
After this operation, 29.7 MB of additional disk space will be used.
Get:1 http://apt.pop-os.org/ubuntu jammy-security/main amd64 libgfortran5 amd64 12.3.0-1ubuntu1~22.04 [879 kB]
Get:2 http://apt.pop-os.org/ubuntu jammy/main amd64 liblapack3 amd64 3.10.0-2ubuntu1 [2,504 kB]
Get:3 http://apt.pop-os.org/ubuntu jammy/universe amd64 python3-libevdev all 0.5-3 [17.5 kB]
Get:4 http://apt.pop-os.org/ubuntu jammy-security/main amd64 python3-numpy amd64 1:1.21.5-1ubuntu22.04.1 [3,467 kB]
Fetched 6,868 kB in 2s (4,534 kB/s)      
Selecting previously unselected package libgfortran5:amd64.
(Reading database ... 213071 files and directories currently installed.)
Preparing to unpack .../libgfortran5_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking libgfortran5:amd64 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package liblapack3:amd64.
Preparing to unpack .../liblapack3_3.10.0-2ubuntu1_amd64.deb ...
Unpacking liblapack3:amd64 (3.10.0-2ubuntu1) ...
Selecting previously unselected package python3-libevdev.
Preparing to unpack .../python3-libevdev_0.5-3_all.deb ...
Unpacking python3-libevdev (0.5-3) ...
Selecting previously unselected package python3-numpy.
Preparing to unpack .../python3-numpy_1%3a1.21.5-1ubuntu22.04.1_amd64.deb ...
Unpacking python3-numpy (1:1.21.5-1ubuntu22.04.1) ...
Setting up python3-libevdev (0.5-3) ...
Setting up libgfortran5:amd64 (12.3.0-1ubuntu1~22.04) ...
Setting up liblapack3:amd64 (3.10.0-2ubuntu1) ...
update-alternatives: using /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3 to provide /usr/lib/x86_64-linux-gnu/liblapack.so.3 (liblapack.so.3-x86_64-linux-gnu) in auto mode
Setting up python3-numpy (1:1.21.5-1ubuntu22.04.1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput is already the newest version (1.6.3-1build2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Testing interface i2c-0 : sucess
Detected laptop: Zenbook UM3402YA_UM3402YA
Automatically recommended numpad layout: up5401ea (associated to UM3402YA-OLED059W). You can specify numpad layout later by yourself. When is recommended layout wrong please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues). Do you want use recommended numpad layout? [y/N]y
Selected key layout up5401ea
Selected key layout specified to up5401ea-31B9 by touchpad ID 31B9
Installing asus touchpad service to /etc/systemd/system/
Attempt to stop, disable and remove service from previous implementation (without using user templates @)
Failed to stop asus_touchpad_numpad.service: Unit asus_touchpad_numpad.service not loaded.
X11 is detected
DISPLAY: :1
AUTHORITY: /run/user/1000/gdm/Xauthority
Installing udev rules to /usr/lib/udev/rules.d/
Added 90-numberpad-external-keyboard.rules
Installed default config which can be futher modified here:
/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev
./install.sh: line 338: unexpected token `;', conditional binary operator expected
./install.sh: line 338: syntax error near `;'
./install.sh: line 338: `        if [[ $(type io.elementary.calculator 2>/dev/null) ; $? -eq 0 ]]; then'
```

### Desktop

- Hardware: ASUSTeK COMPUTER INC. Zenbook UM3402YA_UM3402YA
- Program version: asus-numberpad-driver-2.9.2.zip
- Distro: Pop!_OS 22.04 LTS
- Kernel: Linux pop-os 6.2.6-76060206-generic #202303130630~1689015125~22.04~ab2190e SMP PREEMPT_DYNAMIC Mon J x86_64 x86_64 x86_64 GNU/Linux

levog-1, 2023-08-04:
Hello,

I stumbled upon the same problem. Turns out it can be easily fixed by changing the bash test from 
`  if [[ $(type io.elementary.calculator 2>/dev/null) ; $? -eq 0 ]]; then ` 
to
`  if [[ $(type io.elementary.calculator 2>/dev/null ; echo $? ) -eq 0 ]]; then `

tested with bash version 5.12, Fedora 38

I modified respective lines and submitted a pull request...

ldrahnik, 2023-08-04:
@pvojnisek Merged PR created by @levog-1

closed 2023-08-04 by ldrahnik