asus-linux-drivers/asus-numberpad-driver #139: [Bug]: Printing hex on button click

georgev93, 2023-09-14:
### Describe the bug

The numpad enables and disables properly, but when I click "9", it prints "u39 ". This happens when I'm in the console window, browser, etc. In the calculator it just prints "39"

### Expected behavior

When I hit "9", I would expect it to print "9"

### Relevant log output

```shell
INFO:asus-touchpad-numpad-driver:Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
INFO:asus-touchpad-numpad-driver:Set keyboard 3 from H: Handlers=sysrq kbd leds event3
INFO:asus-touchpad-numpad-driver:Detecting touchpad from string: "N: Name="ASUE1412:00 04F3:3285 Touchpad""
INFO:asus-touchpad-numpad-driver:Set touchpad device id 14 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-ASUE1412:00/0018:04F3:3285.0001/input/input11
INFO:asus-touchpad-numpad-driver:Set touchpad id 8 from H: Handlers=event8 mouse1
INFO:asus-touchpad-numpad-driver:Touchpad min-max: x 0-3987, y 0-2493
INFO:asus-touchpad-numpad-driver:Numpad min-max: x 200-3787, y 200-2413
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3927 y 50
INFO:asus-touchpad-numpad-driver:Touched top_right_icon (numlock) in time: 1694705525.078036
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3900 y 66
INFO:asus-touchpad-numpad-driver:The numpad numlock was pressed longer than the activation time: 1.5440130233764648
INFO:asus-touchpad-numpad-driver:Activation time: 1.0
INFO:asus-touchpad-numpad-driver:Un-touched with NumberPad activation top_right_icon (numlock) in time: 1694705526.6222038
INFO:asus-touchpad-numpad-driver:System numlock activated
INFO:asus-touchpad-numpad-driver:Numpad activated
INFO:asus-touchpad-numpad-driver:grab
INFO:asus-touchpad-numpad-driver:Setting up for config file key: "enabled" with value: "True"
INFO:asus-touchpad-numpad-driver:check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
<Event dir=False mask=0x8 maskname=IN_CLOSE_WRITE name=asus_touchpad_numpad_dev path=/usr/share/asus_touchpad_numpad-driver pathname=/usr/share/asus_touchpad_numpad-driver/asus_touchpad_numpad_dev wd=1 >
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2171 y 492
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2171 y 492
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:9
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:9
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 2184 y 515
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 2176 y 497
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:9
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:9
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3407 y 813
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3408 y 820
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3433 y 720
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3429 y 724
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:Unpressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 3436 y 736
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3429 y 732
INFO:asus-touchpad-numpad-driver:Pressed numpad key
INFO:asus-touchpad-numpad-driver:KEY_BACKSPACE:14
[... 77 lines cut ...]
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 3859 y 1215
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1471 y 1300
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1599 y 307
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1486 y 1214
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1142 y 142
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1212 y 819
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 1188 y 214
INFO:asus-touchpad-numpad-driver:Started new slot
INFO:asus-touchpad-numpad-driver:finger down at x 1745 y 668
INFO:asus-touchpad-numpad-driver:Ended existing slot
INFO:asus-touchpad-numpad-driver:finger down at x 918 y 507
```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2023-09-14:
@georgev93 Hello there, thank you for creating an issue. In your environment, are you able to print Unicode characters via Ctrl+Shift+U+<number>? It look like you are not. For using this driver you have to be able to do that. For example, the shortcut for percent char is left ctrl + left shift + U + 25 = %. Try to enable that in your environment and then try printing number 9 again, should be done properly. I see in the error log no errors.

ldrahnik, 2023-09-15:
@georgev93 Can be reopened.

georgev93, 2023-09-15:
Thanks for the response! As it turns out, I can print unicode characters with left_ctrl + left_shift + (hex code) in my console, but not in most other programs. Note that I don't need to type the u, and if I do I just get an extra u before my unicode character.

A little more background on my keyboard setup: I'm using a custom variant I defined in my /usr/share/X11/xkb/symbols/us file:

```
partial alphanumeric_keys
xkb_symbols "george" {
    include "us(basic)"
    name[Group1]= "English custom layout";

    key <AD06> { type="THREE_LEVEL_SHIFT",   [        y,        Y,     Prior]   };
    key <AD07> { type="THREE_LEVEL_SHIFT",   [        u,        U,      Home]   };
    key <AD08> { type="THREE_LEVEL_SHIFT",   [        i,        I,        Up]   };
    key <AD09> { type="THREE_LEVEL_SHIFT",   [        o,        O,       End]   };
    key <AD10> { type="THREE_LEVEL_SHIFT",   [        p,        P,    Delete]   };

    key <AC04> {
        type[Group1] = "THREE_LEVEL_SHIFT",
        symbols[Group1] = [          f,          F                           ],
        actions[Group1] = [ NoAction(), NoAction(), LockMods(modifiers=Lock) ]
    };

    key <AC06> { type="THREE_LEVEL_SHIFT",   [        h,        H,      Next]   };
    key <AC07> { type="THREE_LEVEL_SHIFT",   [        j,        J,      Left]   };
    key <AC08> { type="THREE_LEVEL_SHIFT",   [        k,        K,      Down]   };
    key <AC09> { type="THREE_LEVEL_SHIFT",   [        l,        L,     Right]   };
    key <AC10> { type="THREE_LEVEL_SHIFT",   [semicolon,    colon, BackSpace]   };

    key <AB10> { type="THREE_LEVEL_SHIFT",   [    slash, question,    Insert]   };

    key <SPCE> { type="THREE_LEVEL_SHIFT",   [    space,    space,    Return]   };
};
```

But this variant is just a superset of the regular us keyboard with some modification to the caps lock behavior, so I'm not sure why that would mess with the ctrl + shift + u unicode entry behavior.

georgev93, 2023-09-15:
Just to confirm, I commented out my modifications to xkb files, set my KDE Plasma environment to chose the default US keyboard layout, and restarted. I'm still having issues. Although now instead of typing "u39 " when I type 9 on the numpad, it is typing "39 ", and repeated presses overwrite the previous "39 ". Num lock is turned on when I activate my numpad (which is the intended behavior, correct?).

ldrahnik, 2023-09-16:
@georgev93 Important is what do you see in ibus settings, me: 

[image]

ldrahnik, 2023-09-16:
@georgev93 Workaround for you might be going back to the previous set up of layout without sending unicode characters except for percent char, more [here](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/acd50fe1d1e5c9c2a2b73a70ac93ba564e400386), but I recommend you figure out what is wrong with ctrl + shift + u.

ldrahnik, 2023-09-16:
@georgev93 
> Num lock is turned on when I activate my numpad (which is the intended behavior, correct?).

Yes

georgev93, 2023-09-16:
My iBus settings for unicode match, however, my ibus daemon wasn't running. When I started the daemon to check I received a system notification that ibus changes won't take effect on Wayland and that I should use systemsettings5 instead.

After enabling the daemon, I still had the same behavior (ctrl+shift+u 25 printed u% and the numpad still printed u39)

georgev93, 2023-09-16:
Looks like this is [a known limitation in Walaynd](https://gitlab.gnome.org/GNOME/gtk/-/issues/5865). Is the virtual key combo of Ctrl+Shift+U hardcoded in this driver? If I was looking to hack this repo to tweak for my machine, where should I start looking?

ldrahnik, 2023-09-17:
@georgev93 `git pull` latest master, reinstall, I reverted numpad layouts which use `KEYS_KP` and not Unicode and set it up as default, it should help you send numpad keys except percent char and hashtag char.

ldrahnik, 2023-09-17:
@georgev93 Does work change of shortcut for unicode sending in gsettings for you? I mean `org.freedesktop.ibus.panel.emoji unicode-hotkey ['<Control><Shift>u']` For me with X11 it does work. I am considering loading the shortcut sequence from there, it is hardcoded in the driver code, you guessed right.

georgev93, 2023-09-17:
Wow! Very kind of you to make such a significant change to your repo to get to the bottom of this! I can confirm that the non-unicode driver is working as expected. Well, the calculator shortcut isn't working, but I bet I can figure that one out (especially with the help available in the readme).

As for the change of unicode shortcut: when I do `gsettings get org.freedesktop.ibus.panel.emoji unicode-hotkey` it does return `<Control><Shift>u`, so as far as gsettings is concerned, I think it was always set to the right key combo. I think (and I'm completely uneducated on this topic so please excuse an obvious or wrong statement here) gsettings was both not running and not in effect when I tried to run it thanks to my Plasma/Wayland environment honoring systemsettings5 instead of gsettings

georgev93, 2023-09-17:
Not that the calculator launch functionality is as big of a deal, but I did see this during the install script:

```
Do you want automatically try install toggling script for XF86Calculator key? Slide from top left icon will then invoke/close detected calculator app. [y/N]y
No such schema “org.gnome.settings-daemon.plugins.media-keys”
is installed io.elementary.calculator: 1
is installed gnome-calculator: 1
Automatic installing of toggling script for XF86Calculator key failed. Please create an issue (https://github.com/asus-linux-drivers/asus-numberpad-driver/issues).
```

The only calculator apps I have installed are `kcalc` and `kalk`. *I use Arch BTW*

ldrahnik, 2023-09-17:
@georgev93 Possibility rebinds the Unicode shortcut is WIP in the branch mentioned to this issue https://github.com/asus-linux-drivers/asus-numberpad-driver/issues/141, atm I know only about way how to do that using mentioned gsettings for Gnome. It concerns even to you as I said for percents or even hash tag char if I am correct.

ldrahnik, 2023-09-17:
@georgev93 Code is setting up shortcut atm only via `org.gnome.settings-daemon.plugins.media-keys`, feel free open another issue and consider a contribution to the code as I can not test it on Arch. Also feel free to add to the code detecting `kcalc` /  `kalk`.

closed 2023-09-17 by ldrahnik