asus-linux-drivers/asus-numberpad-driver #143: [Bug]: Backlight turns off after two minutes

umutd3401, 2023-09-26:
### Describe the bug

I want to keep the backlight on 30 minutes while idle, but after 2 minutes the backlight turns off. I changed 'disable_due_inactivity_time' but doesn't work. It turns back on when I touch anywhere on the touchpad and turns off again after two minutes.

### Expected behavior

Not turning off after two minutes.

### Desktop

- Program version master
- Distro Fedora 38
- Kernel 6.4.15-200.fc38.x86_64

ldrahnik, 2023-09-26:
@umutd3401 Look at this [touchpad power management issue](https://askubuntu.com/questions/1231126/touchpad-power-management-issue-sticky-cursor-input-detection-delay-autosus)

umutd3401, 2023-09-26:
Thank you for your help. I tried to change autosuspend_delay_ms and then control but it didn't work. I even turned everything  from powertop  in case I couldn't find the right one and also tried a different distro.

I don't know much about these stuff but do you think a touchpad input can be emulated to prevent suspend? For example, when I touch the edge of the touchpad, evtest detects event but nothing happens because it is an edge. I don't know if this will work or prevent suspend.

ldrahnik, 2023-09-27:
@umutd3401 Unfortunately what I have done up to this moment is an only attempt for my `i2c` (found during running `sudo ./install.sh` of this driver ```
$ sudo ./install.sh
...
Testing interface i2c-1 : failed
Testing interface i2c-2 : sucess
... ``` 
)

to change appropriate file `/sys/bus/i2c/devices/i2c-2/device/power/autosuspend_delay_ms` / `/sys/bus/i2c/devices/i2c-2/device/power/control` but it looks like that touchpad is disabled every time after almost 2 mins.

umutd3401, 2023-09-27:
I just installed Windows 11 to see if it's the same and yes, it has the same behavior, so I don't think there is a solution. I will contact ASUS but have no hope.

ldrahnik, 2023-10-21:
@umutd3401 Answered?

ldrahnik, 2024-04-02:
Can be reopened. Behaviour with installed official driver for windows is the same.

closed 2024-04-02 by ldrahnik