asus-linux-drivers/asus-numberpad-driver #144: [Bug]: Starting this as a systemd service crashes SDDM, if installed to root, fails to start if installed to user

Rekt3y, 2023-10-05:
### Describe the bug

First of all, in my experience this only happens under Fedora KDE. Workstation works fine.
If during installation I install to root, it crashes systemd and I have to uninstall it.
If I pick install to user, it just fails to start after login.
Starting manually after logging in works, but it's extremely annoying.

### Expected behavior

...It shouldn't crash SDDM, or fail to start.

### Desktop

- Program version 3.0.1
- Distro Fedora 38 KDE Spin
- Kernel 6.5.5-200.fc38.x86_64

ldrahnik, 2023-10-06:
@Rekt3y Hello, thank you that you mentioned this problem. I do not close this issue yet (already exist more specific issues for systemctl service, I mean for X11 branch and Wayland branch) but I need logs from boot, driver error log and log from installation of the driver. Thank you

ldrahnik, 2023-10-08:
Can be reopened.

closed 2023-10-08 by ldrahnik